RIOT-OS/RIOT #10897: sock_dtls: a DTLS sock API

pokgak, 2019-01-29:
#### Description

`sock_dtls` aims to provide a wrapper around DTLS stack in RIOT so that users can choose the underlying stack by a `USEMODULE` in the Makefile. 

I am aware of @rfuentess's [sock_secure]() and @danielinux's [wolfSSL]() effort and have reviewed it together with @tcschmidt, @waehlisch, @miri64 (who also helps me with the API), @MichelRottleuthner and others (feel free to chime in if there is anything I missed/had it wrong here). This API hopes  to solve the comments that we have, such as:

  1. Targeting DTLS and TLS is (mostly) not needed in RIOT use case. The users typically know from the start which underlying stack to use
  2. `connect`, `listen` in traditional socket has a different meaning when used in DTLS compared to TLS.
  3. Sending and receiving of packets using `sock_udp` internally. `sock_dtls` is responsible for encrypting, decrypting of the packets and handshake. This should help by minimizing changes when integrating with existing applications such as gcoap.

To implement this API, the implementors of stack must  provide the definition the structs (like in [sock_udp, sock_tcp](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/sock/include/sock_types.h)) through the modules/pkgs.

`sock_dtls` API:

```C
/**
 * Methods for sock_dtls_create.
 * Defines the DTLS protocol version to use.
 */
#define DTLSv1_SERVER
#define DTLSv1_CLIENT
#define DTLSv12_SERVER
#define DTLSv12_CLIENT

/**
 * Contains `sock_udp` used, keys and certificate.
 */
typedef struct sock_dtls sock_dtls_t;

/**
 * Contains security and handshake parameter of a client, etc.
 */
typedef struct sock_dtls_session sock_dtls_session_t;

/**
 * Array of all connected clients
 */
typedef struct sock_dtls_queue sock_dtls_queue_t;

/**
 * @brief Creates a new DTLS sock object
 *
 * @param[out] sock     The resulting DTLS sock object
 * @param[in] udp_sock  Existing UDP sock to be used
 * @param[in] method    Defines the method for the client or server to use.
 *
 * @return  0 on success.
 * @return value < 0 on error
 */
int sock_dtls_create(sock_dtls_t *sock, sock_udp_t *udp_sock, unsigned method);

/**
 * @brief Initialises the server to listen for incoming connections
 *
 * @param[in] sock          DTLS sock to listen to
 * @param[in] queue         The resulting sessions queue
 * @param[in] queue_array   Array of session objects
 * @param[in] len           Length of @p queue_array / Max allowed session
 *                          at any time
 */
void sock_dtls_init_server(sock_dtls_t *sock, sock_dtls_queue_t *queue,
                           sock_dtls_session_t *queue_array, unsigned len);

/**
 * @brief Establish DTLS session to a server. Execute the handshake step in DTLS.
 *
 * @param[in]  sock      DLTS sock to use
 * @param[in]  ep        Endpoint to establish session with
 * @param[out] session   The established session
 *
 * @return 0 on success
 * @return value < 0 on error
 */
int sock_dtls_establish_session(sock_dtls_t *sock, sock_udp_ep_t *ep,
                                sock_dtls_session_t *session);

/**
 * @brief Close an existing DTLS session
 *
 * @param[in] sock      DTLS session to close
 *
 * @return 0 on success.
 * @return value < 0 on error
 */
int sock_dtls_close_session(sock_dtls_t *sock);

/**
 * @brief Decrypts and reads a message from a remote peer.
 *
 * @param[in] sock      DTLS sock to use
 * @param[out] remote   DTLS session of the received data.
 *                      May be `NULL`, if it is not required by the application
 * @param[out] buf      Pointer where the data should be stored
 * @param[in] maxlen    Maximum space available at @p data
 *
 * @return The number of bytes received on success
 * @return value < 0 on error
 */
ssize_t sock_dtls_recv(sock_dtls_t *sock, sock_dtls_session_t *remote,
                       void *data, size_t max_len, uint32_t timeout);

/**
 * @brief Encrypts and send a message to a remote peer
 *
 * @param[in] sock      DTLS sock to use
 * @param[in] remote    DTLS session to use. Must not be NULL and
 *                      the session must be already established with
 *                      `sock_dtls_establish_session()` or
 *                      use the session returned by `sock_dtls_recv`.
 * @param[in] data      Pointer where the data to be send are stored
 * @param[in] len       Length of @p data to be send
 *
 * @return The number of bytes sent on success
 * @return value < 0 on error
 */
ssize_t sock_dtls_send(sock_dtls_t *sock, sock_dtls_session_t *remote,
                       const void *data, size_t len);

/**
 * @brief Destroys DTLS sock created by `sock_dtls_create`
 *
 * @param sock          DTLS sock to destroy
 *
 * @return 0 on success
 * @return value < 0 on error
 */
int sock_dtls_destroy(sock_dtls_t *sock);
```

### Useful links

rfuentess, 2019-02-01:
Sounds perfect for me :+1: 

Sadly, I was unable to retake my work with my original approach on the last months.   I'll try to help with this new approach.

kb2ma, 2019-02-05:
Thanks for pursuing this! The API looks good at this level: send/recv functions as a wrapper for the underlying UDP sock, with additional setup/teardown infrastructure. The timeout parameter for sock_dtls_recv() is essential for gcoap, as is the ability to interrupt it when listening indefinitely.

I suggest developing a very simple end to end implementation to get a feel for how the API works for applications.

kb2ma, 2019-02-05:
You also might want to look at OSCORE, and the recently open sourced c_OSCORE embedded implementation [1,2]. From an application perspective it would be nice if there was some commonality with a DTLS based approach, but I don't know if that is feasible.

[1] https://datatracker.ietf.org/doc/draft-ietf-core-object-security/
[2] https://mailarchive.ietf.org/arch/msg/core/PYheXvjx0vNZ3wPhB8-NaJ2Es8M

miri64, 2019-02-05:
> From an application perspective it would be nice if there was some commonality with a DTLS based approach, but I don't know if that is feasible.

OSCORE and DTLS are very different approaches to communication encryption (one is on the application layer, one is on the transport layer). For now we should focus to have a common API to one protocol and then the other and then it might make sense to have a common API for both on top of this. But I don't think that we should mix those two on the very entry level into the protocol. Otherwise, we risk getting too complicated and too overengineered.

kb2ma, 2019-02-05:
I agree on starting simply, @miri64! Perhaps the most valuable API comparison between DTLS and OSCORE would be at the level of message encryption/decryption.

pokgak, 2019-03-13:
I just made a PR #11177 providing TinyDTLS implementation of the API. Changes since last initial draft:
1. `sock_dtls_init`, `sock_dtls_close_session` and `sock_dtls_destroy` now have `void` as return value (previously `int`)
2. `sock_dtls_close_session` have additional `sock_dtls_session_t` as parameter

First post will be updated.

Following is what I'm considering right now and hope to get some feedback before I proceed with it:
1. Remove `sock_dtls_init` from the API and move `dtls_init` call that is needed by TinyDTLS to autoinit. wolfSSL doesn't have such init functions.

2.  Remove the queue from `sock_dtls_init_server` because TinyDTLS already has its own
internal queue to keep track of all established session. Having an additional queue will just be
a duplicate of it. Looking at wolfSSL PR, they also doesn't have any queue to to save established
session. So I think it will be okay to remove it.

    The new `sock_dtls_init_server` will then look like this:

```C
void sock_dtls_init_server(sock_dtls_t *sock);
```
Just leaving my thoughts out here. Removing the queue will also mean that  `sock_dtls_init_server` will do nothing more than setting the role of the endpoint to a DTLS_SERVER (see [code][1]). So maybe `sock_dtls_init_server` could be removed altogether from the API but I'm *really* not sure about this and leaning more towards leaving it there in case anything needs it in the future.

[1]: https://github.com/RIOT-OS/RIOT/pull/11177/commits/e3701adeba10c3c2085fd4647968e6871c18e34e#diff-c71021b55895d420e83f1d22409d8c37R260

pokgak, 2019-08-06:
This comment will be used to track the current status of DTLS sock implementation and integration with gcoap. I'll try to keep it updated.

### DTLS Sock TODO List

- ~Add (D)TLS credential management module credman #11564~ DONE
- Introduce DTLS sock API #11909 - IN REVIEW.
- Add implementation of the DTLS sock API for tinydtls #11943 - waiting for #11909 
- Extend credman to handle the second use case as described in this [comment][1] - Not started yet
- Add DTLS integration for gcoap using DTLS sock - WIP

[1]: https://github.com/RIOT-OS/RIOT/pull/11564#issuecomment-513171749

miri64, 2019-08-06:
@pokgak how about adding a checklist (see task list [here](https://guides.github.com/features/mastering-markdown/#GitHub-flavored-markdown) to the OP? This makes it easier to find it later ;-).

pokgak, 2019-08-06:
Thanks for the tip. I updated the OP with the checklist and deleted my comment with the list to avoid confusion.

pokgak, 2019-08-28:
Update: #11909 is now merged. OP updated.

miri64, 2020-07-04:
@pokgak what are your plans going forward on this?

pokgak, 2020-07-07:
I'm currently integrating the new async dtls into gcoap but it'll be a bit slow as I can't focus on this much. Currently, I can already send a simple coap message with some dirty changes here and there. I'll work on making separate PRs for this changes. Current progress can be tracked [here](https://github.com/pokgak/RIOT/tree/gcoap/add-dtls-v2).

miri64, 2020-07-07:
>     * Add implementation of the DTLS sock API for WolfSSL - WIP

Is this still on the table?

pokgak, 2020-07-07:
Yes. I do have a somewhat working implementation [here](https://github.com/pokgak/RIOT/commits/wolfssl_sock_dtls) before I got sidetracked by other things. That will require a refactor now that sock dtls API is a lot more different now.

miri64, 2020-11-05:
@janosbrodbeck since you basically already picked up "Add DTLS integration for gcoap using DTLS sock", might there be a way to coax you into also looking into implementing `sock_dtls` for WolfSSL? :grin:

miri64, 2020-11-05:
Or maybe @fjmolinas, who I know, is also doing `sock_dtls` stuff.

janosbrodbeck, 2020-11-05:
Coaxing would not be necessary ;) But want to finish the DTLS integration first, I expect feedback and lots of change requests when the PR arrives.

`Add general way to set/get sock_dtls_session_t from a sock_udp_ep_t and vice-versa` is also still an open task and could be important/useful. 

But depends on what is really meant by it. Restoring an existing session from an endpoint? (tinydtls seems to store some context internally - but just did a quick look into tinydtls last week). Would like to have a look at it before.

Flole998, 2023-02-28:
@pokgak Are you planning on continuing your work on that wolfSSL implementation? If not maybe I will find some time to pick up your work.

jhermanowski, 2026-08-24:
Im working on implementing the sock_dtls API for wolfSSL.
Im already working on cleaning up my code.
I think i will be able to create a PR in the next weeks, some time after the RIOT Summit.