RIOT-OS/RIOT #10909: fmt_hex_bytes is not safe with respect to both arguments

danpetry, 2019-01-31:
#### Description
The function `fmt_hex_bytes` formats bytes from an input hex string to an output char array. It runs through the hex string until it finds a null termination, copying it into the string buffer. However, it is up to the user to provide an output buffer which is at least as large as the input buffer, otherwise it will overrun the output buffer silently. Suggest putting a check on the size of the output buffer as well and returning and error if it is not at least as large as the input.

https://github.com/RIOT-OS/RIOT/blob/fd2ffd9ef77a8c25a320cdda91628e3e8691a72a/sys/fmt/fmt.c#L150

Also I'm not sure if the pointer argument should be referenced and handled within the function as an array, but correct me if I'm wrong.

kaspar030, 2019-01-31:
> However, it is up to the user to provide an output buffer which is at least as large as the input buffer, otherwise it will overrun the output buffer silently. 

Half as large. :)

> Suggest putting a check on the size of the output buffer as well and returning and error if it is not at least as large as the input.

Well, you're opening a can here. None of fmt's functions check the output buffer's length. This is by design, the caller needs to read the docs and ensure correct output buffer length.
When designing fmt that way, I rationalized ignoring three decades of C experience (leading to str*n*cpy etc) by us being in low-end embedded context. I hope that was not too bad of an idea.

danpetry, 2019-01-31:
Sure. I guess the decision would be validated (or otherwise) if we have a bunch users who are happily writing programs which run without runtime errors here and feeling ok about needing to check the docs (or not).

danpetry, 2019-01-31:
Closing the issue since this is by design. Thanks for explaining.

closed 2019-01-31 by danpetry