RIOT-OS/RIOT #10945: net/gnrc_tcp: timeout of connection establishment

PeterKietzmann, 2019-02-05:
#### Description
Disturbances during connection establishment (see [`gnrc_tcp_open_passive()`](https://github.com/RIOT-OS/RIOT/blob/68dc5b0d6e0422df9c3653c2cb8021fc35974aae/sys/include/net/gnrc/tcp.h#L114)) may lead to half-open state and thus to deadlock of a TCP endpoint. 

The problem occurred in two different scenarios. The first one was figured by @sabyse during testing. She found that a lost ACK after a SYN-ACK during 3-way handshake leads to an infinite number of SYN-ACK retransmissions, as the timer is never stopped. This leads to deadlock as the only TCB we allocate is never freed again. 

The second one will be described below in more detail and is present with the current *tests/gnrc_tcp_** applications due to **an other issue**.

#### Steps to reproduce the issue
- Set up tap bridge and two tap devices (e.g. by running `RIOT/dist/tools/tapsetup/tapsetup -c 2`)
- Open wireshark and monitor traffic on `tapbr0`
- Build and run *tests/gnrc_tcp_server* on Node A (`make all term`)
- Build and run *tests/gnrc_tcp_client* on Node B (`make all term PORT=tap1`)

(Adding a delay (for example`xtimer_sleep(5)`) before connection establishment [here](https://github.com/RIOT-OS/RIOT/blob/master/tests/gnrc_tcp_client/main.c#L67) prevents the initial SYN packet from being sent with `: :` source address and thus, successfully establishes the connection. However, this only fixes "the other issue" not the TCP timeout)

#### Expected results
Successful 3-way handshake establishes a TCP connection.

#### Actual results
- Node A sends a SYN packet with source address `: :`.
- Node B does not establish the connection.
- Node A retransmits the SYN with its correct source address.
- Node B responds RST-ACK and will never establish any connection again.

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "18.04.1 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-45-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.6.7 on Linux
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15rc1
                python2: Python 2.7.15rc1
                python3: Python 3.6.7
             coccinelle: missing
```

brummer-simon, 2019-02-05:
Would you like to create a second issue for the second scenario?

From my investigations from yesterday I am sure that both problems are related but caused by different sources. The referenced PR #10946 only addresses the first scenario.

PeterKietzmann, 2019-02-06:
@brummer-simon I will open an other issue but first wanted to sort out what's been discussed in #10950. Maybe these problems relate. However, if I'm not mistaken #10946 actually does address this issue as well. If you need further insights I could send you tome pcaps w/ and w/o your PR.

brummer-simon, 2019-02-06:
The initiali sending of SYN with Source addr :: is a problem and might be caused by a faulty test setup.
I my point of view thats the outcome of #10950. 

The changed behavior introduced by #10946 solves the deadlock problem but has nothing to do with sending ::. So from point of view they the issues are caused by different problems and should be handled in separate Issues and PR.

PeterKietzmann, 2019-02-06:
Please read the above description carefully. Initially sending from `: :` does **not** relate to TCP **but** it leads to a similar situation as with the missing last ACK of a handshake: The server is not able to establish a new connection. With "new" connection I'm talking about the client re-trying to initiate the handshake with his (correct) link local IPv6 source address the second time.

brummer-simon, 2019-02-06:
Ah okay. There is still room for improvement. A received SYN without any source or destination address should not be accepted as a valid SYN anyway and therefore a sending of a SYN+ACK/RST+ACK should not occur.

However you are right with the proposed change, at least the deadlock vanishes. RST+ACK is like SYN+ACK, something the receiver has to respond with an ACK. In case the the peer never ACKs the previous request, the RIOT Node would stay locked without #10946.
 
From what I have seen the sending of the initial SYN with source addr = "::" is either caused by
1) The setup in the current gnrc_tcp tests.
2) Something else in the network stack (I think this is unlikely).

To improve the situation:
1) The current test should be improved to cause as less problems as possible.
2) The TCP layer handling thread should, assert in cases where an invalid IP-Address enters or leaves the TCP layer.  With that check it might be easier to nail down such issues in the future.

I would like to address both measures soon. Any objections or additional ideas?

miri64, 2019-02-06:
> From what I have seen the sending of the initial SYN with source addr = "::" is either caused by

`::` is set when the source address selection algorithm can't find any address. Are the addresses on the interface all valid?

miri64, 2019-02-06:
With valid I mean: Is the address in [GNRC_NETIF_IPV6_ADDRS_FLAGS_STATE_VALID](http://doc.riot-os.org/gnrc_2netif_2ipv6_8h.html#a3cb79f744d506ffe88e5aafac08f3849) state?

brummer-simon, 2019-02-06:
I can check later today. At the beginning of the tests the IP-Address from the makefile is assigned.
After that I should be able to a the assigned Address is in a valid state.

brummer-simon, 2019-02-06:
> > From what I have seen the sending of the initial SYN with source addr = "::" is either caused by
> 
> `::` is set when the source address selection algorithm can't find any address. Are the addresses on the interface all valid?

This PR #10963 uses a different ip addresses and verifies them. At least on native the "::" goes away with this.

PR #10946 (2019-02-12): gnrc_tcp: syn_rcvd pkt loss fix
### Contribution description
This PR fixes #10945. Peter found a bug in gnrc_tcp() connection establishment procedure.

## Problem description
The Problem can be provoked by starting a RIOT Node with TCP in passive mode by calling gnrc_tcp_open_passive(). The function does not return until a connection was successfully established
or an error occurred. As soon as the RIOT Node received a SYN Packet, a SYN-ACK is added to the retransmission mechanism. By prohibiting the peer TCP to not send any response to the SYN-ACK,
gnrc_tcp() retransmits SYN+ACK infinitely an [... cut]
closed 2019-02-12 by PeterKietzmann