RIOT-OS/RIOT #10947: tests/gnrc_tcp_*: not working on boards

PeterKietzmann, 2019-02-05:
#### Description
Both test applications *tests/gnrc_tcp_server* and *tests/gnrc_tcp_client* do not work on current master using boards (I tried with samr21-xpro and iotlab-m3). It used to work on 2018.07 release branch though.

#### Steps to reproduce the issue
Build/flash/term one of the aforementioned applications to a board and wait for the serial output.

#### Expected results
The respective application should print some status information, among others IPv6 address, port, ...

#### Actual results
You see nothing because the board sees to hard fault before main() is reached. By enabling `ENABLE_DEBUG` in `auto_init` you can see at least some output before the board crashes.

#### Versions
```
Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.6.7 on Linux
                    git: git version 2.17.1
             coccinelle: missing
```

PeterKietzmann, 2019-02-05:
@smlng, @miri64 ideas?

miri64, 2019-02-05:
I'd just do a `git bisect` to find the regression.

miri64, 2019-02-05:
0a9793c49bf7d127f3281aaaa1b5584ee15b02c6 introduced the regression according to my bisect. Will investigate.

miri64, 2019-02-05:
However, there is more to this. In 0a9793c the hardfault handler is still called and I get a dump of all registers. Later I don't see that anymore. I will investigate this as well.

miri64, 2019-02-05:
Ok, it is confirmed to be https://github.com/RIOT-OS/RIOT/commit/0a9793c49bf7d127f3281aaaa1b5584ee15b02c6 now.

PR #10950 (2019-02-05): gnrc_sixlowpan_iphc: fix _compressible()

### Contribution description
When either `gnrc_sixlowpan_iphc_nhc` or `gnrc_udp` is not compiled in `_compressible()` never returns `true`. This causes the `dispatch` snip in `gnrc_sixlowpan_iphc_send()` to be of length 0, meaning `dispatch->data` is `NULL`, causing possible crashes when trying to send IPv6 packets over 6LoWPAN without NHC or UDP.

### Testing procedure
`tests/gnrc_tcp_server` and `tests/gnrc_tcp_client` should work on a 6Lo-based board.
`examples/gnrc_networking` should still be able to exchange UDP packets. Those packets should be co [... cut]
closed 2019-02-05 by PeterKietzmann