RIOT-OS/RIOT #10956: tests/gnrc_tcp_*: #2 not working on 6Lo-based boards

PeterKietzmann, 2019-02-06:
#### Description
In #10947 and #10950 we found and fixed an issue which let a board hard fault running the above mentioned apps. Now that this is fixed it turns out the code is executed but the protocol doesn't work. Looking at a sniffer, all I see is Router Solicitations, nothing more. Digging a bit deeper it seems that TCP packets are dropped within the network stack:

- The packet is dropped in the NIB [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/nib/nib.c#L209) because `netif` is NULL
- `netif` is NULL because it was initialized [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L527) and **not** set again baceause the `pkt->type` is **not** `GNRC_NETTYPE_NETIF`, compare [this statement](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L533)

I think this is missing somewhere:
`gnrc_pktsnip_t *netif = gnrc_netif_hdr_build(NULL, 0, NULL, 0);`

#### Steps to reproduce the issue
- Optionally enable debug messages in [gnrc_ipv6.c](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c) and [nib.c](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/nib/nib.c) 
- Optionally use a 802.15.4 sniffer on channel 20 (RIOT defualt channel)

**server**
``TCP_LOCAL_ADDR=<ll_addr_server> BOARD=samr21-xpro make flash term SERIAL=XXX PORT=/dev/ttyXXX``

**client**
``TCP_TARGET_ADDR=<ll_addr_server> BOARD=samr21-xpro make flash term SERIAL=XXX PORT=/dev/ttyXXX``

#### Expected results
TCP connection should be established.

#### Actual results
No packets are sent -> no connection establishment. Instead, with above mentioned and enabled debug messages, would should get information from the NIB module that host is unreachable.

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "18.04.1 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-45-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.6.7 on Linux
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15rc1
                python2: Python 2.7.15rc1
                python3: Python 3.6.7
             coccinelle: missing
```

miri64, 2019-02-06:
I had a quick look at the stack and the `gnrc_netif_hdr_build()` is there. Here

https://github.com/RIOT-OS/RIOT/blob/3e5658d40739dfa8e482bb3ed1ee595fd656fa96/sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c#L114-L135

and here

https://github.com/RIOT-OS/RIOT/blob/3e5658d40739dfa8e482bb3ed1ee595fd656fa96/sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c#L223-L236

For the first case, I'm not quite sure why it is only build when the source is link-local. To me it seems sensible to always use the information available (i.e. if a netif header is there, use it), but since we are only using link-local addresses and that case is only for constructing reset replies, this can't be the problem I believe.

For the second case it can only be, that `tcb->ll_iface` might not be set or set to a wrong value.

miri64, 2019-02-06:
https://github.com/RIOT-OS/RIOT/blob/3e5658d40739dfa8e482bb3ed1ee595fd656fa96/tests/gnrc_tcp_client/Makefile#L10

Hardcodes the interface identifier to 5. Since we use 6LoWPAN, which adds an extra thread, this might not be the correct one.

PeterKietzmann, 2019-02-06:
> Hardcodes the interface identifier to 5

I used `TCP_TARGET_ADDR=<ll_addr_server>` without any interface info I think that's the probelm

PeterKietzmann, 2019-02-06:
With `%6` at least this issue seems to be fixed. Improving the documentation should do. However, now I see floods of TCP packets which doesn't seem correct. Here I would ask @brummer-simon for clarification

brummer-simon, 2019-02-06:
Sorry guys I can't take a look into the issue currently (I am supposed to do the things I am getting paid for during the day;). Maybe I find the time this evening to find something out.

There is still the Problem that I can test only on native currently.

miri64, 2019-02-06:
> Sorry guys I can't take a look into the issue currently (I am supposed to do the things I am getting paid for during the day;). Maybe I find the time this evening to find something out.
> 
> There is still the Problem that I can test only on native currently. Does the TCP packet flooding occur on real boards or on native too?

I think this goes into the direction I already pointed out when we had the discussion about TCP accepting strings for `connect`. You promised to fix this at some point anyway ;-).

miri64, 2019-02-06:
> Improving the documentation should do.

Yepp.

brummer-simon, 2019-02-06:
> I think this goes into the direction I already pointed out when we had the discussion about TCP accepting strings for `connect`. You promised to fix this at some point anyway ;-).

Jeah we talked about that ;). I think the outcome was that this is part of the integration of gnrc_tcp into the sock interface.

PeterKietzmann, 2019-02-06:
>  Does the TCP packet flooding occur on real boards or on native too

Only on hardware (samr21-xpro - samr21-xpro)

brummer-simon, 2019-02-06:
Hm this is something I can't test at home :(. 

Could you capture the traffic and send me the wireshark dump?

miri64, 2019-02-06:
> Hm this is something I can't test at home :(.

But in the IoT-LAB ;-).

brummer-simon, 2019-02-06:
> > Hm this is something I can't test at home :(.
> 
> But in the IoT-LAB ;-).

That might be true. Do you have any reference on how to use that?

I just want to collect what works and what not:
1) Currently native works without Packet flooding and with the Device ID 5.
2) Non-native works but has Packet flooding. And the Device ID 6 is required, because
there is the additional 6LoPan Layer native doesn't use. 

=> Hardcoding a device identifier is a bad idea. Maybe I could use a Unique local address (fc00/7)
for those tests. Then there is no need for a changing device identifier, that causes obviously problems between running the test on native and non-native. This might improve the situation.

I am not sure what causes the described Packet Flood, the last changes to gnrc_tcp itself were #9233 (there was the previously mentioned discussion). It is from last may but as far as I know it worked on the samr-xpro without any odd behavoir.

miri64, 2019-02-06:
> That might be true. Do you have any reference on how to use that?

https://www.iot-lab.info/tutorials/. There are a diverse number of platforms available: https://www.iot-lab.info/hardware/. You can even sniff remotely: https://www.iot-lab.info/tutorials/monitoring-sniffer-m3/.

miri64, 2019-02-06:
> 2\. Non-native works but has Packet flooding. And the Device ID 6 is required, because
>     there is the additional 6LoPan Layer native doesn't use.

Maybe the MSS is not configured properly for IEEE 802.15.4 traffic and that's causing some kind of flooding?

brummer-simon, 2019-02-06:
GNRC_TCP MSS is configured to be aligned with the minimum IPv6 MTU of 1280 byte via a define.
`
#ifndef GNRC_TCP_MSS
#ifdef  MODULE_GNRC_IPV6
#define GNRC_TCP_MSS (1220U) /**< If IPv6 is used. Get MSS = 1280 - IPv6 Hdr - TCP Hdr = 1220 */
#else
#define GNRC_TCP_MSS (576U) /**< Default MSS */
#endif
#endif
`
The TCP MSS is therefore independent from the 6LoPAN layer.

miri64, 2019-02-06:
> GNRC_TCP MSS is configured to be aligned with the minimum IPv6 MTU of 1280 byte via a define.
`#ifndef GNRC_TCP_MSS #ifdef MODULE_GNRC_IPV6 #define GNRC_TCP_MSS (1220U) /**< If IPv6 is used. Get MSS = 1280 - IPv6 Hdr - TCP Hdr = 1220 */ #else #define GNRC_TCP_MSS (576U) /**< Default MSS */ #endif #endif`
The TCP MSS is therefore independent from the 6LoPAN layer.

Okay. I believe that is in line with the [TCP over constrained node networks draft](https://tools.ietf.org/html/draft-ietf-lwig-tcp-constrained-node-networks-04#section-4.1.1).

miri64, 2019-02-06:
(6Lo only fragments when packets don't fit into the L2's PDU)

brummer-simon, 2019-02-06:
During the test procedure a 2048 byte sized buffer, filled with a test pattern if transmitted in both directions. This should result in two IPv6 Packets. One with 1220 byte payload and the other with 828 byte payload. This transmission happens in both ways. I am pretty sure that there is some fragmentation involved.

miri64, 2019-02-06:
> During the test procedure a 2048 byte sized buffer, filled with a test pattern is transmitted. This should result in two IPv6 Packets. One with 1220 byte payload and the other with 828 byte payload. This transmission happens in both ways. I am pretty sure that there is some fragmentation involved.

Ok, but those should not show up in the sniffer as TCP traffic, but rather 6LoWPAN packets with one (reassembled) TCP datagram in the end.

PeterKietzmann, 2019-02-06:
@brummer-simon check [this pcap](https://owncloud.inet.haw-hamburg.de/index.php/s/ZWVbzFB4NE7KIel). As you can see, handshake seems somewhat ok but then weird stuff happens. However, I'd prefer **not** to analyze remotely and maybe wait for the next Hack'n'ACK.

brummer-simon, 2019-02-06:
> @brummer-simon check [this pcap](https://owncloud.inet.haw-hamburg.de/index.php/s/ZWVbzFB4NE7KIel). As you can see, handshake seems somewhat ok but then weird stuff happens. However, I'd prefer **not** to analyze remotely and maybe wait for the next Hack'n'ACK.

That would play into the fragmentation theory. The handshake has no payload.

@martine - I have some vague memory that the TCP inspection view was somewhat broken when I comes to using TCP over 6LoPAN. This was few years back but it could is still not fixed.

@PeterKietzmann - To be really sure if this is fragmentation on 6LoPAN or really entire TCP Packets, you could add a debug print in "sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c".
Every Packet entering and leaving the TCB comes through there including retransmits. Could you check that for me?

miri64, 2019-02-06:
> @miri64 - I have some vague memory that the TCP inspection view in wireshark was somewhat broken when It comes to using TCP over 6LoPAN. This was few years back but it could be still not fixed.

I have no idea about that. I'm not really used to dissecting TCP traffic ;-D.

miri64, 2019-02-06:
(btw it's either called 6Lo*W*PAN [if you talk about IPv6 over IEEE 802.15.4] or 6Lo [if you talk about IPv6 over any unspecified CNN] not 6LoPAN ;-))

miri64, 2019-02-06:
> => Hardcoding a device identifier is a bad idea. Maybe I could use a Unique local address (fc00/7)
> for those tests. Then there is no need for a changing device identifier, that causes obviously problems between running the test on native and non-native. This might improve the situation.

I think if you are using ULAs you also could use `2001:db8::/32` (the IPv6 prefix version of `example.org`). Just make sure you assign a global IPv6 address marked as *valid* to *both* nodes. Otherwise the network bootstrapping in 6Lo-ND will give you hell ;-) (for DAD a border router is required e.g.).

brummer-simon, 2019-02-06:
> I think if you are using ULAs you also could use `2001:db8::/32` (the IPv6 prefix version of `example.org`). Just make sure you assign a global IPv6 address marked as _valid_ to _both_ nodes. [...]

Thanks. I'll try that and keep you updated.

miri64, 2019-02-06:
(oh and of course routes need to be configured as well)

brummer-simon, 2019-02-06:
Thanks Martine for the valuable input.
This is the PR #10963 what uses the different IP Addresses. I was only able to test it on native.
@PeterKietzmann would you test the PR on a samr21-xpro with 6LoWPAN (;-)) and check if the Packet flooding still occurs?

brummer-simon, 2019-03-24:
I found the cause of the Packet Flooding. This PR should fix it:  #11257
@PeterKietzmann could you verify it please.

brummer-simon, 2019-03-30:
Since #11257 got merged, this Issue can be closed right?

miri64, 2019-03-30:
I think so. @PeterKietzmann please reopen if you disagree.

closed 2019-03-30 by miri64