RIOT-OS/RIOT #10969: net: netdev_driver_t::send() doc unclear

maribu, 2019-02-08:
#### Description
The [documentation of `netdev_driver_t::send()`](http://api.riot-os.org/structnetdev__driver.html#a39585137953a92c1c58b5e0ad3262096) is unclear on the return value.

#### Steps to reproduce the issue
Read the doc regarding the return value.

#### Expected results
The reader is aware what the return value should be.

#### Actual results
The reader is confused, as the correct return value on success could be any of the following:

- The number of bytes of layer 2 PDU send
- The number of bytes of the layer 2 frame send without hardware supplied overhead (e.g. preamble, sync word, CRC, ...)
- The number of bytes of the layer 2 frame send including any overhead

#### Versions
Current master

maribu, 2019-02-11:
Looking into the current drivers, variant 2 seems to be the predominant approach: 

Drivers implementing variant 1. (PDU):
- None

Drivers implementing variant 2. (frame size without FCS, preamble, etc.)
- `at86rf2xx`
- `enc28j60`
- `encx24j600`
- `esp_now`
- `ethos`
- `mrf24j40`
- `nrfmin`
- `kw2xrf`
- `slipdev
- `sx127x` (?)

Drivers implementing variant 3 (frame size with FCS, but without preamble etc.)
- `cc110x` (current master version, not my PR)
- `cc2420`

Drivers implementing variant 4 (frame size including all overhead)
- None

(I might have classified some wrong, as I only looked briefly at it.)

miri64, 2019-08-28:
@maribu I assume this issue still persists? How about converting it into a tracking issue?

miri64, 2020-07-04:
> @maribu I assume this issue still persists? How about converting it into a tracking issue?

Ping?

maribu, 2020-07-04:
Sorry for the late reply.

> I assume this issue still persists?

Yes. "number of bytes sent" is not really clear (at least to me), what it should include.

> How about converting it into a tracking issue?

Sure. But maybe a decision on what the correct behavior is should be done first. If interpretation 2 of the sentence is chosen, only few drivers need to be adapted.

Maybe the first thing to do after the decision is to update the documentation, so that new drivers implement the behavior correctly right away.

miri64, 2020-07-05:
As there is no clear default on what device overhead entails (not even among IEEE 802.15.4 devices), I would also prefer variant 2.

maribu, 2022-09-19:
`netdev_driver_t::confirm_send()` is clear in its wordings on what to return. It will take a bit of time until all drivers will make use of that, but the immediate issue of the doc being unclear is solved.

closed 2022-09-19 by maribu