RIOT-OS/RIOT #10979: Missing SPI pre-scalars for STM32F7

bluephoton, 2019-02-09:
#### Description
**File**: /RIOT/boards/common/stm32/include/cfg_spi_divtable.h
This file doesn't have values corresponding to APBx clock = 54M, and 108M

#### Steps to reproduce the issue
This was discovered while trying to add sd-card support, hence enabling SPI. I got compiler error. ABPx clocks need to match the values above to get this error.

Had to add these values (used existing utility to calculate it):

```
#define CFG_SPIDIV_54               \
    {       /* for 54000000Hz */    \
        7,  /* -> 210937Hz */       \
        6,  /* -> 421875Hz */       \
        5,  /* -> 843750Hz */       \
        3,  /* -> 3375000Hz */      \
        2   /* -> 6750000Hz */      \
    },

#define CFG_SPIDIV_108              \
    {       /* for 108000000Hz */   \
        7,  /* -> 421875Hz */       \
        7,  /* -> 421875Hz */       \
        6,  /* -> 843750Hz */       \
        4,  /* -> 3375000Hz */      \
        3   /* -> 6750000Hz */      \
    },
```

File with complete changes attached: 
[cfg_spi_divtable.h.txt](https://github.com/RIOT-OS/RIOT/files/2847538/cfg_spi_divtable.h.txt)

#### Expected results

Should compile successfully.

#### Actual results
#error "CFG_SPI_DIVTABLE: no prescalers for selected APB1 bus clock defined"

#### Versions
Operating system: Windows 10 Pro
Fairly recent RIOT.

PeterKietzmann, 2019-02-11:
@bluephoton I'm not quite sure how to reproduce. Furthermore, do I see it correctly that this is more a feature request that can be fixed by adding a configuration to *cfg_spi_divtable.h*? To contribute changes to RIOT please have a look at [Contribut code](https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT#contribute-code) and our [Developement procedures](https://github.com/RIOT-OS/RIOT/wiki/Development-procedures).

bluephoton, 2019-02-12:
@PeterKietzmann, I'm not sure of a simple repro. I think if you change the CLOCK_CORECLOCK value on a nucleo board to be 216MHz build will fail. But for this to happen you need to be using SPI (your periph_conf.h needs to include cfg_spi_divtable.h)

It doesn't look like a feature request to my untrained eye, since RIOT already have support for F7 cpu - which support this clock frequency.

Unfortunately I have very little time to spend on this project otherwise I'd have tried to send a proper PR, I just shared the code in case it will help...

PeterKietzmann, 2019-02-13:
@bluephotonthanks for the feedback. Would you be so kind to adapt the issue description? That means, which application did you try to build for which board and what configurations have been changed during that build.

bluephoton, 2019-02-14:
@PeterKietzmann , sure thing! I'm actually using proprietary board that is not part of public RIOT, its mostly a nucleo like. Let me try to cook something simple with existing nucleo that can repro then will update. Stay tuned...

kaspar030, 2019-10-13:
PR'ed unmodified in #12437.

PR #12437 (2019-10-13): boards/stm32: add 54MHz and 108MHz SPI divtable entries

### Contribution description

#10979 attached a fixed file. I just turned that into this PR. The changes look sensible to me

Fixes #10979.

### Testing procedure

### Issues/PRs references

#10979
closed 2019-10-13 by aabadie