RIOT-OS/RIOT #11028: Setting up custom board with supported CPU and radio module

Khayhen, 2019-02-17:
Hi, I'm trying to implement a 6LoWPAN network with multiple boards. Unfortunately, I'm unable to afford the Atmel SAMR21-xpro. However, I managed to purchase some Arduino Mega and the MRF24J40 radio module. I connected the radio module to the Arduino Mega with the CS pin = 53, int pin = 2 and reset pin = 3. I tested this with the Arduino IDE and it works successfully. 

So in order to setup my own board, I duplicated the arduino-mega file in the boards directory and renamed it megaTest. Then in the file I added a Makefile.dep which I added the following code in.
<pre>ifneq (,$(filter gnrc_netdev_default netdev_default,$(USEMODULE)))
  USEMODULE += mrf24j40
endif
</pre>
I then added the following parameters code into include/board.h file.
<pre>#define MRF24J40_PARAMS_BOARD       { .spi = SPI_DEV(0), \
                                      .spi_clk = SPI_CLK_5MHZ, \
                                      .cs_pin = GPIO_PIN(1, 0), \
                                      .int_pin = GPIO_PIN(4, 4), \
                                      .reset_pin = GPIO_PIN(4, 5) }`
</pre>
 I successfully executed the tutorials but got stuck at the most important task at task-05 which is starting a gnrc network. I'm able to flash it but then only this appears and does not response to anything commands such as help or ps.
<pre>No handlers could be found for logger &quot;root&quot;
2019-02-18 02:32:05,301 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type &apos;/exit&apos; to exit.
</pre>  
What am I doing wrong? My apologies for any mistakes as I'm very new to these and need help urgently for my project. Any help or guide will be much appreciated. Thank you

PeterKietzmann, 2019-02-18:
@Khayhen I'm surprised that you could even build and link successfully. The arduino-mega2560 only has 8kB RAM which is not enough for *examples/gnrc_networking*. Maybe have a look at *examples/gnrc_minimal*.

Khayhen, 2019-02-18:
I see. What is the bare minimum to run a system like this? I'm also have a blue pill board which is based of the STM32F103C8T6 which has 64kB flash and 20kB SRAM. Will that be sufficient? I'm also trying to program that but having many difficulties, should I start a new issue on that?

PeterKietzmann, 2019-02-18:
Bluepill should be good I guess. To get an idea about minimum memory requirements you might wanna have a look at [this doc](https://arxiv.org/pdf/1801.02833.pdf). However, *gnrc_networking* is bigger as it comes with a shell and a collection of practical shell handlers.

PeterKietzmann, 2019-02-18:
@Khayhen can we close this issue again?

Khayhen, 2019-02-18:
@PeterKietzmann Noted. Thank you very much. Will open a new issue on programming the blue pill.

closed 2019-02-18 by Khayhen