RIOT-OS/RIOT #11029: Default example does not work on stm32f4discovery

yschroeder, 2019-02-18:
#### Description

The default example outputs nothing on stm32f4discovery and does not accept commands.

#### Steps to reproduce the issue

* `git clone https://github.com/RIOT-OS/RIOT.git`
* `cd RIOT/examples/default`
* `BOARD=stm32f4discovery make flash term`
* Boards flashes normally, pyterm starts up

#### Expected results

Device prints `Welcome to RIOT!` to the serial console. Input of `help` or `ps` prints output to the serial console.

#### Actual results

Device prints nothing to the console. `hello-world` example prints output as expected.

#### Versions

I tested this both on master and 2019.01 release.

```
Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 4.20.7-arch1-1-ARCH x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (GCC) 8.2.1 20181127
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.2.0
                avr-gcc: avr-gcc (GCC) 8.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: clang version 7.0.1 (tags/RELEASE_701/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.13.4
               cppcheck: Cppcheck 1.86
                doxygen: 1.8.15
                 flake8: missing
                    git: git version 2.20.1
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 3.7.2
                python2: Python 2.7.15
                python3: Python 3.7.2
             coccinelle: missing
```

PeterKietzmann, 2019-02-18:
The discovery boards don't support on-board UART over USB and need an external converter connected to `UART_DEV(0)` which is PA2(TX) and PA3(RX). Did you use one?

yschroeder, 2019-02-18:
Yes I did us an FTDI on PA2/PA3, the hello-world example worked fine and printed the expected output, `tests/leds` also works as expected, only the default example seems to hang.

PeterKietzmann, 2019-02-18:
ok I'll give it a try

kaspar030, 2019-02-18:
@yschroeder can you please try "tests/shell"? If that works, basic uart IO is working properly.
I don't have an stm32f4discovery. Does "make debug" work on the board?

yschroeder, 2019-02-18:
`tests/shell` works:

```
/home/yannic/gits/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB1" -b "115200"
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-02-18 09:41:02,055 - INFO # Connect to serial port /dev/ttyUSB1
Welcome to pyterm!
Type '/exit' to exit.
help
2019-02-18 09:41:04,778 - INFO # help
2019-02-18 09:41:04,780 - INFO # Command              Description
2019-02-18 09:41:04,782 - INFO # ---------------------------------------
2019-02-18 09:41:04,783 - INFO # start_test           starts a test
2019-02-18 09:41:04,785 - INFO # end_test             ends a test
2019-02-18 09:41:04,787 - INFO # echo                 prints the input command
2019-02-18 09:41:04,789 - INFO # reboot               Reboot the node
2019-02-18 09:41:04,791 - INFO # ps                   Prints information about running threads.
> reboot
2019-02-18 09:41:11,857 - INFO #  reboot
2019-02-18 09:41:11,857 - INFO # main(): This is RIOT! (Version: 2019.01)
2019-02-18 09:41:11,874 - INFO # test_shell.
> 2019-02-18 09:41:15,851 - INFO # Exiting Pyterm
```
And yes, `make debug` works.

PeterKietzmann, 2019-02-18:
*tests/shell* works, *examples/default* hangs here as will. Will look into it

PeterKietzmann, 2019-02-18:
The bug was introduced in `f2466fb8dc5b931ba4adbf902996985e4e2db71d` and relates to the RTC module. After removing `FEATURES_OPTIONAL += periph_rtc` from the Makefile, the *examples/default* works again.

PeterKietzmann, 2019-02-18:
The RTC initialization hangs [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/stmclk_common.c#L73) because the the low-speed oscillator never gets ready. @haukepetersen, @vincent-d can you help?

yschroeder, 2019-02-18:
According to the schematics, X3 on the discovery board is a 32khz clock crystal which is not mounted by the factory. I can confirm that X3 is missing on all 4 boards I have at hand. Maybe some boards exist with the clock crystal mounted? In any case, I guess RTC operation is not possible without the 32khz crystal, so it might be in error, that the board tries to initialize the RTC peripheral at all.

aabadie, 2019-02-18:
If there's no LSE, can you try with `#define CLOCK_LSE           (0)` [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/common/stm32/include/f4/cfg_clock_168_8_1.h#L41) ?

PeterKietzmann, 2019-02-19:
Same here, no crystals on X3. Disabling the external crystal as proposed by @aabadie fixes the issue. @aabadie do you have any suggestions how we handle the default configuration? Or may document it next to the board?

aabadie, 2019-02-19:
> do you have any suggestions how we handle the default configuration?

This default clock configuration is also used by other boards (nucleo-f429zi and stm32f429i-disc1) for which it works in master. IMHO the best solution would be to add a new default configuration with `CLOCK_LSE` defined to 0 (cfg_clock_168_8_0) that will be used by the stm32f4-discovery board. Maybe this file could share the core clocks defines with the already existing ones. This is just a suggestion.

cladmi, 2019-02-19:
I think the `stm32f3discovery` is also affected by this. Commenting `FEATURES_OPTIONAL += periph_rtc` in `examples/default` makes it work again.
I found during the release tests on `rc3` that `tests/periph_rtc` was failing on the board.

PeterKietzmann, 2019-02-21:
@aabadie that sounds reasonable. Anyone willing to address this, maybe @yschroeder? I'll mark the issue as Hack'n'ACK candidate.

aabadie, 2019-02-26:
@PeterKietzmann @yschroeder, see #11066

cladmi, 2019-02-26:
I applied the same changes as in #11066 for `stm32f3discovery` and `periph_rtc` test succeeds and `examples/default` does not crash.

```
diff --git a/boards/stm32f3discovery/include/periph_conf.h b/boards/stm32f3discovery/include/periph_conf.h
index 79bd8cad5..bf7105bac 100644
--- a/boards/stm32f3discovery/include/periph_conf.h
+++ b/boards/stm32f3discovery/include/periph_conf.h
@@ -40,7 +40,7 @@ extern "C" {
 #define CLOCK_HSE           (8000000U)
 /* 0: no external low speed crystal available,
  * 1: external crystal available (always 32.768kHz) */
-#define CLOCK_LSE           (1)
+#define CLOCK_LSE           (0)
 /* peripheral clock setup */
 #define CLOCK_AHB_DIV       RCC_CFGR_HPRE_DIV1
 #define CLOCK_AHB           (CLOCK_CORECLOCK / 1)
```

Does anyone who can justify this change do a pull request ? I can only say it make the thing work but not justify with hardware arguments.

jcarrano, 2019-02-26:
@cladmi The STM32F3DISCOVERY does _not_ include any crystal. You can verify it yourself by reading page 20 of the [User Manual](https://www.st.com/resource/en/user_manual/dm00063382.pdf).

closed 2019-03-04 by MichelRottleuthner