RIOT-OS/RIOT #11030: Problem flashing the Bluepill (STM32F130C8)

Khayhen, 2019-02-18:
Hi, I'm trying to setup a 6LoWPAN network and due to limited budget and resources, I decided to use the bluepill board (STM32F130C8) with the MRF24J40 radio module. However, I'm unable to flash the board. I'm using a FTDI USB-UART programmer with the pins connected in the following way and with boot 0 set to 1 and boot 1 set to 0:
5V    - 5V
GND - GND
TX    - A9
RX    - A10
 When I try to flash it with task-01, I get the following error.
<pre>### Flashing Target ###
Open On-Chip Debugger 0.10.0+dev-00634-gdb070eb8 (2018-12-31-01:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
none separate
Info : clock speed 1000 kHz
Error: open failed

/home/khayhen/Tutorials/task-01/../RIOT/Makefile.include:390: recipe for target &apos;flash&apos; failed
make: *** [flash] Error 1
</pre>
Based on the [documentation](https://doc.riot-os.org/group__boards__bluepill.html), I tried connecting the TX and RX pins to A3 and A2 respectively but still get the same error. What am I doing wrong? (I'm able to program the board with the Arduino IDE, so its not a defect in the board) Any help will be much appreciated thank you.

PeterKietzmann, 2019-02-18:
@MrKevinWeiss can you help here?

Khayhen, 2019-02-19:
After some more research, I notice the board is configured to use a ST-Link debugger. But in the OpenOCD wiki it mentioned that FTDI programmers can be used. Is this a problem with the programmer I'm using?

MrKevinWeiss, 2019-02-20:
Hey @Khayhen welcome to RIOT, sorry I didn't respond, I have been pretty busy.

The default flasher for the bluepill in RIOT is the openocd stuff.  You can follow the guide [here](http://www.marrold.co.uk/2017/08/flashing-bootloader-on-to-stm32-blue.html) to program with the uart bootloader (the stm32loader.py script comes with RIOT and can be found at <RIOTBASE>/dist/tools/stmloader32/stm32loader.py).  If you have the dfu-util bootloader already programmed in then you can use [this](https://github.com/inetrg/RIOT-Giveaway) to flash with the usb and make flash.

If the guide doesn't work there are a number of things that can be done for flashing,
- You can try using another st-link programmer and program the bluepill with SWD (if you have a nucleo board that can work)
- You can use the dfu-util usb bootloader to program, however that requires the usb bootloader to be flashed already and sometimes the R10 (I think) needs to be changed, though I have not have not done that yet.
- You can use the stm32loader.py and the uart to program it.  Now this is the option I think you are currently trying to do, RIOT doesn't have it integrated with the make flash so you will have to run the script yourself.  The standard way is to use this script to flash the dfu-util usb bootloader then just flash with the usb after that.

Please let me know if this solves your issue.  
_There are also many guides on how to program the bluepill, if you flash the usb bootloader make sure you have an up-to-date version._

Khayhen, 2019-02-20:
Hey @MrKevinWeiss, Thank you so much for your help and your time. I have ordered a ST-Link V2 programmer and should be arriving soon. In the mean time I'm just trying to install the dfu-util bootloader. But I can't find any proper bin file providing the dfu-util bootloader. Is there any link you can recommend?

MrKevinWeiss, 2019-02-21:
I used [this](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/bootloader_only_binaries/generic_boot20_pc13.bin), just be careful as the last commit message for it was:

> "Changes so that generic_pc13 version uses Boot1 jump link as the button pin so that the bootloader holds in DFU mode. Also rebuilt and tested generic_pa1 to check no side effects of these changes"

[here](https://medium.com/@paramaggarwal/programming-an-stm32f103-board-using-usb-port-blue-pill-953cec0dbc86) is another guide.

just using an stlink is the best way I think (once it arrive).

Have fun! (and can I close the issue?)

Khayhen, 2019-02-24:
Hi, so I managed to get my STLink V2 but I'm not sure if its flashing properly. So when I flash with the basic task-01, I get a "Done Flashing" but when I hook up a USB-UART converter to pins A2/A3 and open pyterm, I don't get any messages. I also wrote another simple blink code to blink PC13(which was tested on an arduino board succesfully). I flashed the bluepill and got the same message but the on board LED doesn't blink. This is the message I get upon flashing.
<pre>### Flashing Target ###
Open On-Chip Debugger 0.10.0+dev-00634-gdb070eb8 (2018-12-31-01:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
none separate
Info : clock speed 1000 kHz
Info : STLINK V2J17S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.258964
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 42327 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f1x.cpu       hla_target little stm32f1x.cpu       running
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000464 msp: 0x20000200
auto erase enabled
Info : device id = 0x20036410
Info : flash size = 128kbytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20000200
wrote 18432 bytes from file /home/khayhen/Tutorials/task-01/bin/bluepill/Task01.elf in 1.066452s (16.878 KiB/s)
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20000200
verified 17540 bytes in 0.294756s (58.112 KiB/s)
shutdown command invoked
Done flashing
</pre>

This is the mesage I get upon opening pyterm.
<pre>/home/khayhen/Tutorials/RIOT/dist/tools/pyterm/pyterm -p &quot;/dev/ttyUSB0&quot; -b &quot;115200&quot;
No handlers could be found for logger &quot;root&quot;
2019-02-24 14:31:59,958 - INFO # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type &apos;/exit&apos; to exit.
</pre>
I have a feeling that it is not flashing properly. I notice a couple of things, one of which is that the boards file in the RIOT directory for tutorials and latest RIOT release are different, although I have tried both boards file and still get the same result. Secondly, is that based on the bluepill [doc](https://doc.riot-os.org/group__boards__bluepill.html), the bluepill is defaulted to flash with 64kb but I seem to be flashing it with 128kb but I'm not sure if this a problem. Anything you can recommend to fix this problem? Any help will be much appreciated. Thank you for your time.

MrKevinWeiss, 2019-02-24:
The stdio uart pins usually default to device 0.  According to [the periph configuration](https://github.com/RIOT-OS/RIOT/blob/master/boards/common/stm32f103c8/include/periph_conf.h) device 0 is actually a10 and a9.
```
        .rx_pin     = GPIO_PIN(PORT_A, 10),
        .tx_pin     = GPIO_PIN(PORT_A, 9),
```

So you could try to get terminal output by connecting to those pins.

_(A2 and A3 are for the rom uart bootloader I think)_

As for the LED it should be C13 and you should see it flashing (if something else isn't already enabled).  A few things to note about the LED is that it is sinking, so a 0 turns it on and a 1 turns it off.  Also the QA for the bluepills are not the best (that is one thing that keeps the cost so low) some have the LED connected to different pins.  This is the reason why there are so many different binaries for the [dfu_util bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/bootloader_only_binaries).  Now if you already tested the hardware it is probably not that.  You could also try to flash the RIOT/tests/led/ application.

Your almost there!

Khayhen, 2019-02-24:
@MrKevinWeiss I have tried all that you mentioned and still no luck. No LEDs blinking or anything on pyterm. I have also tried with other bluepills and also programmed the board normally with arduino IDE to ensure that the on-board LED is PC13. I have a feeling its something to do with the software side, since I can even program it with the ST-link using the arduino IDE on windows.

MrKevinWeiss, 2019-02-24:
Ok, I will try to confirm tomorrow morning on one of my devices.

Khayhen, 2019-02-25:
@MrKevinWeiss Thank you very much for your help and time. Is it possible if I could have your email? As I would like to if possible ask you some questions regarding the whole idea of this project I'm working on. It would really help a lot. Thank you.

MrKevinWeiss, 2019-02-25:
My contact info is [here](https://inet.haw-hamburg.de/members/kevin-weiss).

I was able to flash the bluepill with `BOARD=bluepill make flash -C test/leds/` using the swd pinout from the nucleo and it seems to flash correctly.

Khayhen, 2019-02-25:
@MrKevinWeiss May I know your boot configuration, gcc-arm-none-eabi version and st-link version please. Thank you

MrKevinWeiss, 2019-03-01:
_sorry for the late reply, I was at embedded world_
Could it be the boot mode of the bluepill?  Make sure that the BOOT0 and BOOT1 jumper are at 0 otherwise it will try to boot the rom bootloaders not the application code.  Like I said, it appears that the flashing is correct.

Here is my system info:
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "16.04.6 LTS (Xenial Xerus)"
                 Kernel: Linux 4.15.0-45-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: avr-gcc (GCC) 4.9.2
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
   xtensa-lx106-elf-gcc: xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6) 4.8.5
                  clang: clang version 5.0.0-3~16.04.1 (tags/RELEASE_500/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: "2.2.0"
xtensa-lx106-elf-newlib: "2.0.0"
               avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.12.3
               cppcheck: Cppcheck 1.72
                doxygen: 1.8.11
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.6.3 on Linux
                    git: git version 2.7.4
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.10.0+dev-00399-g09076d1 (2018-04-12-17:12)
                 python: Python 2.7.12
                python2: Python 2.7.12
                python3: Python 3.6.3
             coccinelle: missing
```

Khayhen, 2019-03-07:
Hi I manage to solve the problem. The problem was mainly in the drivers and compilers as I manage to program it when I setup a VM with vagrant. However, I fixed my Linux system reformatting the whole thing and installing the compilers step by step again. I had some trouble installing the stlink driver as there is no up-to-date guide on how to do that but I manage to do so after lots of experimenting. Thanks for the help and I will close this question.

closed 2019-03-07 by Khayhen