RIOT-OS/RIOT #11043: boards/z1: tests/rng resets board

PeterKietzmann, 2019-02-21:
#### Description
z1 boards crash running the application *tests/rng*. Changing the default PRNG to `prng_xorshift` doesn't help. telosb nodes don't seem to be affected.

#### Steps to reproduce the issue
- Build/flash/term *tests/rng* on a z1 node
- Run the `entropy` command

#### Expected results
- The application outputs a (float) entropy estimation and a number of 1000 samples

#### Actual results
- The board reboots shortly after printing some malformed output
`Calculated %02f bits of entropy from 1090509387 samples`
- The malformed output is fixed in #10999

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "18.04.2 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-45-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: 3.6.0 (mccabe: 0.6.1, pycodestyle: 2.4.0, pyflakes: 2.0.0) CPython 3.6.7 on Linux
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15rc1
                python2: Python 2.7.15rc1
                python3: Python 3.6.7
             coccinelle: missing
```

cladmi, 2019-02-21:
With the `wsn430-v1_3b` the board does not answer anymore after running the `source 0` test suite.
I imagine it can be linked with a stack size issue. I could not re-run the test because of the testbed maintenance in Strasbourg.

cladmi, 2019-02-22:
I re-tested on `wsn430-v1_3b`. I added `ps` command to see the stack usage output.
And indeed it was a stack issue.

I needed to add this to make it work on `wsn430-v1_3b`:

```CFLAGS += -DTHREAD_STACKSIZE_MAIN=\(3*THREAD_STACKSIZE_DEFAULT\) ```

And the relevant `ps` output:

```
2 | main                 | running  Q |   7 |    768 (  640) |     0x11a0 |     0x13c
```

I will re-check on other boards too.

cladmi, 2019-02-22:
Then with 2 times the default stack size:

* `arduino-mega2560`:
    `2 | main                 | running  Q |   7 |   1024 (  566) |      0xa65 |      0xd58`
* `msba2`:
    `2 | main                 | running  Q |   7 |   1024 (  860) | 0x4000017c | 0x400003c4 `

cladmi, 2019-02-22:
To be more precise with the stack, I put the `line_buffer` static and defined the stack with its usage:

``` diff
diff --git a/tests/rng/Makefile b/tests/rng/Makefile
index 3c75d2f5e..690dfa74a 100644
--- a/tests/rng/Makefile
+++ b/tests/rng/Makefile
@@ -14,6 +14,9 @@ USEMODULE += xtimer

 FEATURES_OPTIONAL += periph_hwrng

+# Default stack with printf + the tests buffer uint32_t[32]
+CFLAGS += -DTHREAD_STACKSIZE_MAIN=\(THREAD_STACKSIZE_DEFAULT+THREAD_EXTRA_STACKSIZE_PRINTF+32*sizeof\(uint32_t\)\)
+
 TEST_ON_CI_WHITELIST += native

 include $(RIOTBASE)/Makefile.include
diff --git a/tests/rng/main.c b/tests/rng/main.c
index ff26a95a4..621c618f6 100644
--- a/tests/rng/main.c
+++ b/tests/rng/main.c
@@ -260,7 +260,7 @@ static int cmd_speed(int argc, char **argv)
 int main(void)
 {
     puts("Starting shell...");
-    char line_buf[SHELL_DEFAULT_BUFSIZE];
+    static char line_buf[SHELL_DEFAULT_BUFSIZE];
     shell_run(shell_commands, line_buf, SHELL_DEFAULT_BUFSIZE);

     return 0;
```

When testing with ps enabled:

``` diff
diff --git a/tests/rng/Makefile b/tests/rng/Makefile
index 690dfa74a..c90e0243b 100644
--- a/tests/rng/Makefile
+++ b/tests/rng/Makefile
@@ -10,6 +10,8 @@ BOARD_INSUFFICIENT_MEMORY += arduino-duemilanove arduino-uno
 USEMODULE += fmt
 USEMODULE += random
 USEMODULE += shell
+USEMODULE += shell_commands
+USEMODULE += ps
 USEMODULE += xtimer
 
 FEATURES_OPTIONAL += periph_hwrng
diff --git a/tests/rng/tests/01-run.py b/tests/rng/tests/01-run.py
index e6313058c..3eb78bad9 100755
--- a/tests/rng/tests/01-run.py
+++ b/tests/rng/tests/01-run.py
@@ -12,6 +12,9 @@ from testrunner import run
 
 
 def testfunc(child):
+    child.sendline("ps")
+    child.expect("ps")
+    child.expect("> ")
     # RNG source
     child.sendline("source 0")
     child.sendline("seed 1337")
@@ -64,6 +67,10 @@ def testfunc(child):
     child.sendline("entropy")
     child.expect(re.compile(r"Calculated 0\.017\d{3} bits of entropy from 10000 samples\."))
 
+    child.sendline("ps")
+    child.expect("ps")
+    child.expect("> ")
+
 
 if __name__ == "__main__":
```

I get these ps stack usage outputs after the test so would work for these.

* `wsn430-v1_3b`: `640 (  512)`
* `arduino-mega2560`: `768 (  474)`
* `msba2`: `2688 (  732)`

We see that we may not need the added `32*sizeof(int)` but it would be 100% measured usage on `wsn430-v1_3b` but we could still be going over without being measured.

PeterKietzmann, 2019-02-25:
@cladmi thanks for the investigation. I agree and confirm that the stack size is the problem. However, I don't quite understand what you propose as a fix. `THREAD_STACKSIZE_DEFAULT` is relatively small for MSP430 platforms and *tests/rng* uses a reasonable amount of memory for the shell, all the test functions and the PRNG. I would say we either add `THREAD_STACKSIZE_DEFAULT*2` to the applications Makefile or we think of some Makefile dependency to increase stack size when `random` is used on MSP430 platforms.

cladmi, 2019-03-11:
The stack usage does come from `random` but from the implementation of `tests/rng/test.c`. It is this file that allocates tests buffers on the stack. That is why I add the `32*sizeof\(uint32_t\)`.

What I define is `THREAD_STACK_SIZE = defaut_stack_size_for_a_printf_thread + stack_used_for_the_tests_buffers` so no magic `*2` that increase by a platform specific version.

However, I have no real idea how the thread stack is supposed to be set in applications and such. What is the normal way of handling this ?

PeterKietzmann, 2019-03-13:
I propose to simply set the stack size for this application: `CFLAGS += -DTHREAD_STACKSIZE_MAIN=1024` no matter of the platform default. 

> What is the normal way of handling this ?

I think we have no "normal way" defined yet

cladmi, 2019-03-13:
By saying it is `normal stack + printf required stack + what you allocate on the stack` it is clear.
I do not like putting a `random magic value not tested on every platforms`.

PeterKietzmann, 2019-03-13:
Fine with me. Although setting `what you allocate on the stack = 32*sizeof(int)` seems a bit odd to me too. Anyway, I don't care so much as long as we limit this to *tests/rng* and eventually leave a note in the Makefile.

cladmi, 2019-04-17:
The implementation saying `32*sizeof(int)`is indeed unusual but I would like to keep the idea.
If you want to allocate a 2k buffer on your stack, you must configure the stack and not expect to work out of the box.
Do you have other suggestions on how it could be written ?

Another fix would be to move the buffers to a shared buffer statically allocated but it would be a different implementation.

PeterKietzmann, 2019-05-21:
No other suggestions. Please lets not change the actual implementation (static buffer etc.) now.

> `normal stack + printf required stack + what you allocate on the stack`

Is ok with me.  If you agree, would you open a PR?

cladmi, 2019-05-28:
@PeterKietzmann TODO for Hack'N'Ack :)

closed 2019-05-28 by PeterKietzmann