RIOT-OS/RIOT #11051: NimBLE port should be always building all NimBLE files

sjanc, 2019-02-21:
Building NimBLE for RIOT is part of Apache NimBLE CI now. Whenever new file is added to tree our CI fails on RIOT port due to hardcoded list of files in controller.mk and nimble.host.mk makefiles.

Could RIOT port simply build all .c file for NimBLE? This is how NimBLE is build with newt anyway so building all files is expected to work even if some of NimBLE features are disabled.

This way RIOT port wouldn't require updating per release and would easily allow us to use it for master branch testing.

Example hackish workaround we need to use now:(
https://github.com/JuulLabs-OSS/mynewt-travis-ci/pull/33/commits/235c309f8f0c42a3b23dfe4bc755898cfeb49901

haukepetersen, 2019-02-21:
Yes, this is certainly possible. Will create the needed fix in the next day(s).

haukepetersen, 2019-02-22:
A first draft in #11055, might have to still improve a little more, but hope that branch does work fine already with the NimBLE CI?!

sjanc, 2019-02-25:
Hi,

I've tested this with nimBLE CI and build works just fine.
Thanks for fixing this so swiftly!

haukepetersen, 2019-03-06:
the fix is merged, should be all good now. If not, let me know :-)

PR #11055 (2019-03-06): pkg/nimble: improve structure of Makefiles
### Contribution description
This PR reworks the Makfiles for the `nimble` package a little bit, to make it more robust and improve modularity a bit. I think I still have to do some improvements to the `Makefile.include`, but aside from that the situation should be improved with this PR already...

### Testing procedure
Simply verify that `exampels/nimble_gatt` is still working as expected

### Issues/PRs references
fixes #11051
closed 2019-03-06 by haukepetersen