RIOT-OS/RIOT #11088: Configure board to act as a Slave

Khayhen, 2019-03-01:
Hi, I'm trying to configure a board running RIOT OS to act as a slave to communicate with other MCUs through a standard communication protocol. My preferred communication would be SPI, but it seems that there is no support for SPI slave mode for RIOT. Which current communication protocol supports slave mode and how do you implement it?

MrKevinWeiss, 2019-07-30:
...UART so far.  There is some talk about doing an I2C slave implementation.

mrusme, 2019-08-28:
Are there any plans/timeline on implementing SPI slave mode? At least in my case UART is not an option unfortunately. :(

MrKevinWeiss, 2019-08-29:
Not anything from me.  I first want to get SPI master completed with regression tests before adding any additional features.  I don't know about others?
@dylad or @maribu might have something in their back pocket?

mrusme, 2019-08-29:
Thanks for getting back! Well, then I guess the SC16IS750 it is, as I'm simply short of UART pins on my master.

maribu, 2019-08-29:
I have considered to connect the power monitor board of our testbed nodes via I2C or SPI for a while now, but I have not yet written a single line of code. Sorry.

It appears however, that @roberthartung has an I2C slave implementation for the ATmega chips: https://github.com/RIOT-OS/RIOT/issues/10587. In case @mrusme is using an ATmega based slave and I2C would be an option, @roberthartung might be able to provide a solution?

I also have recently talked with @PeterKietzmann about SPI slave in RIOT. If I remember correctly, he is able to point some pitfalls to avoid, if someone wants to implement SPI slave support in RIOT.

mrusme, 2019-08-29:
@maribu thank you for providing further info on this topic!

Maybe we could re-label/re-name this issue as a feature request then? :)

roberthartung, 2019-08-29:
@maribu I would be happy to integrate my self into RIOT. However, the I2C interface should be changed / adopted accordingly. I would be happy to have such a discussion. The same could be applied to other periphs, such as SPI.

closed 2020-04-01 by stale[bot]