RIOT-OS/RIOT #11101: `nib route add` for routing to Raspberry Pi gateway not working

ducktalez, 2019-03-04:
I want to (dtls-secure) connect a Laptop with a nucleo Board over 6LoWPAN and a Raspberry Pi as router between the eth0 and 6LoWPAN interface. Connections between the RPi and the board work fine.

When I try to add the RPi as gateway (in examples/gnrc_networking) with 
`nib route add 7 :: <RPi 6lowpan IP>`, 
`nib route` shows `default* via <RPi 6lowpan IP> dev #7`.
I guess this should forward messages/pings, but nothing arrives on the RPi when sent to another IP. Wireshark on the RPi does not recognize any packages besides router solicitations.
Help would be largely appreciated, searching for a solution for weeks now.

ducktalez, 2019-03-05:
If someone has the same issue, nib works fine. Don't know, how i got it to work now, but I:
- deleted and cloned the whole repository again
- used the global iface address
- had to ping multiple times before it worked
...maybe this helps you out.

miri64, 2019-03-07:
The problem might have been the neighbor cache (check with `nib neigh` on the RIOT side or `ip -6 neigh` on the RaspPi side), but I'm not sure.