RIOT-OS/RIOT #11119: How to handle shared UART ISR for STM32F0

niziak, 2019-03-07:
#### Description

### Useful links

PeterKietzmann, 2019-03-07:
I don't quite understand the problem. Maybe @MrKevinWeiss or @yegorich can say more

MrKevinWeiss, 2019-03-07:
I don't think that is a bad solution.
When there is multiple things that can cause an interrupt you need some way of distinguishing between.

The only thing that I would do is try to make change the actual uart definitions (in this case 3 and 6) to the dev number (in this case 1 and 2)

so in the board periph config file you would have
```
#define UART_0_ISR          (isr_usart1)
#define UART_1_2_ISR        (isr_usart3_6)
```

and in the cpu common file
```
#ifdef UART_1_2_ISR
void UART_1_2_ISR(void)
{
    if (dev(UART_DEV(1))->ISR & (USART_ISR_RXNE | USART_ISR_ORE)) {
        irq_handler(UART_DEV(1));
    } else  {
        irq_handler(UART_DEV(2));
    }
}
#endif
```

The only downside is that if we want to support every combination it should add up (so if the uarts are not ordered in a nice way we could have `UART_1_2_ISR` and `UART_1_3_ISR` and `UART_0_2_ISR`).  Since they are #ifdef it shouldn't take up any extra space though.

On the other side  maybe have some generic function `UART_ISR` that iterates through each device to see if there is an interrupt that occured, that wouldn't be that fast though and would be a pain to coordinate (you have check both the int_flags and if they are enabled and bla bla bla).  I don't think that would be viable.

niziak, 2019-03-07:
Ok, I will create some universal macrology to support maximum possible number of shared USARTs (max I found is 6 for USART3 to 8).

Let explain on my configuration:
```c
static const uart_conf_t uart_config[] = {
    {        .dev        = USART1,        .irqn       = USART1_IRQn    },
    {        .dev        = USART3,        .irqn       = USART3_6_IRQn    },
    {        .dev        = USART5,        .irqn       = USART3_6_IRQn    }
};
```

Enable normal (not shared) USART1:
```c
#define UART_0_ISR        (isr_usart1)
```
Two shared USARTs:
```c
#define UART_SHARED_ISR      (isr_usart3_6)
```
And now it is needed to know which UARTs should be handled inside `UART_SHARED_ISR`.
I can use exsting defines:
```c
#define UART_1_ISR          (isr_usart3_6)
#define UART_2_ISR          (isr_usart3_6)
```
and add some additional preprocesor conditions (in uart.c) to do not enable generic handlers and handle specified UARTs inside shared handler.

Another way (to do not complicate) is to define new symbols, like: 
```c
#define UART_SHARED_ISR      (isr_usart3_6)
#  define UART_1_ISR_SHARED
#  define UART_2_ISR_SHARED
```

What way do you prefer?

aabadie, 2019-07-04:
I know it's a bit late for a reply but you could simply use a custom `arg` identifying each UART when initializing them:
```c
int uart_init(uart_t uart, uint32_t baudrate, uart_rx_cb_t rx_cb, void *arg);
```

That would give something like this:

```c
static void rx_cb(void *uart, uint8_t c)
{
    /* This is quite unsafe, but it's just an example */
    printf("Received '%c' from UART%d\n", c, (int)uart);
}

int main(void)
{
    /* Initialize USART3, e.g. UART0 in RIOT */
    uart_init(UART_DEV(0), 115200, rx_cb, (void *)0);

    /* Initialize USART5, e.g. UART1 in RIOT */
    uart_init(UART_DEV(1), 115200, rx_cb, (void *)1);
}
```

Instead of the printf it's safe to send an ipc message to another thread (with a message queue) that will process the characters out of the interrupt context.

aabadie, 2019-07-04:
> I know it's a bit late for a reply

Ok it's even a bit too late. After thinking of it a second time, my proposition makes no sense, so forget about it...

fjmolinas, 2020-01-06:
@niziak did you end up implementing this?

closed 2020-08-09 by stale[bot]