RIOT-OS/RIOT #11137: Setting up border router which connects to the internet

Khayhen, 2019-03-07:
Hi, I'm trying to setup 6LoWPAN network where the nodes can transmit data to a server someway or other. My preferred method would be MQTT-SN. My current hardware that I'm using is a SAMR21 Xplained-Pro board for the border router and the nodes. By following the [tutorial](https://github.com/RIOT-OS/RIOT/blob/master/examples/gnrc_border_router/README.md), I manage to setup the border router through ethos and ping the nodes from my Linux PC where my border router is connected to. However, it still can't connect to the internet. Hence, I attempted to follow the slip method but still unsuccessful. 

The method I'm using to test the connection to the internet is simply pinging the google.com IPv6 address. Here is what I did, I first connected a FTDI UART connector to the board PA22 and PA23 pins respectively and also connected a cable to the debug port. Then I flashed the board with the example/gnrc_border_router file with making changes to the Makefile to enable slip routing and commenting out ethos uhcp. The content of the Makefile is as below:
 <pre># name of your application
APPLICATION = gnrc_border_router

# If no BOARD is found in the environment, use this default:
BOARD ?= samr21-xpro

# This has to be the absolute path to the RIOT base directory:
RIOTBASE ?= $(CURDIR)/../..

BOARD_INSUFFICIENT_MEMORY := airfy-beacon arduino-duemilanove arduino-mega2560 \
                             arduino-uno b-l072z-lrwan1 blackpill bluepill calliope-mini \
                             cc2650-launchpad cc2650stk hifive1 maple-mini \
                             mega-xplained microbit msb-430 msb-430h \
                             nrf51dk nrf51dongle nrf6310 nucleo-f031k6 nucleo-f042k6 \
                             nucleo-f303k8 nucleo-l031k6 nucleo-f030r8 \
                             nucleo-f070rb nucleo-f072rb nucleo-f103rb \
                             nucleo-f302r8 nucleo-f334r8 nucleo-l053r8 \
                             nucleo-l073rz opencm904 saml10-xpro saml11-xpro \
                             spark-core stm32f0discovery telosb waspmote-pro \
                             weio wsn430-v1_3b wsn430-v1_4 yunjia-nrf51822 z1

# The following boards do not have an available UART
BOARD_BLACKLIST += mips-malta pic32-wifire pic32-clicker ruuvitag thingy52

# use ethos (ethernet over serial) for network communication and stdio over
# UART, but not on native, as native has a tap interface towards the host.
# ifeq (,$(filter native,$(BOARD)))
#   GNRC_NETIF_NUMOF := 2
#   USEMODULE += ethos

#   # ethos baudrate can be configured from make command
#   ETHOS_BAUDRATE ?= 115200
#   CFLAGS += -DETHOS_BAUDRATE=$(ETHOS_BAUDRATE) -DUSE_ETHOS_FOR_STDIO
# else
#   GNRC_NETIF_NUMOF := 2
#   TERMFLAGS += -z [::1]:17754
#   USEMODULE += socket_zep
# endif

# SLIP legacy compatibility
# Uncomment the lines below if you want to use SLIP with this example and don't
# forget to comment the lines above for ethos.
ifeq (,$(SLIP_UART))
# set default (last available UART)
SLIP_UART="UART_DEV(UART_NUMOF-1)"
endif   
ifeq (,$(SLIP_BAUDRATE))    
# set default   
SLIP_BAUDRATE=115200
endif

GNRC_NETIF_NUMOF := 2
INCLUDES += -I$(CURDIR)
CFLAGS += -DSLIP_UART=$(SLIP_UART)
CFLAGS += -DSLIP_BAUDRATE=$(SLIP_BAUDRATE)
# Include SLIP package for IP over Serial communication
USEMODULE += slipdev

# Include packages that pull up and auto-init the link layer.
# NOTE: 6LoWPAN will be included if IEEE802.15.4 devices are present
USEMODULE += gnrc_netdev_default
USEMODULE += auto_init_gnrc_netif
# Specify the mandatory networking modules for 6LoWPAN border router
USEMODULE += gnrc_sixlowpan_border_router_default
# Add forwarding table
USEMODULE += fib
# Additional networking modules that can be dropped if not needed
USEMODULE += gnrc_icmpv6_echo
# Add also the shell, some shell commands
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps

# include UHCP client
# USEMODULE += gnrc_uhcpc

# Optionally include RPL as a routing protocol. When includede gnrc_uhcpc will
# configure the node as a RPL DODAG root when receiving a prefix.
#USEMODULE += gnrc_rpl

# Comment this out to disable code in RIOT that does safety checking
# which is not needed in a production environment but helps in the
# development process:
DEVELHELP ?= 1

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1
TAP ?= tap0
IPV6_PREFIX ?= 2001:db8::/64

# ifeq (native,$(BOARD))
# TERMDEPS += uhcpd-daemon

# .PHONY: uhcpd-daemon

# uhcpd-daemon: host-tools
# 	$(RIOTTOOLS)/uhcpd/bin/uhcpd $(TAP) $(IPV6_PREFIX) &
# else
# # We override the `make term` command to use ethos
# # TERMPROG ?= sudo sh $(RIOTTOOLS)/ethos/start_network.sh
# # TERMFLAGS ?= $(PORT) $(TAP) $(IPV6_PREFIX)

# # We depend on the ethos host tools to run the border router, we build them
# # if necessary
# TERMDEPS += host-tools
# endif

include $(RIOTBASE)/Makefile.include

.PHONY: host-tools

host-tools:
	$(Q)env -u CC -u CFLAGS make -C $(RIOTTOOLS)

# Set a custom channel if needed
ifneq (,$(filter cc110x,$(USEMODULE)))          # radio is cc110x sub-GHz
  DEFAULT_CHANNEL ?= 0
  CFLAGS += -DCC110X_DEFAULT_CHANNEL=$(DEFAULT_CHANNEL)
else
  ifneq (,$(filter at86rf212b,$(USEMODULE)))    # radio is IEEE 802.15.4 sub-GHz
    DEFAULT_CHANNEL ?= 5
    CFLAGS += -DIEEE802154_DEFAULT_SUBGHZ_CHANNEL=$(DEFAULT_CHANNEL)
  else                                          # radio is IEEE 802.15.4 2.4 GHz
    DEFAULT_CHANNEL ?= 26
    CFLAGS += -DIEEE802154_DEFAULT_CHANNEL=$(DEFAULT_CHANNEL)
  endif
endif
</pre>

I then followed to run the tunslip program and got this output:
<pre><font color="#8AE234"><b>khayhen@UbuKhay</b></font>:<font color="#729FCF"><b>~/RIOT/dist/tools/tunslip</b></font>$ sudo ./tunslip6 2001:db8::1/64 -t tun0 -s /dev/ttyUSB0
********SLIP started on ``/dev/ttyUSB0&apos;&apos;
opened tun device ``/dev/tun0&apos;&apos;
ifconfig tun0 inet `hostname` up
ifconfig tun0 add 2001:db8::1/64
ifconfig tun0 add fe80::cb8:0:0:1/64
ifconfig tun0

tun0: flags=4305&lt;UP,POINTOPOINT,RUNNING,NOARP,MULTICAST&gt;  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 2001:db8::1  prefixlen 64  scopeid 0x0&lt;global&gt;
        inet6 fe80::40b2:af97:3dfd:6cc  prefixlen 64  scopeid 0x20&lt;link&gt;
        inet6 fe80::cb8:0:0:1  prefixlen 64  scopeid 0x20&lt;link&gt;
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
</pre>
I then open pyterm for the border router and added the following commands as instructed:
<pre>&gt; ifconfig 6 add 2001:db8::2
2019-03-08 00:11:30,045 - INFO #  ifconfig 6 add 2001:db8::2
2019-03-08 00:11:30,049 - INFO # success: added 2001:db8::2/64 to interface 6
&gt; nib neigh add 6 2001:db8::1
2019-03-08 00:12:06,079 - INFO #  nib neigh add 6 2001:db8::1
&gt; ifconfig 5 add 2001:db8::3
2019-03-08 00:12:20,590 - INFO #  ifconfig 5 add 2001:db8::3
2019-03-08 00:12:20,594 - INFO # success: added 2001:db8::3/64 to interface 5
</pre>
I managed to ping the tunslip and add the fibroute to ping google's addr but then it didnt work.
<pre>2019-03-08 00:13:07,792 - INFO #  ping6 2001:db8::1
2019-03-08 00:13:07,815 - INFO # 12 bytes from 2001:db8::1: icmp_seq=0 ttl=64 time=17.781 ms
2019-03-08 00:13:08,818 - INFO # 12 bytes from 2001:db8::1: icmp_seq=1 ttl=64 time=15.970 ms
2019-03-08 00:13:09,821 - INFO # 12 bytes from 2001:db8::1: icmp_seq=2 ttl=64 time=14.042 ms
2019-03-08 00:13:09,822 - INFO # 
2019-03-08 00:13:09,825 - INFO # --- 2001:db8::1 PING statistics ---
2019-03-08 00:13:09,830 - INFO # 3 packets transmitted, 3 packets received, 0% packet loss
2019-03-08 00:13:09,834 - INFO # round-trip min/avg/max = 14.042/15.931/17.781 ms
fibroute add 2001:4860:4860::8888 via 2001:db8::1 dev 6
2019-03-08 00:14:57,003 - INFO #  fibroute add 2001:4860:4860::8888 via 2001:db8::1 dev 6
ping6 2001:4860:4860::8888
2019-03-08 00:14:59,643 - INFO #  ping6 2001:4860:4860::8888
2019-03-08 00:15:02,658 - INFO # 
2019-03-08 00:15:02,662 - INFO # --- 2001:4860:4860::8888 PING statistics ---
2019-03-08 00:15:02,667 - INFO # 3 packets transmitted, 0 packets received, 100% packet loss
</pre>
Sorry for the long post, a few main questions are:
1. What is the difference between ethos and slip? (Can ethos connect to the internet)
2. How can I check if my tunslip is running properly? (If plug out the TX/RX cables the program still runs with no error)
3. How to get it to successfully transfer data with the internet?
Any help will be much appreciated. Thank you in advanced.

PeterKietzmann, 2019-03-13:
@miri64 can you help out here?

benemorius, 2019-07-07:
I actually got here as the top result of googling "riot border router" and I've realized after reading it that this will perhaps be a very commonly encountered thing so I thought it important to answer even if it's too late to help @Khayhen.

The information provided is very good but the other thing that helps complete the picture is the ipv6 routing table of the `UbuKhay` host that is running the tunslip6 command. It looks like this, although this is from a setup using ethos instead of tunslip:
```shell
UbuKhay# ip -6 r
2001:470:4bb0:ffff::/64 via fe80::2 dev tap0 metric 1024  pref medium
unreachable fd00:dead:beef::1 dev lo proto kernel metric 256  error -101 pref medium
fe80::/64 dev eth0 proto kernel metric 256  pref medium
fe80::/64 dev tap0 proto kernel metric 256  pref medium
default via fe80::ba27:ebff:fe7d:f917 dev eth0 metric 202  pref medium
default via fe80::2 dev tap0 metric 221  pref medium
```

I don't know whether there are additional problems but the one that sticks out is that you've used an ip range from `2001:db8::/32` which is the designated ipv6 address space for examples and documentation. It isn't routable on the public internet, so if you want to ping the internet you need to replace this with an ip network that gets routed to you. If you need some ipv6 address space it's easy to get an allocation from https://tunnelbroker.net/ and route it to your border router through a tunnel.

closed 2020-02-08 by stale[bot]