RIOT-OS/RIOT #11147: 6LoWPAN over BLE: Device not pingable

Citrullin, 2019-03-08:
#### Description
I am not able to ping my device with 6LoWPAN on BLE from my linux machine.
Possible issue: As @miri64 [mentioned in an issue](https://github.com/RIOT-OS/RIOT/pull/11067#issuecomment-470487587), the header compression might be the issue. So, I added ```USEMODULE += gnrc_sixlowpan_iphc``` to my Makefile. 

#### Steps to reproduce the issue
1. Since [gnrc networking with BLE does not work](https://github.com/RIOT-OS/RIOT/issues/11091) at the moment. You need to checkout the commit 60008cbb2e6afbf3a39116192fcac02b5454e87b. 
2. Flash the gnrc_networking example
3. Follow the [6LoWPAN BLE guide](https://github.com/RIOT-OS/RIOT/blob/master/pkg/nordic_softdevice_ble/README-BLE-6LoWPAN.md)
4. The ping ```ping6 [DEVICE_LOCAL_ADDRESS]%bt0``` responds with address unreachable.
In my case:
```From fe80::1a:7dff:feda:7113%bt0 icmp_seq=1 Destination unreachable: Address unreachable```

#### Expected results
Device to be pinagble.

#### Actual results
The device doesn't respond on pings. 

#### Versions
```
Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.13.3
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.17.1
             coccinelle: missing
```

miri64, 2019-03-08:
Can you also post the OS version you are using? Given that the Nordic Softdevice [only works with certain kernel versions](https://github.com/RIOT-OS/RIOT/pull/10816) this is kind of important.

miri64, 2019-03-08:
> Possible issue: As @miri64 [mentioned in an issue](https://github.com/RIOT-OS/RIOT/pull/11067#issuecomment-470487587), the header compression might be the issue. So, I added `USEMODULE += gnrc_sixlowpan_iphc` to my Makefile.

As you are using `gnrc_networking` (which includes `gnrc_sixlowpan_iphc`) and the referenced issue is about `gnrc_minimal` (which does not include it), I highly doubt that.

Citrullin, 2019-03-08:
> Can you also post the OS version you are using? Given that the Nordic Softdevice [only works with certain kernel versions](https://github.com/RIOT-OS/RIOT/pull/10816) this is kind of important.

"but newer kernel versions than 4.12 definitely don't work"
Shit... Thanks.

miri64, 2019-03-08:
@haukepetersen is currently working on a GNRC-integration for nimBLE which should be ready soon™, so the boards currently using the softdevices could hopefully use nimBLE in the future instead. Alternatively, (if possible) the softdevice needs to be ported to a newer version.

Citrullin, 2019-03-08:
@miri64 I would be really happy with nimBLE. I always prefer the open source solution. So does that mean @haukepetersen is implementing the IPSP characteristics for Bluetooth in nimBLE?

Citrullin, 2019-03-09:
@miri64 Not sure if it is really the linux kernel.
```
4.12.14-041214-generic
```
and same distro
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic
```
I get an older version, to check if it works there.

Citrullin, 2019-03-21:
Okay @miri64, I guess you are right. It probably has something to do with a legacy system and the usage of pre-specification features. I use a Bluetooth 4.0 USB Adapter with the nRF52832. So, the common ground should be Bluetooth 4.0. But Bluetooth 4.0 does not define the IPSP characteristics. The first time the IPv6 over Bluetooth specification appears is in the ["Internet Protocol Support Profile v1"](https://www.bluetooth.org/docman/handlers/DownloadDoc.ashx?doc_id=296307) specification. The specification defines Bluetooth 4.1 or higher as a requirement. So, the characteristics was introduces with Bluetooth 4.1. The complete specification of the [6LoWPAN over Bluetooth is defined in Bluetooth 4.2](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=286439).
Therefore, Bluetooth 4.2 is required to run 6LoWPAN over Bluetooth, but you can use the legacy software implementations to get it running in previous generations.

haukepetersen, 2019-03-28:
> @haukepetersen is implementing the IPSP characteristics for Bluetooth in nimBLE?

Yes, thats exactly what I am doing right now. PR should be coming up 2nd week of April (I am on holiday next week...).

miri64, 2019-03-28:
As #10816 was merged now, I think we can close this.

closed 2019-03-28 by miri64