RIOT-OS/RIOT #11149: xtimer: hang on xtimer_spin_until (corner case)

vac, 2019-03-10:
I've been analyzing xtimer code and stoped on `xtimer_spin_until` function:

```c
static inline void xtimer_spin_until(uint32_t target)
{
#if XTIMER_MASK
    target = _xtimer_lltimer_mask(target);
#endif
    while (_xtimer_lltimer_now() > target) {}
    while (_xtimer_lltimer_now() < target) {}
}
```

`_xtimer_lltimer_now` will return uint32_t values within range: 0..UINT32_MAX. `xtimer_spin_until` will hang when the `target` value will be one of the following:
* 0,
* UINT32_MAX (4294967295U).

For target=0 it will hang on the first while loop.
For target=UINT32_MAX it will hang on the second while loop.

Please correct me if I'm wrong.

vac, 2019-03-10:
It looks like It is not as bad.
As long as _xtimer_lltimer_now() will manage to report the same value twice in a row loops will not hang in my cases.
I think I was too tired to analyze this code;)
As it is not such a big deal it is ok if you close it.
Sorry for disturbing;

vac, 2019-03-12:
I just found out that when I'm using 32768 Hz for my xtimer, the `_xtimer_lltimer_now()` function takes over 120us on samd21  - which is about 4 cycles of xtimer. So when the `target` is close to the uint32 limits then there a risk that target value will be omitted is much higher and from time to time device will spin for much longer than it should.

We may minimize (but not eliminate) the risk of this issue by changing `timer_read()` implementation so it won't take so long (on samd21 - most of the time is taken by synchro wait loop there).

I think that the right solution is to change these two while loops with something more robust.

PeterKietzmann, 2019-03-13:
@MichelRottleuthner could you check if this issue is a duplicate of one of the already reported xtimer issues?

vac, 2019-03-29:
Just found out that it is not possible ready every value of the async timer in sam21d.
If we would read the count register of TC, even in continuous mode, it will result in same value for 6 async (slow) clocks 😢 

If we agree that such platforms exist it makes handling timer functions much more complicated (for example mentioned `xtimer_spin_until` function).

I've tested it with 48MHz DFPLL main clock and 32768Hz xtimer clock.
With the 32768Hz clock, the theoretical resolution should be ~30us but on samd21, when polling count value, the resolution is ~180us... So RIOT xtimer solutions such us `XTIMER_OVERHEAD` handling need to be verified and improved.

pokgak, 2020-07-02:
Since `xtimer_spin_until()` is removed in #9530, should this issue be closed @miri64, @MichelRottleuthner?

miri64, 2020-07-02:
Yes, if the function described by the issue does not exist anymore we should indeed close it.

closed 2020-07-02 by miri64