RIOT-OS/RIOT #11151: drivers/l3g42000d: scale has wrong data type

TorbenPetersen, 2019-03-11:
#### Description

Introduced by 2ace1a2dfe scale changed from int32_t to int, which breaks conversion of data form the sensor. PR will follow soon.

PeterKietzmann, 2019-03-13:
@TorbenPetersen thanks for reporting. Please always use the template when opening an issue and give some more description. Do you know why the data type was changed? What exactly breaks? How do we reproduce this bug? Which platform, which application...

PeterKietzmann, 2019-03-13:
@dylad, @aabadie I assigned you as you were the authors of the @l3g42000d or [2ace1a2](https://github.com/RIOT-OS/RIOT/commit/2ace1a2dfeed7ac59e74753b9bed9ff619f53596)

TorbenPetersen, 2019-03-14:
> @TorbenPetersen thanks for reporting. Please always use the template when opening an issue and give some more description. Do you know why the data type was changed? What exactly breaks? How do we reproduce this bug? Which platform, which application...
1. I do not know why the data type was changed, according to the commit (2ace1a2) they reworked the driver API and the param definitions
2. All data, that is read from the sensor, is converted to zero during the conversion. That is the issue this bug causes
3. I used the driver test (tests/driver_l3g4200d). And as stated in my PR (#11152) I'm using an Atmega1284p

PR #11152 (2019-04-05): l3g4200d: Fixes conversion bug
There is a bug in the l3g4200d driver, which converted all the values my MCU (atmega1284p) received from the sensor to 0. 

### Contribution description

This PR fixes a conversion bug in the l3g4200d driver.

### Issues/PRs references

Fixes #11151.
closed 2019-04-05 by dylad