RIOT-OS/RIOT #11155: boards: boards/common directory as separate variable

bergzand, 2019-03-11:
#### Description

TLDR: I propose to add a `$(RIOTBOARD_COMMON) ?= $(RIOTBOARD)/common`

I rather often have the case where I use a board that is a simple variation of a supported board. For example where a module was added to the board or where peripheral pinouts are modified compared to the original. My workflow here consists of creating an out of tree directory containing all board files and pointing `$(RIOTBOARD)` to the parent directory.

The issue here is that when I want to use this board for a longer time out of tree, I have to flatten the full makefile includes from `$(RIOTBOARD)/$(BOARD)` and from `$(RIOTBOARD)/common`. 
my current example for this is a board based on the nrf52840 chip. Most of the makefiles in `$(RIOTBOARD)/common` still apply (`boards/common/nrf52/Makefile.include` as example), but I can't simply include those files into my board because they reference `$(RIOTBOARD)/common` which doesn't resolve as I've modified $(RIOTBOARD).

Having a separate value for the `common` directory inside `boards` would solve my issue here as I can override it to point to the original `boards/common` directory.

I don't mind writing the changes required for this myself as it should be an additional variable to the makefiles and a simple regexp replace. But before I do the work I'd like to hear some opinions on this and whether I'm missing something obvious in my workflow.

cladmi, 2019-03-11:
i was thinking about something similar, was just waiting for other changes to get in to handle it.

I would prefer to do it with a new `BOARDSDIR ?= $(RIOTBOARD)` variable, which should also then have a `$(BOARD)` directory in it, instead of just putting `common` in a variable.

Currently a `board` directory cannot be re-used to be extended and it is something I would like to change.
Like you could define your "this_board_with_a_shield_on_top" that could re-use `this_board`.

I was 

Now that https://github.com/RIOT-OS/RIOT/pull/9646 is merged, it will be easier to handle

cladmi, 2019-03-11:
TLDR; I would prefer `BOARDSDIR ?= $(RIOTBOARD)` but I agree, and it is in my todo list.

i was thinking about something similar, with the `boards` directory.

I would prefer to do it with a new `BOARDSDIR ?= $(RIOTBOARD)` variable, which should also then have a `$(BOARD)` directory in it, instead of just making  `common` a variable.
RIOTBOARD would then always be `RIOTBASE/boards`.

* I think that normal `board` directories can also be re-used and extended, not only the `common` ones. You could then define your board variation (with a plugged sensor) depending on a normal board.
    * it is currently not possible due to hard coded values and all implementing the `board` module, but possible to fix. It is in my/our todo list somehow :) and depends on this change I think.
* I think about defining `BOARDSDIR` instead of `BOARDDIR` as it would be also possible on your external directory to have multiple directories and common. And having only one external directory to mount in docker makes it easier to handle.
* It would also apply to the current `cpu` directory that does not have a `common` directory inside, and would be a good thing.

I was just waiting for the `docker` mount handling changes https://github.com/RIOT-OS/RIOT/pull/9646 to start looking at it as it requires mounting an extra directory  (thanks @MrKevinWeiss for merging).

cladmi, 2019-03-11:
If you have examples of an external board, I would gladly use it for testing :)

bergzand, 2019-03-11:
> If you have examples of an external board, I would gladly use it for testing :)

https://github.com/bergzand/boards_dmn

cladmi, 2019-03-11:
Would having this `BOARDSDIR` work for your use-cases or do you see any issues ?

I remember a dependency I would like to handle before changing this. All the pattern of `VAR := $(apspath $(VAR)`,  `override` and `MAKEOVERRIDES` of `__DIRECTORY_VARIABLES`.

I would prefer to require the user to provide an absolute path himself and trigger an error if it is invalid instead of trying to force overwrite the given configuration.

Do you think it would be appropriate to ask `BOARDSDIR` to be given as absolute ?
I want to ask now, as changing it later would change this API.

bergzand, 2019-03-11:
Am I correct in my assumption that this `BOARDSDIR` variable is a list of directories to use for looking up boards and related files? If that is the case this would solve my issue I think.

cladmi, 2019-03-11:
No, currently only the directory where the boards are implemented for this compilation. So that you can point it outside and still use `RIOT/boards` through `RIOTBOARD`.

So BOARDSDIR and not BOARDSDIRS. In the same way as `RIOTBOARD` currently points to `boards`, but you could make it point outside without copying the entire directory as you could still use `RIOTBOARD` inside `RIOT` and you could provide the main `board` implementation inside. Overwriting `boards/common` would not work with this.

Doing a search path is more complicated with the current constraints and would be hard to migrate in the current state. There are other dependencies to handle before at east.

When I think about the variable, I think also of adding a `CPUSDIR` to replace `RIOTCPU`. As currently it cannot be overwritten either as even `makefiles/arch` uses `RIOTCPU`, so `makefies/arch` is not more generic than the `cpu` directory.

jcarrano, 2019-03-13:
> Am I correct in my assumption that this BOARDSDIR variable is a list of directories to use for looking up boards and related files? If that is the case this would solve my issue I think.

I think that would be the best solution, like a kind of "PATH" for boards. It would also make "common" not special, just a convention.

We could set BOARDDIRS to a list of directories and then use some wildcard magic to include the board we want:

```make
# just set it to some random list, cpu is here just to add a bit of diversity
BOARDSDIRS=boards boards/common cpu

get_board=$(lastword $(wildcard $(addsuffix /$(1),$(BOARDSDIRS))))

$(info $(call get_board,esp32))
```

Same thing with cpus.

bergzand, 2019-03-13:
@cladmi I think your solution would solve my issue yes. I think having a `PATH`-like variable would be nice to have, but 95% of the time it is probably sufficient to have a solution as you proposed.

cladmi, 2019-03-13:
@jcarrano The dependency list for adding a search path is big, even more for boards and cpus, if you want to address it, please do.
But there are many things in the way before it can work, so would not want to wait for it for just this.

cladmi, 2019-09-16:
Different solution done here: https://github.com/RIOT-OS/RIOT/pull/12183

fjmolinas, 2020-01-15:
@bergzand I think this can be closed right?

miri64, 2020-07-04:
I assume yes. @bergzand please reopen if you disagree.

closed 2020-07-04 by miri64