RIOT-OS/RIOT #11171: Add further description of CoAP resources

leandrolanzieri, 2019-03-13:
#### Current situation
Right now our CoAP resources are described in [`coap_resource_t`](http://doc.riot-os.org/structcoap__resource__t.html) structures which include the resource path, the methods it accepts, a pointer to a handler and a context. That is all the information gcoap or nanocoap have about that resource.

With that limited information only a basic payload can be send when registering to a resource directory or when a GET request to `/.well-known/core` is replied. That is, only the paths of the registered resources expressed in [link format](https://tools.ietf.org/html/rfc6690):

```C
/* registration taken from cord_ep example application */
</node/info>,</sense/hum>,</sense/temp>
```
Those resources have no attributes, not metadata and no semantics that might indicate, for instance, that `/sense/temp` is actually a temperature sensor. Something like:
```
</sense/temp>;rt="http://sweet.jpl.nasa.gov/2.0/phys.owl#Temperature";
if="http://www.example.org/myapp.wadl#sensor"
```
This has consequences like:
- A CoAP client that tries to learn the server API has not enough information to figure out how to use it without out-of-band information (e.g. API documentation).
- A CoAP client that tries to filter resources results from a resource directory has not information to do so, neither has the directory if it tries to accomplish a lookup filtering.

#### Proposal
This issue aims to figure out if we should add this information to CoAP resources, and if yes how we should implement that.

**Some ideas**:
1- Add fields to the `coap_resource_t` structure, that add information about the resource (e.g. `rt`, `sz`, `if`, etc.). This option does not seem escalable and it is really attached to link format.

2- Add a sort of description callback to the `coap_resource_t` with the form of `ssize_t get_resource_desc(uint8_t *buf, size_t maxlen, uint8_t format)` that would be called if defined to describe the resource in detail in a given format. This would entail more effort from the point of view of the resource implementation, but only if a detailed description is wanted. It would also mean more control over the description.

3- Add some description that is resolved in compile time (maybe just a string) that would override the basic `</path/to/resource>` one. This option is not as flexible as (2) but is cheaper. We should figure out how different formats could be handled in this case.

**Things we should keep in mind**:
- Right now only link format is used for the description of the resources (both on `/.well-known/core` and registration to a resource directory) but [that might change in the future](https://tools.ietf.org/html/draft-hartke-t2trg-coral-reef-01).
- Should this descriptions be resolved in compile time or we would like to change them in runtime (e.g. a resource or an operation is not available under certain circumstances)?
- Should the resource handle the description or nanocoap/gcoap modules?
- ...?

### Useful links
- [RFC 6690 - CoRE Link Format](https://tools.ietf.org/html/rfc6690)
- [CoRE Resource Directory draft](https://tools.ietf.org/html/draft-ietf-core-resource-directory-20)

kb2ma, 2019-03-31:
To determine the best way to further describe CoAP resources, we need to identify the main use cases for this description. In general I am wary of mandating inclusion of *any* additional state for metadata for a constrained node. I think the simple scenario mostly pushes resource state off to a resource directory or other store on some more capable machine. I agree with the resource directory draft that a separate commissioning tool is likely to push this data to the RD.

Simple, static case
-------------------
Besides the case of external management, a simple node may know some static state about its resources. In this case the simple state includes basic insights into the resource, like resource type and content type. 

I would use the simplest approach that could work in this situation. I suggest addition of a new attribute to `coap_resource_t` that includes the parameter text already encoded in link format, like you describe in Idea 3:

```
typedef struct {
    const char *path;               /**< URI path of resource               */
    unsigned methods;               /**< OR'ed methods this resource allows */
    coap_handler_t handler;         /**< ptr to resource handler            */
    void *context;                  /**< ptr to user defined context data   */
#if NANOCOAP_CLIF_STATIC
    const char *clif_params;        /**< link format attributes             */
#endif
} coap_resource_t;
```

which looks like this example when used:

```
static const coap_resource_t _resources[] = {
    { "/cli/stats", COAP_GET | COAP_PUT, _stats_handler, NULL
#if NANOCOAP_CLIF_STATIC
      , .clif_param = "rt=counter;ct=40"
#endif
    }, 
...
```

I would make these parameters compile-time optional to avoid burdening the simplest scenarios that don't need them at all. Use of `const` also means the attribute string is stored in read-only memory with the rest of the resource. In this scenario a parameter like 'sz', that might change, could just use a value guaranteed to be large enough.

This approach makes it simple to write parameters to a resource directory in this scenario. Simply append the literal from the resource array as the parameters.

Capable, dynamic case
---------------------
A more capable scenario is use of a framework like LwM2M, where state is maintained internally, but also may be pushed to an external data manager. In this case, the state manager that will be able to interpret resource state in a generic way to write the state dynamically in link format.
 
In this situation, I don't think changes are needed to the `coap_resource_t`. The entity responsible for writing the links will know how to handle resources in a generic way. So maybe any conditional code for this scenario would reside in `gcoap_get_resource_list()` or similar. Maybe there would be a handler like in Idea 2 in the issue description. Notice the function below adds a `coap_resource_t` argument.

```
ssize_t get_resource_desc(const coap_resource_t *resource, uint8_t *buf, size_t maxlen, uint8_t format)
```

We likely would want a compile time constant to identify this scenario, something like, `NANOCOAP_CLIF_DYNAMIC`.

leandrolanzieri, 2019-04-01:
> In this situation, I don't think changes are needed to the `coap_resource_t`. The entity responsible for writing the links will know how to handle resources in a generic way.

So in this case you are suggesting that there is only one implementation of `get_resource_desc`, is that right?

kb2ma, 2019-04-02:
Good question about the implementation of `get_resource_desc()`. My mindset here is how a framework like an LwM2M implementation would handle this. I don't think there is a single implementation. Instead it seems reasonable to put the callback into the `gcoap_listener_t` struct, which manages the collection of resources. I would expect the framework to somehow check its internal state to provide the description.

leandrolanzieri, 2019-04-02:
Ah ok, I see, the callback is defined in the gcoap listener. Now I understand.

Which would be the main advantage of doing this in the listener and not in the `coap_resource_t` structure? I would imagine that a LWM2M implementation would only register a resource in `/lwm2m` for example with the `COAP_MATCH_SUBTREE` option set, and handle everything internally.

leandrolanzieri, 2019-04-03:
(updated the description to include @kaspar030 proposal)

leandrolanzieri, 2019-07-30:
Resolved for Gcoap by #11765.

closed 2019-07-30 by leandrolanzieri