RIOT-OS/RIOT #11207: CI: do not check labels

jcarrano, 2019-03-19:
### Description

Currently, the CI tests will fail if the PR has the "CI : needs squashing" label. After removing the label a rebuild must be triggered. If one does not want to waste CI resources building the same thing again, then the only way to do it now is to remove the label _before_ squashing. This has some problems:

- Non-members cannot touch the labels.
- It is extremely easy to forget.
- It is not really correct to remove the label before squashing.

Using the CI only when necessary would mean PR authors spend less time waiting in the queue. 

### Proposal

Could we have the CI tests not check the flags, and maybe another (way faster) test only for flags?

kaspar030, 2019-09-20:
still relevant

miri64, 2020-07-04:
> still relevant

Question is: how to solve this? Murdock was build around checking labels.

aabadie, 2021-05-20:
This is now addressed by the "check-pr" github: it's faster and doesn't require a Murdock rebuild. This can be closed.

closed 2021-05-20 by aabadie