RIOT-OS/RIOT #11208: Switch recommended Windows setup to msys2?

jcarrano, 2019-03-19:
### Description

I recently learned about [msys2](http://www.msys2.org/) and the closely related [mingw-w64](https://mingw-w64.org/doku.php/start). Both are a "revival" of msys and mingw. I tried the briefly in a VM:

- They have more updated packages. In particular, a decently recent version of GNU make (4.2)
- Msys2 uses pacman to install packages.
- Python 2 and 3 get installed correctly, and python libraries are also supported by the package manager.
  - I just tried it and the command `python3` works. This means that  #10768 would not have happened with msys2.
- It is actively mantained.

### Question

Who thinks it is a good idea to change the recommended Windows setup to Msys2?

@bluephoton, @vac, do you have experience with msys2?

vac, 2019-03-20:
Well, I'm currently happy with standard windows command line + MSYS make (only). 
MSYS make is needed to understand Linux style paths (like /c/programs/riot).
RIOT makefiles are not ready for Windows-style paths.
I didn't like using full MSYS as, for example, it used different Python than on my system - so if I wanted to do some fancy thing like automated tests, device programming, reports connected with RIOT compilation I needed to maintain another Python installation.

I totally understand sticking with python3, but I also found that it is not hard to make RIOT Python files compatible with python2 - most of them are tools that don't use any bleeding edge python3 feature.

Note that I didn't use native targets - just ARM.

I totally ok with MSYS2 but you/we may consider using Windows bash instead - not sure about the current status but it was easy to install kind of Ubuntu for Windows in MS Store a few months ago.
It looks like I can now just run bash in Windows that runs seamlessly as standard command line or power shell and it understands Linux style paths (/mnt/c/) and bash scripts, ... and it has apt:)

For me, the worst thing about the RIOT building system is a building system...
I know that I should create a separate issue about it but maybe it's just because of Windows so then it's fine, my fault:)
Incremental compilation on Windows takes a very long time (minutes) comparing to similar software made in Atmel studio for example (few seconds). I'm also not sure if RIOT makefiles declaring make dependencies right...: When I change header file in RIOT I always making sure to clean everything before compilation - otherwise, changes won't be included in a build. Clean RIOT build of some Atmel SAM project takes 3-4 minutes. So it is very time-consuming to tune RIOT headers. 
For me, it's a much bigger problem than things like #10768. If something is not working on Windows than we are always able to compile RIOT on Vagrant machine for that. It's working out of a box.

Sorry if I sound pessimistic... I'm really glad that "RIOT is happening", I like it, thanks for that:)

bluephoton, 2019-03-20:
I have very little experience in both and don't mind any one really! I totally agree with using WSL as @vac suggested. I started doing this initially, however I stopped as at the time interop between the two file systems was not supported. Coming windows 10 update however seems to have this taken care of.

see [Linux Files inside of File Explorer (Build 18342)
](https://docs.microsoft.com/en-us/windows-insider/at-home/whats-new-wip-at-home-19h1#linux-files-inside-of-file-explorer-build-18342)

jcarrano, 2019-03-20:
@vac 

> Sorry if I sound pessimistic...

Not at all. Honest feedback is _greatly_ appreciated.

> I'm also not sure if RIOT makefiles declaring make dependencies right

It's not, sadly. We are working on it.

> I totally ok with MSYS2 but you/we may consider using Windows bash instead

In addition to the build system problems (that's _our_ problem) I think there might be an issue with how MSYS/MSYS2 implements the unix compatibility. I also feel some things are slower than they should.

@bluephoton 
> see Linux Files inside of File Explorer (Build 18342)

mmm.... that would seem to take care of everything. I'll try the "ubuntu for windows" thing.

I'm closing this issue as I believe the question to be answered with "no". The next question will be "Should we switch the recommended setup to Ubuntu for windows?". I'll try to make it work before asking.

closed 2019-03-20 by jcarrano