RIOT-OS/RIOT #11212: POSIX TCP API: bad file descriptor

pipex, 2019-03-19:
#### Description

I've been testing the POSIX TCP/IP implementation in RIOT and I've come up with a few problems for the same code depending on the platform.

For testing I've created a simple [echo server/client](https://github.com/niclabs/two/tree/echo-v2) combining the [posix sockets example](https://github.com/RIOT-OS/RIOT/tree/2019.01-branch/examples/posix_sockets) and the [LWIP TCP example](https://github.com/RIOT-OS/RIOT/tree/2019.01-branch/tests/lwip). I tested the code on linux and it runs without issue (see the branch [echo-v2-linux](https://github.com/niclabs/two/tree/echo-v2-linux)).

#### Steps to reproduce the issue

Testing the code on different platforms I get the following

* On native
```
$  dist/tools/tapsetup/tapsetup -c 2
$ make all 

# For tap0 
$ PORT=tap0 make term
> ifconfig 
ET_00:  inet6 fe80::d864:daff:fed3:909
> echo server start 1234
Success: started TCP server on port 1234

# For tap1
$ PORT=tap0 make term
> echo send fe80::d864:daff:fed3:909 1234 hello
Success: connected to TCP server on port 1234
client send: Bad file descriptor

# If I enter the wrong port
> echo send fe80::d864:daff:fed3:909 12345 hello
Success: connected to TCP server on port 12345
client send: Bad file descriptor
```

* On samr21-xpro 
```
# For both boards
$ BOARD=samr21-xpro make all flash

# On board 1
$ BOARD=samr21-xpro PORT=<tty1> make term
> echo server start
INFO # Success: started TCP server on port 1234

# On board2
> echo send <addr> 1234 hello
INFO # Error in connect: Invalid argument
INFO # errno is unknown error: 22

# Same error with wrong port
```

* iotlab-m3 I get the same issue that with samr21-xpro

I figured that the error 22 was the result of a an uninitialized netif on the call to [_sockaddr_to_ep](https://github.com/RIOT-OS/RIOT/blob/9876e93b6b601411e0bf40e68e109f67100a284c/sys/posix/sockets/posix_sockets.c#L542) inside `_bind_connect` leading to a NULL return in [_netif_to_bind_addr](https://github.com/RIOT-OS/RIOT/blob/9876e93b6b601411e0bf40e68e109f67100a284c/pkg/lwip/contrib/sock/lwip_sock.c#L98)

However if y change the following code 
```
diff --git a/sys/posix/sockets/posix_sockets.c b/sys/posix/sockets/posix_sockets.c
index ae7195b32..8c7f16708 100644
--- a/sys/posix/sockets/posix_sockets.c
+++ b/sys/posix/sockets/posix_sockets.c
@@ -222,6 +222,7 @@ static int _sockaddr_to_ep(const struct sockaddr *address, socklen_t address_len
             }
             struct sockaddr_in *in_addr = (struct sockaddr_in *)address;
             out->family = AF_INET;
+            out->netif = SOCK_ADDR_ANY_NETIF;
             out->addr.ipv4_u32 = in_addr->sin_addr.s_addr;
             out->port = ntohs(in_addr->sin_port);
             break;
@@ -233,6 +234,7 @@ static int _sockaddr_to_ep(const struct sockaddr *address, socklen_t address_len
             }
             struct sockaddr_in6 *in6_addr = (struct sockaddr_in6 *)address;
             out->family = AF_INET6;
+            out->netif = SOCK_ADDR_ANY_NETIF;
             memcpy(&out->addr.ipv6, &in6_addr->sin6_addr, sizeof(out->addr.ipv6));
             out->port = ntohs(in6_addr->sin6_port);
             break;
```

I get the "Bad file number" message on iotlab-m3 and a hard fault on samr21-xpro. 

Am I missing something obvious? I still don't have such a good grasp of the lwip API to see what is wrong with the code.

#### Expected results

Similar to the result on [echo-v2-linux](https://github.com/niclabs/two/tree/echo-v2-linux)

```
# Server
$ gcc echo.c main.c 
$ ./a.out server start 12345
Success: started TCP server on port 12345
TCP Client [::1]: 42963 connected
Received TCP data from client [::1]:42963:
Received: "hello"
Received TCP data from client [::1]:42963:
Disconnected
TCP connection to [::1]:42963 reset, starting to accept again

# Client 
$ ./a.out send ::1 12345 hello
Success: connected to TCP server on port 12345
hello
```

#### Actual results

Bad file descriptor on iotlab-m3 and native, and  hardfault on samr21-xpro.

#### Versions

```
Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.14.3
                 Kernel: Darwin 18.2.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Apple LLVM version 10.0.0 (clang-1000.10.44.4)
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: Apple LLVM version 10.0.0 (clang-1000.10.44.4)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: missing
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.17.2 (Apple Git-113)
                   make: GNU Make 3.81
                openocd: Open On-Chip Debugger 0.10.0+dev-00730-ge2430759 (2019-03-06-17:21)
                 python: Python 3.7.2
                python2: missing
                python3: Python 3.7.2
             coccinelle: missing
```

miri64, 2019-03-20:
I'll have a look at this in the next few days.

miri64, 2019-05-23:
Sorry for the delay... :-( I'm not really understanding what the problem is here. You say on `samr21-xpro` and that you have the same results on `iotlab-m3` and `native` in "Actual results" but your initial description seems to hint at a different error?

Is the problem that the error numbers have different values on different platforms? Since those are provided by the libc we don't have any influence on that. I recommend to use the defines to identify them.

pipex, 2019-05-24:
Hi Martine. Yes, you're right that the description is unclear. 

What I meant is that I get the same results on samr21-xpro and iotlab-m3 BEFORE performing the following change (i think it's a bug)

```diff
diff --git a/sys/posix/sockets/posix_sockets.c b/sys/posix/sockets/posix_sockets.c
index ae7195b32..8c7f16708 100644
--- a/sys/posix/sockets/posix_sockets.c
+++ b/sys/posix/sockets/posix_sockets.c
@@ -222,6 +222,7 @@ static int _sockaddr_to_ep(const struct sockaddr *address, socklen_t address_len
             }
             struct sockaddr_in *in_addr = (struct sockaddr_in *)address;
             out->family = AF_INET;
+            out->netif = SOCK_ADDR_ANY_NETIF;
             out->addr.ipv4_u32 = in_addr->sin_addr.s_addr;
             out->port = ntohs(in_addr->sin_port);
             break;
@@ -233,6 +234,7 @@ static int _sockaddr_to_ep(const struct sockaddr *address, socklen_t address_len
             }
             struct sockaddr_in6 *in6_addr = (struct sockaddr_in6 *)address;
             out->family = AF_INET6;
+            out->netif = SOCK_ADDR_ANY_NETIF;
             memcpy(&out->addr.ipv6, &in6_addr->sin6_addr, sizeof(out->addr.ipv6));
             out->port = ntohs(in6_addr->sin6_port);
             break;
```

AFTER the change, I get a different error: "Bad file descriptor" on iotlab-m3 and native, and "hardfault" on samr21-xpro.

miri64, 2019-05-24:
22 [is EINVAL in newlib](https://www.sourceware.org/git/gitweb.cgi?p=newlib-cygwin.git;a=blob;f=newlib/libc/include/sys/errno.h;h=025b461d482a7ca74eca5e65aa0ff7b48207736e;hb=572687310059534b2da9428ca19df992509c8a5d#l54) btw

miri64, 2019-05-24:
> I'll try to find out, why the `samr21-xpro` is crashing.

The stack is to small for all the IP stacked on IP. If I patch your Makefile

```diff
diff --git a/src/Makefile b/src/Makefile
index e36691b..2784a3a 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -59,6 +59,7 @@ endif
 
 CFLAGS += -DSO_REUSE
 CFLAGS += -DLWIP_SO_RCVTIMEO
+CFLAGS += -DTHREAD_STACKSIZE_MAIN="(2*THREAD_STACKSIZE_DEFAULT)"
 #CFLAGS += -DLWIP_SOCK_TCP_ACCEPT_TIMEOUT=500
 #CFLAGS += -DPOSIX_SETSOCKOPT
```

I get a bad file descriptor for `samr21-xpro` as well.

miri64, 2019-05-24:
For the `EINVAL` bug there is now #11575 (I decided to use `memset()` instead to make it more future proof; I don't expect `struct _sock_tl_ep` to change, but who knows ;-)).

miri64, 2019-05-24:
So, now for the question, why `write()` returns `EBADF` in the first place.

As you may have noticed, that `socket()` uses the `vfs` module to create the file descriptor for the socket

https://github.com/RIOT-OS/RIOT/blob/f6ee0ace8ed2953f8ba6c13a4858076ffbbe8fc5/sys/posix/sockets/posix_sockets.c#L346

For write it basically just wraps around `send()`.

miri64, 2019-05-24:
Appearently the flags for this check

https://github.com/RIOT-OS/RIOT/blob/f6ee0ace8ed2953f8ba6c13a4858076ffbbe8fc5/sys/vfs/vfs.c#L325-L328

are not set accordingly.

pipex, 2019-05-24:
That's great, thanks for your help. I don't think I could've solved it myself.

I'll patch the Makefile and wait for the PRs to be merged.

Thanks again!

miri64, 2019-08-02:
https://github.com/RIOT-OS/RIOT/pull/11575 still needs to be merged for this issue to be considered fixed

miri64, 2020-07-04:
#11575 was merged.

PR #11576 (2019-08-02): posix_sockets: set read/write permissions for socket fd

### Contribution description
In https://github.com/RIOT-OS/RIOT/issues/11212 it was discovered, that the permissions for the file descriptors of POSIX sockets are not set so `write` and `read` can be used. This fixes that.

### Testing procedure
Flash and run the application provided in #11212 on 2 `samr21-xpro`s. Without this PR `echo send` returns "Bad file descriptor" as its `write()` call ends up here

https://github.com/RIOT-OS/RIOT/blob/f6ee0ace8ed2953f8ba6c13a4858076ffbbe8fc5/sys/vfs/vfs.c#L325-L328

### Issues/PRs references
Fixe [... cut]
closed 2020-07-04 by miri64