RIOT-OS/RIOT #11219: [Question/Problem] Cannot Compile ESP8266 with make

mmaxus35, 2019-03-20:
Hi all,
I have done steps in http://doc.riot-os.org/group__cpu__esp8266.html

I have added PATH's to .profile 

XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
PATH=/home/burak/bin:/home/burak/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:

/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin -> Here

MANDATORY_PATH=/usr/share/gconf/ubuntu.mandatory.path
COMPIZ_BIN_PATH=/usr/bin/

and which xtensa gave this output " /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
"
However i still get same error
"Compiler xtensa-lx106-elf-gcc is required but not found in PATH. Aborting."
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
PATH=/home/burak/bin:/home/burak/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:

/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin -> Here

MANDATORY_PATH=/usr/share/gconf/ubuntu.mandatory.path
COMPIZ_BIN_PATH=/usr/bin/

and which xtensa gave this output " /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc"

When I use make -C examples/gnrc_networking, it says;

mkdir: cannot create directory ‘/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/’: Permission denied
/home/burak/RIOT-master/examples/gnrc_networking/../../Makefile.include:686: recipe for target '/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/riotbuild.h' failed
make: *** [/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/riotbuild.h] Error 1
make: Leaving directory '/home/burak/RIOT-master/examples/gnrc_networking'

When I use "sudo make -C examples/gnrc_networking", it says;
"Compiler xtensa-lx106-elf-gcc is required but not found in PATH. Aborting."

* I have done chmod 777 , chown  etc. however it does not compile.

I could not understand the problem sadly. Thanks for your help.

mmaxus35, 2019-03-20:
I think that i manage to compile and tried simple example hello-world but output is like this ;

Building application "hello-world" for "esp8266-esp-12x" with MCU "esp8266".

"make" -C /home/burak/RIOT-master/boards/esp8266-esp-12x
"make" -C /home/burak/RIOT-master/boards/common/esp8266
"make" -C /home/burak/RIOT-master/core
"make" -C /home/burak/RIOT-master/cpu/esp8266
/home/burak/RIOT-master/cpu/esp8266/thread_arch.c: Assembler messages:
/home/burak/RIOT-master/cpu/esp8266/thread_arch.c:222: Error: invalid register 'interrupt' for 'wsr' instruction
/home/burak/RIOT-master/Makefile.base:83: recipe for target '/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/cpu/thread_arch.o' failed
make[2]: *** [/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/cpu/thread_arch.o] Error 1
/home/burak/RIOT-master/Makefile.base:20: recipe for target 'ALL--/home/burak/RIOT-master/cpu/esp8266' failed
make[1]: *** [ALL--/home/burak/RIOT-master/cpu/esp8266] Error 2
/home/burak/RIOT-master/examples/hello-world/../../Makefile.include:464: recipe for target '/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/application_hello-world.a' failed
make: *** [/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/application_hello-world.a] Error 2
make: Leaving directory '/home/burak/RIOT-master/examples/hello-world'

gschorcht, 2019-03-20:
@mmaxus35 Could you edit your message please and embed commands and outputs from terminal in \`xxxx\` when they are embedded in text or in
\`\`\`
yyyy
\`\`\`
when you have multiple line outputs to get output like the following.

... `xxxx` ...
```
yyyy
```
...
Your report is almost unreadable.

gschorcht, 2019-03-20:
> @gschorcht
> Hi all,
> I have done steps in http://doc.riot-os.org/group__cpu__esp8266.html
> 
> I have added PATH's to .profile
> 
> XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
> XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
> DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
> PATH=/home/burak/bin:/home/burak/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:
> 
> /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin -> Here
> 
> MANDATORY_PATH=/usr/share/gconf/ubuntu.mandatory.path
> COMPIZ_BIN_PATH=/usr/bin/
> 
> and which xtensa gave this output " /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
> "
> However i still get same error
> "Compiler xtensa-lx106-elf-gcc is required but not found in PATH. Aborting."
> XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
> XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
> DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
> PATH=/home/burak/bin:/home/burak/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:
> 
> /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin -> Here
> 
> MANDATORY_PATH=/usr/share/gconf/ubuntu.mandatory.path
> COMPIZ_BIN_PATH=/usr/bin/
> 
> and which xtensa gave this output " /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc"
> 
> When I use make -C examples/gnrc_networking, it says;
> 
> mkdir: cannot create directory ‘/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/’: Permission denied
> /home/burak/RIOT-master/examples/gnrc_networking/../../Makefile.include:686: recipe for target '/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/riotbuild.h' failed
> make: *** [/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/riotbuild.h] Error 1
> make: Leaving directory '/home/burak/RIOT-master/examples/gnrc_networking'
> 
> When I use "sudo make -C examples/gnrc_networking", it says;
> "Compiler xtensa-lx106-elf-gcc is required but not found in PATH. Aborting."
> 
> * I have done chmod 777 , chown  etc. however it does not compile.
> 
> I could not understand the problem sadly. Thanks for your help.

You copied everything before in a single unstructured text. So, lets first figure out what your remaining problem is. Obviously, you could solve your `PATH` problem. Otherwise, the compilation wouldn't start and you would not see the message

> ```mkdir: cannot create directory ‘/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/’: Permission denied```

Right?

mmaxus35, 2019-03-20:
Sorry for delay and sloppy messages. 

-  I solved  PATH problem, i can use

- [ ]  make BOARD=esp8266-esp-12x -C examples/hello-world

 to compile hello-world example , during complilation it fails and output as follows;

```
`Building application "hello-world" for "esp8266-esp-12x" with MCU "esp8266".

"make" -C /home/burak/RIOT-master/boards/esp8266-esp-12x
"make" -C /home/burak/RIOT-master/boards/common/esp8266
"make" -C /home/burak/RIOT-master/core
"make" -C /home/burak/RIOT-master/cpu/esp8266

/home/burak/RIOT-master/cpu/esp8266/thread_arch.c: Assembler messages:
**/home/burak/RIOT-master/cpu/esp8266/thread_arch.c:222**: Error: invalid register 'interrupt' for 'wsr' instruction**
/home/burak/RIOT-master/Makefile.base:83: recipe for target '/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/cpu/thread_arch.o' failed**
**make[2]: *** [/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/cpu/thread_arch.o] Error 1
/home/burak/RIOT-master/Makefile.base:20: recipe for target 'ALL--/home/burak/RIOT-master/cpu/esp8266' failed
make[1]: *** [ALL--/home/burak/RIOT-master/cpu/esp8266] Error 2**
/home/burak/RIOT-master/examples/hello-world/../../Makefile.include:464: recipe for target '/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/application_hello-world.a' failed
make: *** [/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/application_hello-world.a] Error 2
make: Leaving directory '/home/burak/RIOT-master/examples/hello-world'`
```

gschorcht, 2019-03-20:
> ```
> mkdir: cannot create directory ‘/home/burak/RIOT-master/examples/gnrc_networking/bin/esp8266-esp-12x/riotbuild/’: Permission denied [...]

This error message is not related to the ESP8266 port. Might it be that you have cloned the repository as `sudo` by mistake? What is the `user:group` of directory `/home/burak/RIOT-master/examples/gnrc_networking`?
```
ls -l /home/burak/RIOT-master/examples/gnrc_networking
```

> When I use "sudo make -C examples/gnrc_networking", it says;
>
> ```Compiler xtensa-lx106-elf-gcc is required but not found in PATH. Aborting.```

That's very clear. Because of the `sudo` command, you become `root` while executing the command. Since you added the `PATH` only to your user's `.profile`, the `PATH` variable of `root` does not contain the path to the toolchain.

> * I have done chmod 777 , chown  etc. however it does not compile.

This is definitly the wrong way. Please check, whether you have cloned the repository as user `burak` and delete and reclone it again if not.

mmaxus35, 2019-03-20:
@gschorcht No, please do not consider PATH problem anymore. I can trigger make. 
New problem is below.

> 
> 
> @gschorcht Sorry for delay and sloppy messages.
> 
>     * I solved  PATH problem, i can use
> 
>     * [ ]   make BOARD=esp8266-esp-12x -C examples/hello-world
> 
> 
> to compile hello-world example , during complilation it fails and output as follows;
> 
> ```
> `Building application "hello-world" for "esp8266-esp-12x" with MCU "esp8266".
> 
> "make" -C /home/burak/RIOT-master/boards/esp8266-esp-12x
> "make" -C /home/burak/RIOT-master/boards/common/esp8266
> "make" -C /home/burak/RIOT-master/core
> "make" -C /home/burak/RIOT-master/cpu/esp8266
> 
> /home/burak/RIOT-master/cpu/esp8266/thread_arch.c: Assembler messages:
> **/home/burak/RIOT-master/cpu/esp8266/thread_arch.c:222**: Error: invalid register 'interrupt' for 'wsr' instruction**
> 
> /home/burak/RIOT-master/Makefile.base:83: recipe for target '/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/cpu/thread_arch.o' failed**
> 
> **make[2]: *** [/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/cpu/thread_arch.o] Error 1
> 
> /home/burak/RIOT-master/Makefile.base:20: recipe for target 'ALL--/home/burak/RIOT-master/cpu/esp8266' failed
> 
> make[1]: *** [ALL--/home/burak/RIOT-master/cpu/esp8266] Error 2**
> 
> /home/burak/RIOT-master/examples/hello-world/../../Makefile.include:464: recipe for target '/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/application_hello-world.a' failed
> 
> make: *** [/home/burak/RIOT-master/examples/hello-world/bin/esp8266-esp-12x/application_hello-world.a] Error 2
> make: Leaving directory '/home/burak/RIOT-master/examples/hello-world'`
> ```

gschorcht, 2019-03-20:
@mmaxus35 I'm really sorry but I can't reproduce your problem. That's what I have done to reproduce your problem:
```
cd /opt
sudo git clone https://github.com/gschorcht/RIOT-Xtensa-ESP8266-toolchain.git esp
export PATH=$PATH:/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin
cd ~/src
git clone https://github.com/RIOT-OS/RIOT
cd RIOT
make BOARD=esp8266-esp-12x -C examples/hello-world
```
And that's the output as expected:
```
make: Verzeichnis „/home/gs/src/RIOT/examples/hello-world“ wird betreten
ESP8266_NEWLIB_DIR should be defined as /path/to/newlib directory
ESP8266_NEWLIB_DIR is set by default to /opt/esp/newlib-xtensa
ESP8266_SDK_DIR should be defined as /path/to/sdk directory
ESP8266_SDK_DIR is set by default to /opt/esp/esp-open-sdk/sdk
Building application "hello-world" for "esp8266-esp-12x" with MCU "esp8266".

"make" -C /home/gs/src/RIOT/boards/esp8266-esp-12x
"make" -C /home/gs/src/RIOT/boards/common/esp8266
"make" -C /home/gs/src/RIOT/core
"make" -C /home/gs/src/RIOT/cpu/esp8266
"make" -C /home/gs/src/RIOT/cpu/esp8266/periph
"make" -C /home/gs/src/RIOT/cpu/esp8266/sdk
"make" -C /home/gs/src/RIOT/cpu/esp8266/vendor
"make" -C /home/gs/src/RIOT/cpu/esp8266/vendor/esp
"make" -C /home/gs/src/RIOT/cpu/esp8266/vendor/xtensa
"make" -C /home/gs/src/RIOT/cpu/esp_common
"make" -C /home/gs/src/RIOT/drivers
"make" -C /home/gs/src/RIOT/drivers/mtd
"make" -C /home/gs/src/RIOT/drivers/periph_common
"make" -C /home/gs/src/RIOT/sys
"make" -C /home/gs/src/RIOT/sys/auto_init
"make" -C /home/gs/src/RIOT/sys/luid
"make" -C /home/gs/src/RIOT/sys/ps
"make" -C /home/gs/src/RIOT/sys/random
"make" -C /home/gs/src/RIOT/sys/random/tinymt32
   text	   data	    bss	    dec	    hex	filename
  70947	   4300	   7184	  82431	  141ff	/home/gs/src/RIOT/examples/hello-world/bin/esp8266-esp-12x/hello-world.elf
make: Verzeichnis „/home/gs/src/RIOT/examples/hello-world“ wird verlassen
```
As you can see, I followed exactly the documentation.

Please send me the output of following command:
```
make BOARD=esp8266-esp-12x -C examples/hello-world QUIET=0 >& make.prn
```
This will produce `make.prn` in RIOT's root directory. The best way for uploading large file contents is to use [Gist](https://gist.github.com).

mmaxus35, 2019-03-20:
I followed your steps and the output changed. Here is the output of .prn file.

[https://gist.githubusercontent.com/mmaxus35/cb73320e73e522c84af030888538d3d4/raw/46e414134546b7aa4c5bb2aa484c49460cabe563/make.prn](url)

gschorcht, 2019-03-21:
@mmaxus35 I really don't know what the cause of your problem is, but the output looks really strange. It seems that your `/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc` is executed as it would be a shell script. Please send me the output of following commands:

```
which xtensa-lx106-elf-gcc
file `which xtensa-lx106-elf-gcc`
ls -l `which xtensa-lx106-elf-gcc`
uname -a
cat /etc/issue
```

mmaxus35, 2019-03-21:
@gschorcht I will give necessary outputs tomorrow. I left my notebook at University.
However, today i tried to install esp32. I followed the docs and i made installation in two ways,

- 1) I followed manuel installation for esp32 and when i tried to make it gave error outputs as same as esp8266 hello-world example , build target for = esp32-wroom -> /cpu/thread_arch.o' failed** ( this is the same error with esp8266.)

- Then i tried to use docker image installation in docs for esp32 and it built it successfully for esp32-wroom build target and hello-world example.

As result, i think that we should consider these similarities too. Again, i will give necessary outputs tomorrow at 14.00 (GMT+3) .

gschorcht, 2019-03-22:
@mmaxus35
> I will give necessary outputs tomorrow.

Please send me also the output of command
```
xtensa-lx106-elf-gcc -v
```
> i tried to use docker image installation

The toolchain installation in docker images follows exactly the steps described in section [Precompiled toolchain](https://riot-os.org/api/group__cpu__esp8266.html#esp8266_precompiled_toolchain).

mmaxus35, 2019-03-22:
@gschorcht 
1.a)Docker image way for esp32 has complied hello-world example.  b.)Manual installation lead to fail as same as esp8266.

2-) Outputs of the following commands for esp8266

`which xtensa-lx106-elf-gcc -> /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc`

file `which xtensa-lx106-elf-gcc` ->` /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/l, for GNU/Linux 2.6.32, BuildID[sha1]=0e79c6bf859b6040cef89d0880e4a8bdbddf56ca, stripped`

ls -l `which xtensa-lx106-elf-gcc ->` `-rwxr-xr-x 1 burak root 776336 Mar 19 17:06 /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
`
`uname -a -> Linux Burak-PC 4.10.0-37-generic #41~16.04.1-Ubuntu SMP Fri Oct 6 22:42:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

cat /etc/issue -> `Ubuntu 16.04.6 LTS \n \l`

`xtensa-lx106-elf-gcc -v` -> [https://gist.githubusercontent.com/mmaxus35/2de4eee4c94f60d277fdf6f603580691/raw/06383747e38ac3461bedfd868b403e1f47d39858/xtensa-lx106-elf-gcc%2520-v](url)

gschorcht, 2019-03-22:
@mmaxus35 Hm, everything looks fine. But, the following lines in your output [make.prn](https://gist.githubusercontent.com/mmaxus35/cb73320e73e522c84af030888538d3d4/raw/46e414134546b7aa4c5bb2aa484c49460cabe563/make.prn) gives me the hint that `/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc` might be called as shell script by mistake:
```
/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: 1: /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: ELF: not found
/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: 3: /opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Syntax error: "(" unexpected
```
The output in this lines are typical shell outputs. ELF binaries are starting with `ELF` signature, see
```
head -1 `which xtensa-lx106-elf-gcc`
```
Hexdump of `/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc` starts with. That's why you get the strange characters erround `ELF`.
```
00000000  7f 45 4c 46 02 01 01 00  00 00 00 00 00 00 00 00  |.ELF............|
```
At the moment, I have definitely no idea what the reason of your problem might be. It is probably not related to the toolchain since it is working in docker and I have never heart about problems from others. The only difference is that you are using Ubuntu 16.04 instead of Ubuntu 18.04. But this shouldn't be a problem. When I have time, I can check this with a virtual machine. BTW, docker was also using Ubuntu 16.04 until fall of last year and there were no problems too.

mmaxus35, 2019-03-22:
@gschorcht Thanks for your interest, i will be keep in contact in case of you need anything to solve this problem.

gschorcht, 2019-03-22:
@mmaxus35 I used a fresh Ubuntu 16.04.6 LTS standard desktop installation and just executed the following steps:
```
sudo apt-get install git
cd /opt
sudo git clone https://github.com/gschorcht/RIOT-Xtensa-ESP8266-toolchain.git esp
export PATH=$PATH:/opt/esp/esp-open-sdk/xtensa-lx106-elf/bin
cd ~
https://github.com/RIOT-OS/RIOT.git
cd RIOT
make BOARD=esp8266-esp-12x -C examples/hello-world
```
And ... compilation works as expected without any problems. So, everything is fine with the toolchain.

The last thing you could try is to execute the failed command by hand and see what happens:
```
xtensa-lx106-elf-gcc \
		-DRIOT_FILE_RELATIVE=\"examples/hello-world/main.c\" \
		-DRIOT_FILE_NOPATH=\"main.c\" \
		-Werror -Wno-unused-parameter -Wformat=0 -mlongcalls -mtext-section-literals -fdata-sections -fzero-initialized-in-bss -Os -fno-common -Wall -Wextra -Wmissing-include-dirs -include '/home/user/src/RIOT/examples/hello-world/bin/esp8266-esp-12x/riotbuild/riotbuild.h'  -I/home/user/src/RIOT/core/include -I/home/user/src/RIOT/drivers/include -I/home/user/src/RIOT/sys/include -I/home/user/src/RIOT/boards/esp8266-esp-12x/include -I/home/user/src/RIOT/cpu/esp8266/include -I/opt/esp/newlib-xtensa/xtensa-lx106-elf/include -I/home/user/src/RIOT/boards/common/esp8266/include -I/home/user/src/RIOT/cpu/esp8266 -I/home/user/src/RIOT/cpu/esp8266/vendor -I/home/user/src/RIOT/cpu/esp8266/vendor/espressif -I/home/user/src/RIOT/sys/libc/include  -MD -MP -c -o /home/user/src/RIOT/examples/hello-world/bin/esp8266-esp-12x/application_hello-world/main.o /home/user/src/RIOT/examples/hello-world/main.c
```

mmaxus35, 2019-03-22:
@gschorcht I downloaded clean Ubuntu 16.04. and everything worked fine! Thanks. I'll shift my study to new installation, also i will investigate to build by hand in previous installation.

PeterKietzmann, 2019-03-24:
So I consider this issue as solved

closed 2019-03-24 by PeterKietzmann

potier97, 2020-05-24:
I have the same problem, i am using a RPI 3, and i execute 
>  make BOARD=esp8266-esp-12x -C examples/hello-world

And show in red: 
Compiler xtensa-esp8266-elf-gcc is required but not found in PATH.  Aborting.

mmaxus35, 2020-05-24:
İ used docker you should use it too.

24 May 2020 Paz 06:28 tarihinde potier97 <notifications@github.com> şunu
yazdı: