RIOT-OS/RIOT #11312: sx127x with gnrc_networking example

Khayhen, 2019-03-29:
Hi, I'm making some custom boards using an arduino mkr-zero with different radio modules to work in 6LoWPAN network with a border router. I have successfully created a board with MRF24J40 radio module but I'm having trouble with the SX1278 module. To implement the board I made a copy of the arduino MKR-Zero file and added the following lines to the file include/boards.h
<pre>
#define SX127X_PARAM_SPI                    (SPI_DEV(0))
#define SX127X_PARAM_SPI_NSS                GPIO_PIN(0, 21)	//PA21 Arduino Pin 7 

#define SX127X_PARAM_RESET                  GPIO_PIN(0, 20)	//PA20 Arduino Pin 6 
#define SX127X_PARAM_DIO0                   GPIO_PIN(1, 11)	//PB11 Arduino Pin 5 
#define SX127X_PARAM_DIO1                   GPIO_PIN(1, 10)	//PB10 Arduino Pin 4 
#define SX127X_PARAM_DIO2                   GPIO_PIN(0, 11)	//PA11 Arduino Pin 3 
#define SX127X_PARAM_DIO3                   GPIO_PIN(0, 10)	//PA10 Arduino Pin 2 
</pre>
and these lines to the file Makefile.dep
<pre>
ifneq (,$(filter netdev_default,$(USEMODULE)))
  USEMODULE += sx1276
endif
</pre>
I tested the board by flashing with tests/driver_sx127x, and manage to successfully communicate with the radio module and do some communication. However, when I flash it with examples/gnrc_networking, the RIOT crashes and I get the following error.
<pre>Welcome to pyterm!
Type &apos;/exit&apos; to exit.
2019-03-30 01:32:08,910 - INFO # 0x743d
2019-03-30 01:32:08,911 - INFO # *** RIOT kernel panic:
2019-03-30 01:32:08,955 - INFO # FAILED ASSERTION.
2019-03-30 01:32:08,955 - INFO # 
2019-03-30 01:32:08,957 - INFO # 	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
2019-03-30 01:32:08,960 - INFO # 	  - | isr_stack            | -        - |   - |    512 (  152) | 0x20000000 | 0x200001b8
2019-03-30 01:32:08,963 - INFO # 	  1 | idle                 | pending  Q |  15 |    256 (  152) | 0x20000564 | 0x200005cc 
2019-03-30 01:32:08,966 - INFO # 	  2 | main                 | bl mutex _ |   7 |   1536 (  264) | 0x20000664 | 0x20000b64 
2019-03-30 01:32:08,969 - INFO # 	  3 | pktdump              | bl rx    _ |   6 |   1536 (  248) | 0x200032d4 | 0x200037dc 
2019-03-30 01:32:09,005 - INFO # 	  4 | ipv6                 | bl rx    _ |   4 |   1024 (  280) | 0x200011bc | 0x200014a4 
2019-03-30 01:32:09,007 - INFO # 	  5 | udp                  | bl rx    _ |   5 |   1024 (  280) | 0x20003f24 | 0x2000420c 
2019-03-30 01:32:09,008 - INFO # 	  6 | sx127x               | running  Q |   2 |   1024 (  668) | 0x20000dbc | 0x2000102c 
2019-03-30 01:32:09,010 - INFO # 	  7 | RPL                  | bl rx    _ |   5 |   1024 (  212) | 0x20003928 | 0x20003c54 
2019-03-30 01:32:09,011 - INFO # 	    | SUM                  |            |     |   7936 ( 2256)
2019-03-30 01:32:09,011 - INFO # 
2019-03-30 01:32:09,012 - INFO # *** halted.
2019-03-30 01:32:09,012 - INFO # 
</pre>

Is this module even possible to run a gnrc network stack? If I can get the basics working, will be good enough already as I just need to show a proof of concept for now. Any help will be much appreciated. Thank you.

jia200x, 2019-04-01:
hello @Khayhen 

Could you run the same example with the DEBUG_ASSERT_VERBOSE? Compile with:
```
CFLAGS=-DDEBUG_ASSERT_VERBOSE BOARD=<your_board> make all flash term
```

That will show the file and line where the failed assertion occurs.

MrKevinWeiss, 2019-04-01:
Currently GNRC is not ported for any LoRa devices (which is what I thought the SX1278 module is for).  It is on it's way.  You could try the `examples/lorawan`

Also you are talking about the sx1278 module but you are using `USEMODULE += sx1276`.

I am not a networking guy but maybe that could clear up some issues.

Khayhen, 2019-04-02:
@jia200x I ran the code with the verbose output and it tends to hang here.
<pre>INFO # sys/net/gnrc/netif/gnrc_netif.c:1250 =&gt; 0x2d35
</pre>

Khayhen, 2019-04-02:
@MrKevinWeiss Thats what I suspected also that lora modules has no support for GNRC yet. Correct me if I'm wrong, to use lorawan, I need to have a subscription to the network and it works independent from the 6lowpan stack? 
Regarding the sx1278, I am only able to source sx1278 locally and from what I see RIOT only supports sx1272 and sx1276. However, from looking at the [datasheet](https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf) sx1278 and sx1276 works exactly the same and its just a different variant of it.

jia200x, 2019-04-02:
> @MrKevinWeiss Thats what I suspected also that lora modules has no support for GNRC yet. Correct me if I'm wrong, to use lorawan, I need to have a subscription to the network and it works independent from the 6lowpan stack?
Regarding the sx1278, I am only able to source sx1278 locally and from what I see RIOT only supports sx1272 and sx1276. However, from looking at the datasheet sx1278 and sx1276 works exactly the same and its just a different variant of it.

GNRC doesn't support LoRa modules yet. There's an ongoing PR #11022 which adds a GNRC specific LoRaWAN implementation, but it's independent from 6lowpan stack. It won't be possible to have IPv6 on top of LoRaWAN because we still don't have [SCHC compression and fragmentation for LPWAN](https://tools.ietf.org/id/draft-ietf-lpwan-ipv6-static-context-hc-17.html)

jia200x, 2019-04-02:
> @jia200x I ran the code with the verbose output and it tends to hang here.

Are you using the latest master? The reason why that could fail is in case the device type is not listed in GNRC netif. This was added a couple of weeks ago for LoRa (#11136)

Does this solve your issue?

Khayhen, 2019-04-05:
> Are you using the latest master? The reason why that could fail is in case the device type is not listed in GNRC netif. This was added a couple of weeks ago for LoRa (#11136)

Yes you are right. It turns out my master version was just a few days older than that pull request. I manage to flash it and operate it now. However, I noticed that the modules don't have a unique hardware address and all of them have the same IPv6 address of fe80::. Is this an expected outcome?  Is it possible to give them a specific address?

jia200x, 2019-11-07:
@Khayhen are you still experiencing this issue?