RIOT-OS/RIOT #11320: [Quick Question] ESP8266 esp-wifi ssid and password parameters in .c file ?

mmaxus35, 2019-04-01:
#### Description
It is important to define parameters in main.c

USEMODULE=esp_wifi \
CFLAGS='-DESP_WIFI_SSID=\"MySSID\" -DESP_WIFI_PASS=\"MyPassphrase\"' \
make -C examples/gnrc_networking BOARD=...
### Useful links
http://doc.riot-os.org/group__cpu__esp8266.html

Sorry for the noob question. But, as i read the document, there are two ways to give credientials to esp8266 either in make or in makefile. However, in main.c i try to re-define these parameters however it does not re-writed and always tries to connect Riot-AP ssid. I inspected library, but could not manage to run healty results. What should i do?

Thanks for your help!

fedepell, 2019-04-03:
Try adding:
`#include "net/ethernet.h"`
before the include you mention? (as `ETHERNET_MAX_LEN` is defined there, and the .c file of esp_wifi_netdev will do so)

mmaxus35, 2019-04-04:
@fedepell  Thanks for your answer, i added ` #include "net/ethernet.h"` however, it gave another error. Actually, it is not important to add another library from net. 
`#define ESP_WIFI_SSID "TPBD9E_2.4GHz" 
#define ESP_WIFI_PASS "dguKuYPc123x" `

These two definitions are not recognized even if i have added in main.c at top.

then, in  `RIOT/cpu/esp8266/esp-wifi/esp_wifi_params.h` 

```
#ifndef ESP_WIFI_SSID
#define ESP_WIFI_SSID           "RIOT_AP"
#endif

#ifndef ESP_WIFI_PASS
#define ESP_WIFI_PASS           "ThisistheRIOTporttoESP"
#endif

```
are used when flash my esp8266. this means my definitions in main.c are not recognized.

gschorcht, 2019-04-04:
@mmaxus35 Documentation says:
> These configuration parameter definitions ... can be done either
> - in the `makefile of the project` or
> - at make command line`

There is no word that this could be done in `main.c` as well. You can't do that in your `main.c`. The reason is quite simple and more a basic C question: _Who needs the definitions and who includes what?_ 

`ESP_WIFI_SSID` and `ESP_WIFI_PASS` are not required in `main.c` but in `cpu/esp8266/esp_wifi/esp_wifi_netdev.c`. Since `cpu/esp8266/esp_wifi/esp_wifi_netdev.c` does not include `main.c` (of course), it cannot know your definitions you made in `main.c`. The definitions could only be overriden in a file that is includes by `cpu/esp8266/esp_wifi/esp_wifi_netdev.c`. Since `cpu/esp8266/esp-wifi/esp_wifi_params.h` is the only file that is included, the definitions are used from there.

Please defined according definitions either in your `makefile` or at command line as documented.

mmaxus35, 2019-04-04:
@gschorcht Thanks for your answer, your study on this has important role for my Msc thesis, and i really thank you for your attention. I work on model based code generation to ESP8266 based on RIOT OS. I thought that there might be a way to auto generate ssid and password parameters in main.c not to complex the code generation phase. In conclusion, thanks for your answer and i will generate  ` esp_wifi_params.h` then.  Thanks for your answer. However, in our university we keen on esp8266 , esp32 and RIOT OS. In further, we would like to work on both mesh topology, using IEEE 802.11 and IEEE 802.15.4 etc ( esp as border router and able to create mesh topology).

I will close this issue after your comment

gschorcht, 2019-04-04:
@mmaxus35 Model based code generation is a very interesting topic. I was working on this topic some years ago.

Maybe, the easiest way to override the definitions is not to overwrite `cpu/esp8266/esp-wifi/esp_wifi_params.h` but to generate a file `esp_wifi_params.h` in your application directory which looks like
```
#define ESP_WIFI_SSID "TPBD9E_2.4GHz"
#define ESP_WIFI_PASS "dguKuYPc123x"

#include_next "esp_wifi/esp_wifi_params.h"
```
You would aslo have to generate following a makefile with first line
```
INCLUDES += -I$(APPDIR)
```
to ensure that `esp_wifi_params.h` in your application directory is found first. Please refer section [Application-Specific Board Configuration](https://doc.riot-os.org/group__cpu__esp8266.html#esp8266_application_specific_board_configuration) in documentation.

mmaxus35, 2019-04-05:
@gschorcht Thanks for your answer. I will make these implementations asap. You are big help.

closed 2019-04-05 by mmaxus35

mugekuskon, 2022-11-27:
@mmaxus35 Could you please provide your code if possible ? I tried to flash my arduino code with RIOT in my eps12 based board but various types of errors have arised. I was wondering if the same problem could happen with your code too. This would be really helpful if possible thank you in advance.