RIOT-OS/RIOT #11329: Proposal: use SPDX headers.

jcarrano, 2019-04-02:
### Description

Current practice in RIOT is to place a (somewhat bulky) standard header in files describing the license. These are parsed by a shell script (https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/licenses/check.sh)

Long license headers bloat source files and complicate parsing (think of all that that poor script has to do, try running it on few hundred files and you will see).

As it usually happens, other people already ran into the same issue and developed SPDX license identifiers: https://spdx.org/ids-how .

### Proposal

What if we start using SPDX for new files?

maribu, 2021-09-13:
IMO we should pick this up again. [SPDX has recently been recognised as international standard](https://lwn.net/Articles/868882/), maybe this good occasion to resume this.

miri64, 2021-09-15:
Whoever is implementing this, should also remember to update the [license section in the README](https://github.com/RIOT-OS/RIOT#license).

crasbe, 2025-07-19:
Closed in favor of #21515.

closed 2025-07-19 by crasbe