RIOT-OS/RIOT #11334: Non-blocking ADC

Alexander-M1, 2019-04-03:
Hi,

I'm trying out ADC in RIOT and I see the sampling function is a blocking function.
I would like to know, if there any solutions available for a non-blocking ADC peripheral interface, or if anyone has written some sort of wrapper around this problem?

Thanks,
Alex

chrysn, 2019-05-14:
This would scratch an itch of mine too. There's a lot of variations to this (from separating the "start the ADC conversion" from the "block until there is a result" code, over providing a nonblocking "is there a result" fetcher over to "and {wake up this thread, send it this message, set that flag, run this callback}" code that'd need to run from the ADC interrupt handler).

I don't know the other asynchronous parts of RIOT well enough to suggest a solution, but would be happy to review any proposal.

PeterKietzmann, 2019-05-21:
> or if anyone has written some sort of wrapper around this problem?

Not that I know. AFAIK blocking the `adc_sample()` function was a design decision back then. However, I agree that some asynchronous features, wakeup on threshold for example, would be helpful.  @haukepetersen what's your opinion?

fjmolinas, 2020-01-22:
Re-open, this could be a useful feature.

wosym, 2020-03-05:
I agree that this is something very much needed. It took me a while to figure out why my main thread seemed to freeze, but apparently if the AVR fails to sample the ADC, there is no other option than to reset.

Adding a time-out options seems quite crucial to me. I added this quick&dirty solution:
```
    xtimer_ticks64_t start = xtimer_now64();
    /* Wait until the conversion is complete */
    while (ADCSRA & (1 << ADSC)) {
        if(xtimer_diff64(xtimer_now64(), start).ticks64 > MAX_SAMPLE_TIME) {
            return -1;
        }
    }
```