RIOT-OS/RIOT #11335: Arduino Zero ADC

Alexander-M1, 2019-04-03:
Hi,

I'm trying out ADC in RIOT on an Arduino Zero board.
The board `arduino-zero` hasn't got [ADC support in RIOT](http://doc.riot-os.org/group__boards__arduino-zero.html), but I managed to get it going by defining `board=samd21-xpro`. Since the SAMD21 chips are not exactly equal and this is not a good long term solution, I would like to know, how others using the Arduino Zero board are tackling the problem?
I'm also curious as to how come the Xpro board has ADC support and the Zero board hasn't got it - are the chips really that different, or is it just a matter of changing a few registers?

Thanks,
Alex

kaspar030, 2019-04-03:
Welcome @Alexander-M1!

General questions like this are supposed to be asked on our mailing lists (e.g., users@riot-os.org or devel@riot-os.org).

> I'm also curious as to how come the Xpro board has ADC support and the Zero board hasn't got it - are the chips really that different, or is it just a matter of changing a few registers?

The actual periph implementation should work as is. The ADC configuration is definitely missing.
If you could get it working by just compiling for samr21-xpro, try copying the corresponding parts and add periph_adc to the Makefile.features.

Try this patch (it compiles but otherwise completely untested):

https://gist.github.com/98402b82ba7b871bc23cd4f5736b598f

Alexander-M1, 2019-04-03:
@kaspar030 thanks for the warm welcome.

In `adc_channels[]` how are macros `ADC_INPUTCTRL_MUXPOS_PINxx` defined?
Looking at the patch you sent its pretty logical, however looking at samd21-xpro I can't find a relation between port/pin and muxpos.
example: `{GPIO_PIN(PB, 0), ADC_INPUTCTRL_MUXPOS_PIN8},      /* EXT1, pin 3 */`

I found the literal definition on lines `334-353` in `cpu/sam0_common/include/vendor/samd21/include/component/adc.h`, but I can't seem to find how these are related to pins.

Alexander-M1, 2019-04-03:
@kaspar030 I think I found the answer. I'll get back once I (don't) get it going. Do you prefer continuing the thread, or moving the conversation elsewhere (e-mail)? Also, is there some sort of "open" forum for all kinds of questions regarding RIOT?

[image]

Alexander-M1, 2019-04-08:
I got it working with the bellow configuration. It is valid for the [Rhomb Duino Zero](https://wiki.rhomb.io/index.php?title=S200_Master_-_Duino_Zero) board, which includes the same chip as the Arduino Zero, the Atsamd21g18a. I also implemented specific ADC settings for the board like external mux control and reference voltage, and I added ADC routines with shorter execution times.
If anyone is using the Duino Zero board and wishes to use ADC, write me a message.
 
```
static const adc_conf_chan_t adc_channels[] = {
   /* port, pin, muxpos (for internal signals port and pin are dummy values) */
	{GPIO_PIN(PB, 8), ADC_INPUTCTRL_MUXPOS_PIN2},	// A0
	{GPIO_PIN(PA, 5), ADC_INPUTCTRL_MUXPOS_PIN5},	// A1
	{GPIO_PIN(PB, 9), ADC_INPUTCTRL_MUXPOS_PIN3},	// A2
	{GPIO_PIN(PB, 2), ADC_INPUTCTRL_MUXPOS_PIN10},	// A3
	{0x0, 0x18},					// Temp
	{0x0, 0x19},					// Bandgap
	{0x0, 0x1A},					// SCALEDCOREVCC
	{0x0, 0x1B}					// SCALEDIOVCC
};
```

fjmolinas, 2019-04-19:
@Alexander-M1 Its awesome you made it work! Great work!, would you wan't to contribute your solution to RIOT so the next person that wants to use adc on arduino-zero can use your solution? You can do this be isolating your changes in a separate branch and making a PR (pull request) to RIOT. You can find a general description at the end of [THIS](https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT) wiki page. Cheers!

Alexander-M1, 2019-05-15:
> @Alexander-M1 Its awesome you made it work! Great work!, would you wan't to contribute your solution to RIOT so the next person that wants to use adc on arduino-zero can use your solution? You can [...]

Hi,

I'd love to. Ho

Alexander-M1, 2019-05-15:
> @Alexander-M1 Its awesome you made it work! Great work!, would you wan't to contribute your solution to RIOT so the next person that wants to use adc on arduino-zero can use your solution? You can [...]

Hi,
I'd be glad to. I think the Rhomb board deserves a place in RIOT and since I'm getting to know it better, I will try to include it in the official RIOT release.
_...better late, than never_
Best,
Alex

closed 2019-05-15 by Alexander-M1