RIOT-OS/RIOT #11354: ESP32: `write(STDIO_FILENO, ...)` not working

maribu, 2019-04-08:
#### Description
The `print()`-family of functions in `fmt.h` use `write(STDIO_FILENO, str, str_len)` under the hood to write to stdio. This does not work on the ESP32 platform, even though `puts()` and `printf()` work.

#### Steps to reproduce the issue

Apply the following patch to `examples/hello-world` and flash and run it on an ESP32 board.

``` diff
diff --git a/examples/hello-world/Makefile b/examples/hello-world/Makefile
index 258d8e9ba..6e4a946c9 100644
--- a/examples/hello-world/Makefile
+++ b/examples/hello-world/Makefile
@@ -15,4 +15,6 @@ DEVELHELP ?= 1
 # Change this to 0 show compiler invocation lines by default:
 QUIET ?= 1
 
+USEMODULE += fmt
+
 include $(RIOTBASE)/Makefile.include
diff --git a/examples/hello-world/main.c b/examples/hello-world/main.c
index f51bf8c0a..5f06e2560 100644
--- a/examples/hello-world/main.c
+++ b/examples/hello-world/main.c
@@ -21,9 +21,12 @@
 
 #include <stdio.h>
 
+#include "fmt.h"
+
 int main(void)
 {
-    puts("Hello World!");
+    puts("Hello World via puts!");
+    print_str("Hello world via print_str!");
 
     printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
     printf("This board features a(n) %s MCU.\n", RIOT_MCU);
```

#### Expected results
Console output:
```
2019-04-08 14:00:36,051 - INFO # Starting RIOT kernel on PRO cpu
2019-04-08 14:00:36,061 - INFO # I (276) [main_trampoline]: main(): This is RIOT! (Version: 2017.10-devel-7371-g4dd09)
2019-04-08 14:00:36,062 - INFO # Hello World via puts!
2019-04-08 14:00:36,062 - INFO # Hello World via print_str!
2019-04-08 14:00:36,063 - INFO # You are running RIOT on a(n) esp32-mh-et-live-minikit board.
2019-04-08 14:00:36,063 - INFO # This board features a(n) esp32 MCU.
```

#### Actual results
Console output:
```
2019-04-08 14:00:36,051 - INFO # Starting RIOT kernel on PRO cpu
2019-04-08 14:00:36,061 - INFO # I (276) [main_trampoline]: main(): This is RIOT! (Version: 2017.10-devel-7371-g4dd09)
2019-04-08 14:00:36,062 - INFO # Hello World via puts!
2019-04-08 14:00:36,063 - INFO # You are running RIOT on a(n) esp32-mh-et-live-minikit board.
2019-04-08 14:00:36,063 - INFO # This board features a(n) esp32 MCU.
```

#### Versions
Current master

gschorcht, 2019-04-11:
@maribu Thanks for figuring out this problem. It seems that ESP32's newlibc `_write_r` function does't write to the UART interface for `stdout`. The reason isn't really clear to me at the moment. Maybe, it requires that `stdout` is opened explicitly at startup.

Since I'm quite busy at the moment, I have no time to investigate it in more deep. Therefore, I will provide a PR which uses `newlib_syscalls_default` module by default as many other architectures do. That will fix the problem for now.

maribu, 2019-04-12:
> Therefore, I will provide a PR which uses `newlib_syscalls_default` module by default as many other architectures do. That will fix the problem for now.

Being honest, I don't think users will mind to have that enabled by default. This will reduce testing effort by having less configurations to test and, thus, will reduce maintenance effort and opportunities of bugs slipped through. The cost is a slight increase in memory requirements. In the context of the quite resourceful ESP32, this seems to be a good deal to me.

closed 2019-04-12 by maribu