RIOT-OS/RIOT #11390: gnrc networking crashs on nRF51dk

Citrullin, 2019-04-12:
#### Description
Flashing the gnrc_networking example on the nRF51dk leads to a crash on the device.
```bash
RIOT network stack example application
All up, running the shell now
> 
Context before hardfault:
```
Since it only uses nrfmin, I suppose is intended to not work yet. I guess this will be fixed when RIOT switches to nimBLE. @haukepetersen is on it as described in https://github.com/RIOT-OS/RIOT/issues/11147.

#### Steps to reproduce the issue
1. Checkout 0ea3c4f984992c13d42c7923647ffd63c3234af9 
2. cd into examples/gnrc_networking
3. Flash to the nRF51dk with ``` BOARD=nrf51dk make flash```

#### Expected results
Working IPv6 on BLE.

#### Actual results
Crash

#### Versions

```bash
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "18.04.1 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-47-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.13.3
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.17.1
                   make: GNU Make 4.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15rc1
                python2: Python 2.7.15rc1
                python3: Python 3.6.7
             coccinelle: missing
```

aabadie, 2019-07-04:
I can't reproduce your issue on master (3f141e9184ef7db8435d3683373d809e92eaf74d). Flashing gnrc_networking works, ping works between 2 nrf51dk.

@Citrullin do you still have the issue or can this be closed ?

aabadie, 2019-10-08:
I cannot reproduce this issue at all with 0ea3c4f (on one nrf51dk of iot-lab):

<details>

```
$ git checkout 0ea3c4f
Note: checking out '0ea3c4f'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 0ea3c4f984 Merge pull request #11383 from aabadie/pr/cpu/stm32l0_pm_initial_blocker
$ make BOARD=nrf51dk IOTLAB_NODE=auto-ssh -C examples/gnrc_networking flash term
make: Entering directory '/work/riot/RIOT/examples/gnrc_networking'
Building application "gnrc_networking" for "nrf51dk" with MCU "nrf51".

"make" -C /work/riot/RIOT/boards/nrf51dk
"make" -C /work/riot/RIOT/core
"make" -C /work/riot/RIOT/cpu/nrf51
"make" -C /work/riot/RIOT/cpu/cortexm_common
"make" -C /work/riot/RIOT/cpu/cortexm_common/periph
"make" -C /work/riot/RIOT/cpu/nrf51/periph
"make" -C /work/riot/RIOT/cpu/nrf5x_common
"make" -C /work/riot/RIOT/cpu/nrf5x_common/periph
"make" -C /work/riot/RIOT/cpu/nrf5x_common/radio/nrfmin
"make" -C /work/riot/RIOT/drivers
"make" -C /work/riot/RIOT/drivers/periph_common
"make" -C /work/riot/RIOT/sys
"make" -C /work/riot/RIOT/sys/auto_init
"make" -C /work/riot/RIOT/sys/auto_init/netif
"make" -C /work/riot/RIOT/sys/div
"make" -C /work/riot/RIOT/sys/evtimer
"make" -C /work/riot/RIOT/sys/fmt
"make" -C /work/riot/RIOT/sys/isrpipe
"make" -C /work/riot/RIOT/sys/luid
"make" -C /work/riot/RIOT/sys/net/crosslayer/inet_csum
"make" -C /work/riot/RIOT/sys/net/gnrc
"make" -C /work/riot/RIOT/sys/net/gnrc/netapi
"make" -C /work/riot/RIOT/sys/net/gnrc/netif
"make" -C /work/riot/RIOT/sys/net/gnrc/netif/hdr
"make" -C /work/riot/RIOT/sys/net/gnrc/netreg
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/icmpv6
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/icmpv6/echo
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/icmpv6/error
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/ipv6
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/ipv6/hdr
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/ipv6/nib
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/ndp
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/sixlowpan
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/sixlowpan/ctx
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/sixlowpan/frag
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/sixlowpan/iphc
"make" -C /work/riot/RIOT/sys/net/gnrc/network_layer/sixlowpan/nd
"make" -C /work/riot/RIOT/sys/net/gnrc/pkt
"make" -C /work/riot/RIOT/sys/net/gnrc/pktbuf
"make" -C /work/riot/RIOT/sys/net/gnrc/pktbuf_static
"make" -C /work/riot/RIOT/sys/net/gnrc/pktdump
"make" -C /work/riot/RIOT/sys/net/gnrc/routing/rpl
"make" -C /work/riot/RIOT/sys/net/gnrc/transport_layer/udp
"make" -C /work/riot/RIOT/sys/net/link_layer/l2util
"make" -C /work/riot/RIOT/sys/net/network_layer/icmpv6
"make" -C /work/riot/RIOT/sys/net/network_layer/ipv6/addr
"make" -C /work/riot/RIOT/sys/net/network_layer/ipv6/hdr
"make" -C /work/riot/RIOT/sys/net/network_layer/sixlowpan
"make" -C /work/riot/RIOT/sys/net/transport_layer/udp
"make" -C /work/riot/RIOT/sys/newlib_syscalls_default
"make" -C /work/riot/RIOT/sys/od
"make" -C /work/riot/RIOT/sys/ps
"make" -C /work/riot/RIOT/sys/random
"make" -C /work/riot/RIOT/sys/random/tinymt32
"make" -C /work/riot/RIOT/sys/shell
"make" -C /work/riot/RIOT/sys/shell/commands
"make" -C /work/riot/RIOT/sys/stdio_uart
"make" -C /work/riot/RIOT/sys/trickle
"make" -C /work/riot/RIOT/sys/tsrb
"make" -C /work/riot/RIOT/sys/xtimer
   text	   data	    bss	    dec	    hex	filename
  81748	    192	  14856	  96796	  17a1c	/work/riot/RIOT/examples/gnrc_networking/bin/nrf51dk/gnrc_networking.elf
iotlab-node --jmespath='keys(@)[0]' --format='int'  --list saclay,nrf51dk,2 --update /work/riot/RIOT/examples/gnrc_networking/bin/nrf51dk/gnrc_networking.elf | grep 0
0
ssh -t abadie@saclay.iot-lab.info 'socat - tcp:nrf51dk-2.saclay.iot-lab.info:20000'
> help
help
Command              Description
---------------------------------------
udp                  send data over UDP and listen on UDP ports
reboot               Reboot the node
ps                   Prints information about running threads.
ping6                Ping via ICMPv6
random_init          initializes the PRNG
random_get           returns 32 bit of pseudo randomness
nib                  Configure neighbor information base
ifconfig             Configure network interfaces
rpl                  rpl configuration tool ('rpl help' for more information)
> ifconfig
ifconfig
Iface  7  HWaddr: 8F:6F  Channel: 0  NID: 0xaffe
           TX-Power: 0dBm  State: IDLE 
          L2-PDU:200 MTU:1280  HL:64  RTR  
          6LO  IPHC  
          Link type: wireless
          inet6 addr: fe80::ff:fe00:8f6f  scope: local  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff00:8f6f
          inet6 group: ff02::1a
          
          Statistics for Layer 2
            RX packets 0  bytes 0
            TX packets 0 (Multicast: 0)  bytes 72
            TX succeeded 0 errors 0
          Statistics for IPv6
            RX packets 0  bytes 0
            TX packets 3 (Multicast: 3)  bytes 162
            TX succeeded 3 errors 0
```

</details>

It's working like a charm, even on latest master (and I found a bug with IoT-LAB integration for nrf51dk, currently looking into it).

Should/can we close this one ?

aabadie, 2019-10-16:
I think this one is a non issue (when strictly following the description). Can someone (@fjmolinas, @miri64) try what I did in https://github.com/RIOT-OS/RIOT/issues/11390#issuecomment-539472417 and confirm this non problem ? It takes 2 minutes.

miri64, 2019-10-16:
If @fjmolinas does not react, can you ping me again next week if I forget? Currently, I'm on a conference ;-).

fjmolinas, 2019-10-16:
If I understand the issue correctly this is a `FEATURE_REQUEST` to be able to use `ble/nimble` in `examples/gnrc_networking` (ipv6 over ble) with an `nrf51dk` board.

The first part was addressed in #11578. 

AFAIK the full request can't be fulfilled for the specified board since `nrf51` is too small to fit `gnrc_networking` + `nimble`:

```
"make" -C /home/francisco/workspace/RIOT-test/sys/xtimer
/opt/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: /home/francisco/workspace/RIOT-test/examples/gnrc_networking/bin/nrf51dk/gnrc_networking.elf section `.bss' will not fit in region `ram'
/opt/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: region `ram' overflowed by 10576 bytes
collect2: error: ld returned 1 exit status
```

I would say this issue is a "wont fix", since it doesn't depend on us but on `nimble` size.