RIOT-OS/RIOT #11398: gnrc_netif_ieee802154: Security issue with malformed headers

shr70, 2019-04-15:
Hi,
I am currently working on writing a netdev driver for a prototype transceiver created at the IAS of RWTH Aachen. Since the transceiver is not working perfectly right now it will also receive noise and send it to RIOT. This setup therefore created a crude fuzzer. 
I noticed that I run in a problem in `gnrc_netif_ieee802154.c` when malformed packets are detected.

`static gnrc_pktsnip_t *_recv(gnrc_netif_t *netif)` will call `nread = dev->driver->recv(dev, pkt->data, bytes_expected, &rx_info);`. Therefore `nread` will contain the amount of bytes read. In my case e.g. 10 bytes. Now `size_t mhr_len = ieee802154_get_frame_hdr_len(pkt->data);` will be called to determine the size of the header by evaluating the FCS bytes. In my case this equaled 21 bytes. The problem then happens when `nread -= mhr_len;` is called. `nread` underflows and `nread` equals now 65525. `od_hex_dump(pkt->data, nread, OD_WIDTH_DEFAULT);` will therefore dump the content of the stack. Further I believe that this might be able to cause a bufferoverflow somewhere in the stack, as the space is only allocated for e.g. 10 bytes and the wrong length will be passed on. However I did not research this any further. 

I believe a check would require to be added in order to make sure the underflow can't happen.

miri64, 2019-04-15:
Thanks for reporting. For future reference, please send security related to security@riot-os.org.

closed 2019-04-15 by SemjonKerner

miri64, 2019-04-15:
> `od_hex_dump(pkt->data, nread, OD_WIDTH_DEFAULT);` will therefore dump the content of the stack.

Since this is only true with `ENABLE_DEBUG` in `gnrc_netif_ieee802154.c`. I see can confirm that this is a bug (and that's why I fixed it), but I don't see the security risk since ENABLE_DEBUG must be explicitly set within the file by the developer:

https://github.com/RIOT-OS/RIOT/blob/245a499f89ecb9095857500c79e0d4880805d060/sys/net/gnrc/netif/ieee802154/gnrc_netif_ieee802154.c#L24-L25

Due to a missing check here 

https://github.com/RIOT-OS/RIOT/blob/245a499f89ecb9095857500c79e0d4880805d060/sys/net/gnrc/netif/ieee802154/gnrc_netif_ieee802154.c#L211 

(which is the only place where `nread` is used after the missing check) the only "dangerous" thing is that the upper layer is receiving some bogus data from the packet buffer instead of the packet being ignored, as the `gnrc_pktbuf_realloc_data()` is erroring due to a very large size provided, but the error isn't checked (since we expected `nread` to be lesser or equal to `pkt->size` which with the fix is now the case again) so the packet is given un-realloced to the upper layer.

Have a look at my test case in #11401 for further reference.