RIOT-OS/RIOT #11404: calliope-mini: flash script not very cleanly written

miri64, 2019-04-16:
While testing #11395, I noticed that the [flash script for the calliope mini](https://github.com/RIOT-OS/RIOT/blob/master/boards/calliope-mini/dist/flash.sh) isn't that well written. It makes some assumptions that just aren't given or expects given configuration that isn't documented anywhere:

- It assumes that Linux always auto-mounts at `/media/${USER}` which it doesn't (e.g. in Arch it is `/run/media/${USER}`)
  https://github.com/RIOT-OS/RIOT/blob/5e789c659584534ffb2e7eb9c86a56a9f6127dcf/boards/calliope-mini/dist/flash.sh#L24-L26
- Appearently it assumes that the storage device of the calliope mini is configured in the `fstab` and can somehow be mounted this way without root. Otherwise I can't explain https://github.com/RIOT-OS/RIOT/blob/5e789c659584534ffb2e7eb9c86a56a9f6127dcf/boards/calliope-mini/dist/flash.sh#L42 to myself.
- It exits on 0 in its error states, so make returns success even if flashing failed.

Sadly, after a short while of trying to fix it I had to give up, since I'm short on time, but I suggest instead of trying to infer the device somehow from the mount-point to use tools like [`lsblk`](https://linux.die.net/man/8/lsblk) (on Linux) and [`diskutil`](https://www.unix.com/man-page/All/8/diskutil) (on OSX) to determine the device name from the `MINI` label (maybe even confirm the USB-ID) and mount the device using `udiskctl` [[1]] [[2]] if it is available to mount it (don't know the equivalent on OSX though).

[1]: https://wiki.ubuntuusers.de/DeviceKit/
[2]: https://wiki.archlinux.org/index.php/Udisks

aabadie, 2019-04-16:
Maybe, you can try to flash this board using openocd, reusing the same configuration that is used for the microbit ? I suspect this board is using a DAPLink programmer. Maybe pyocd would also work.

aabadie, 2019-04-16:
Seems to be the case indeed: https://github.com/calliope-mini/calliope-project-template/blob/master/openocd.cfg

aabadie, 2019-07-04:
The flash script was removed in #11608 so we can close this one.

closed 2019-07-04 by aabadie