RIOT-OS/RIOT #11421: GoMacH on nRF52840

dragondgold, 2019-04-18:
#### Description
I'm really interested and amazed by the RIOT GoMacH protocol I still don't know why these protocols are not as widespread compared to Zigbee, Thread or BLE. They offer mesh networking allowing for the entire network to go to sleep, not just 'end devices', this is awesome for low power sensors networks.

I currently have a DIY synchronized mesh network like this done with XBee devices and the DigiMesh protocol. I just keep all the sensors timers in sync so they can all wake and go to sleep at almost the same time (I can keep them synchronized with an error of around 50 ms, which is a lot compared to what is achievable with GoMacH).

I'm now planning to move away from XBee and instead use an nRF52840 from Nordic. I'm trying to have a synchronized network to achieve low power because all my nodes need to go to sleep, even the 'central node', they all run from batteries.

I see [here](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_networking_mac) that the GoMacH protocol is only supported for the samr21-xpro and iotlab-m3 boards. Is this correct or are there new boards support? If not, is there any other protocol that can achieve this working for this chipset? Or if that's not the case neither, is there any guide to port GoMacH to the nRF52840, I will be more than happy to port it if it's something achievable.

miri64, 2019-04-18:
Hey @dragondgold, thank you for your interest. Currently, I'd say GoMacH is more of an experimental feature, so that's why it is not as advertised ;-). Same goes for the IEEE 802.15.4 support for the nRF52840 btw (which was only merged quite recently).

I assigned @zhuoshuguo to this issue who developed the GoMacH protocol and @SemjonKerner who is the maintainer of the the nrf802154 module, maybe they can find some time to answer your questions.

dragondgold, 2019-04-18:
@miri64 hooooo I see, that explains a lot! Thank you for taking the time to reply. I will wait to see if @zhuoshuguo or @SemjonKerner can suggest me any other protocol or any suggestion about GoMacH, thank you! :)

miri64, 2019-04-18:
BTW: We are hoping to get 6TISCH-support (IEEE 802.15.4e, see [RFC 8180](https://tools.ietf.org/html/rfc8180)) layer some day (see #8570), but the development on that is sadly somewhat slow. This would be the standard-compliant protocol for the MAC layer.

dragondgold, 2019-04-18:
So 6TISCH would be like a standard-compliant version GoMacH? With mesh support and being able to have the entire network go to sleep?

miri64, 2019-04-18:
> So 6TISCH would be like a standard-compliant version GoMacH? With mesh support and being able to have the entire network go to sleep?

I'm not sure enough about the feature sets of both 6TISCH and GoMacH to answer that question in a satisfying manner ;-).

dragondgold, 2019-04-19:
Ok, thank you @miri64 :) Do you know if LWMAC is also an experimental feature not ready for production yet?

miri64, 2019-04-19:
> Ok, thank you @miri64 :) Do you know if LWMAC is also an experimental feature not ready for production yet?

As far as I understand GoMacH is an extension of LWMAC, so yes.

zhuoshuguo, 2019-04-25:
Hi, thank you for your interests on GoMacH. :-)

> is there any guide to port GoMacH to the nRF52840, I will be more than happy to port it if it's something achievable.

The reason why I listed that GoMacH only supports samr21-xpro and iotlab-m3  is that GoMacH requires RTT (the two boads have RTT module) when I was developing it.

Since GoMacH's features are developed based on the general `gnrc-netdev` interface.
So, if nRF52840 is well supported in RIOT with `gnrc-netdev` interface, and also with the RTT module, then GoMacH should theoretically support nRF52840 as well.

zhuoshuguo, 2019-04-27:
@dragondgold By the way, it should be noted that, since RIOT didn't support `deep-sleep` the time when I was developing GoMacH, so currently GoMacH (also LWMAC) only puts the radio to sleep periodically, not the whole chip.

@miri64 Does RIOT support `deep-sleep` now? If yes, I will schedule a that tries to include this function into GoMacH and LWMAC.

dragondgold, 2019-04-27:
Hi @zhuoshuguo, thank you for your replies! I will be porting OpenWSN for the nRF52840, is there any difference between GoMacH and OpenWSN? Because as far as I can understand they are both TSCH (time synchronized channel hopping) networks.

zhuoshuguo, 2019-04-28:
@dragondgold Hi, actually, GoMacH is a distributed solution (doesn't have global synchronization ) and as far as I know OpenWSN is based on network wide synchronization. 
A node in GoMacH will only maintain sychronized with its receiver (phase-lock) when it tries to communicate with the node. 
Also, the way how they utilize the multi-channel scheme is different. I would say, GoMacH just behaves like TSCH in some way, but not exactly TSCH. 
The throughput boost schemes should also be different, GoMacH is based on its queue-length based dynamic slots allocation scheme while I don't know how that is exactly done in OpenWSN.
You can check the basic scheme in [PR-5618](https://github.com/RIOT-OS/RIOT/pull/5618) (or the related publication listed in the PR) for more detailed information about GoMacH.

SemjonKerner, 2019-04-29:
> Hi @zhuoshuguo, thank you for your replies! I will be porting OpenWSN for the nRF52840, is there any difference between GoMacH and OpenWSN? Because as far as I can understand they are both TSCH (time synchronized channel hopping) networks.

SemjonKerner, 2019-04-29:
Sorry, wrong button.

closed 2019-12-01 by stale[bot]