RIOT-OS/RIOT #11423: cpu/kinetis: features provided not properly defined according to series

fjmolinas, 2019-04-19:
#### Description

Since in base Makefile.include, Makefile.fatures for a BOARD is included after `$(RIOTCPU)/$(CPU)/Makefile.include` there is not definition of KINETIS_SERIES when calling this statement

https://github.com/RIOT-OS/RIOT/blob/9a6473104d27e050a5e9318d58dc92bb55c4f268/cpu/kinetis/Makefile.features#L5-L9

Therefore the statement has not effect and the FEATURE_PROVIDED is allways periph_mcg.

#### Steps to reproduce the issue

Add `$(info Kinetis series: $(KINETIS_SERIES))` just before the above statement and call the below statement for any Kinetis board.

`make -C examples/hello-world/ BOARD=pba-d-01-kw2x`

#### Expected results

See in terminal:
```

make: Entering directory '/home/francisco/workspace/RIOT/examples/hello-world'
Kinetis series: W
Building application "hello-world" for "pba-d-01-kw2x" with MCU "kinetis".
```

#### Actual results

```
make: Entering directory '/home/francisco/workspace/RIOT/examples/hello-world'
Kinetis series:
Building application "hello-world" for "pba-d-01-kw2x" with MCU "kinetis".
```

#### Versions

aabadie, 2020-01-03:
@fjmolinas, is this issue still valid ? I don't think as now the check is done on `CPU_MODEL` variable. Do you confirm that we can close ?

fjmolinas, 2020-01-03:
> @fjmolinas, is this issue still valid ? I don't think as now the check is done on CPU_MODEL variable. Do you confirm that we can close ?

Yep, closed. Thanks for the reminder.

closed 2020-01-03 by fjmolinas