RIOT-OS/RIOT #11447: frdm-k64f hwrng support is broken

cladmi, 2019-04-25:
#### Description

I noticed that many tests fails on the `frdm-k64f` when `hwrng` was enabled if available by 64489e8b444efde859e9f048b17707e5da88a983.

However, before this `tests/periph_hwrng` was crashing anyway.

Bisecting the problem led me to 925a908d957f11e40d2935130232322109cedd83

#### Steps to reproduce the issue

Run the `tests/periph_hwrng` firmware for `frdm-k64f`. There is no test yet but the term output should be explicit enough.

#### Expected results

```
2019-04-25 15:10:14,184 - INFO # generating 1 random byte(s)                                                                                                                                                          
2019-04-25 15:10:14,184 - INFO # Got: 0x3c                                                                                                                                                                             
2019-04-25 15:10:14,184 - INFO # generating 2 random byte(s)                                                                                                                                                           
2019-04-25 15:10:14,185 - INFO # Got: 0x93 0x7c
2019-04-25 15:10:14,185 - INFO # generating 3 random byte(s)
2019-04-25 15:10:14,186 - INFO # Got: 0x99 0x24 0x68
2019-04-25 15:10:14,186 - INFO # generating 4 random byte(s)
2019-04-25 15:10:14,187 - INFO # Got: 0xfc 0x9d 0xe5 0x39
2019-04-25 15:10:14,187 - INFO # generating 5 random byte(s)
2019-04-25 15:10:14,188 - INFO # Got: 0x4a 0x5b 0x62 0x0f 0x22
2019-04-25 15:10:14,189 - INFO # generating 6 random byte(s)
2019-04-25 15:10:14,189 - INFO # Got: 0xf5 0x6f 0x29 0xf1 0xc9 0xdc
2019-04-25 15:10:14,190 - INFO # generating 7 random byte(s)
2019-04-25 15:10:14,190 - INFO # Got: 0x8f 0x27 0x21 0xe2 0xfb 0x2b 0x01
2019-04-25 15:10:14,191 - INFO # generating 8 random byte(s)
2019-04-25 15:10:14,192 - INFO # Got: 0x16 0x37 0xf7 0x0c 0x9e 0x7c 0x4c 0x15
2019-04-25 15:10:14,192 - INFO # generating 9 random byte(s)
2019-04-25 15:10:14,193 - INFO # Got: 0x0b 0x0a 0x28 0x4c 0xe3 0x9f 0xd7 0x1c 0x67
2019-04-25 15:10:14,194 - INFO # generating 10 random byte(s)
2019-04-25 15:10:14,195 - INFO # Got: 0xd6 0xcd 0x5f 0x03 0xad 0xfe 0xb2 0x5c 0x27 0xcd
2019-04-25 15:10:14,195 - INFO # generating 11 random byte(s)
2019-04-25 15:10:14,196 - INFO # Got: 0xa4 0xa3 0x79 0x50 0x4c 0xb9 0x3f 0x3a 0x94 0xef 0xd9
2019-04-25 15:10:14,197 - INFO # generating 12 random byte(s)
2019-04-25 15:10:14,198 - INFO # Got: 0xef 0x7f 0x44 0x13 0x9c 0x01 0x60 0x2d 0xfe 0x82 0x90 0x0a
2019-04-25 15:10:14,198 - INFO # generating 13 random byte(s)
2019-04-25 15:10:14,200 - INFO # Got: 0xe4 0xbd 0xe2 0x97 0x84 0x98 0x56 0x5e 0x29 0xa2 0x91 0xc0 0xd7
2019-04-25 15:10:14,200 - INFO # generating 14 random byte(s)
2019-04-25 15:10:14,201 - INFO # Got: 0x5f 0xcb 0x28 0x0d 0x1f 0x7b 0xc7 0x64 0xbd 0xea 0x97 0xb9 0x72 0x3f
2019-04-25 15:10:14,202 - INFO # generating 15 random byte(s)
2019-04-25 15:10:14,203 - INFO # Got: 0xbe 0xae 0xa0 0x7e 0x41 0x17 0xcf 0x7c 0x7b 0xec 0x0f 0xd3 0xbe 0x0f 0x31
2019-04-25 15:10:14,204 - INFO # generating 16 random byte(s)
2019-04-25 15:10:14,205 - INFO # Got: 0x16 0xea 0xb0 0xb2 0x59 0x8a 0xac 0x7d 0x8e 0xf8 0x58 0x76 0x2c 0x51 0xf3 0x3a
2019-04-25 15:10:14,205 - INFO # generating 17 random byte(s)
2019-04-25 15:10:14,207 - INFO # Got: 0x9a 0x53 0x5a 0x03 0x36 0xe3 0x83 0xc0 0x69 0x94 0x29 0x97 0xe5 0xb6 0xaa 0x0e 0x26
2019-04-25 15:10:14,207 - INFO # generating 18 random byte(s)
2019-04-25 15:10:14,208 - INFO # Got: 0x2e 0x59 0xc7 0x4f 0x59 0x78 0x79 0x50 0x13 0xa9 0xa8 0x8b 0xd7 0xae 0xb9 0xd2 0x96 0x8c
2019-04-25 15:10:14,208 - INFO # generating 19 random byte(s)
2019-04-25 15:10:14,210 - INFO # Got: 0xdc 0x94 0x35 0x9d 0x68 0xde 0x07 0xb6 0xcc 0x1f 0xb3 0x1d 0x25 0x90 0x26 0xb4 0xaa 0xed 0x69
2019-04-25 15:10:14,210 - INFO # generating 20 random byte(s)
2019-04-25 15:10:14,211 - INFO # Got: 0xa4 0x15 0xfd 0x82 0xda 0x2f 0x5e 0x55 0x73 0x55 0xc8 0x7e 0x5c 0xdd 0x9a 0x12 0x09 0x8f 0x76 0x1c
```

#### Actual results

A hardfault

```
2019-04-25 15:17:05,547 - INFO # HWRNG peripheral driver test
2019-04-25 15:17:05,547 - INFO #
2019-04-25 15:17:05,553 - INFO # This test will print from 1 to 20 random bytes about every second
2019-04-25 15:17:05,554 - INFO #
2019-04-25 15:17:05,555 - INFO # generating 1 random byte(s)
2019-04-25 15:17:05,556 - INFO #
2019-04-25 15:17:05,558 - INFO # Context before hardfault:
2019-04-25 15:17:05,559 - INFO #    r0: 0x1fff0a44
2019-04-25 15:17:05,561 - INFO #    r1: 0x00000001
2019-04-25 15:17:05,562 - INFO #    r2: 0x1fff0ce4
2019-04-25 15:17:05,564 - INFO #    r3: 0x40029000
2019-04-25 15:17:05,565 - INFO #   r12: 0x1fff09db
2019-04-25 15:17:05,566 - INFO #    lr: 0x000018fd
2019-04-25 15:17:05,568 - INFO #    pc: 0x00001a46
2019-04-25 15:17:05,570 - INFO #   psr: 0x21000200
2019-04-25 15:17:05,570 - INFO #
2019-04-25 15:17:05,571 - INFO # FSR/FAR:
2019-04-25 15:17:05,572 - INFO #  CFSR: 0x00008600
2019-04-25 15:17:05,574 - INFO #  HFSR: 0x40000000
2019-04-25 15:17:05,575 - INFO #  DFSR: 0x00000000
2019-04-25 15:17:05,577 - INFO #  AFSR: 0x00000000
2019-04-25 15:17:05,579 - INFO #  BFAR: 0x40029004
2019-04-25 15:17:05,579 - INFO # Misc
2019-04-25 15:17:05,581 - INFO # EXC_RET: 0xfffffffd
2019-04-25 15:17:05,585 - INFO # Attempting to reconstruct state for debugging...
2019-04-25 15:17:05,586 - INFO # In GDB:
2019-04-25 15:17:05,587 - INFO #   set $pc=0x1a46
2019-04-25 15:17:05,588 - INFO #   frame 0
2019-04-25 15:17:05,589 - INFO #   bt
```

#### Versions

All versions from 925a908d957f11e40d2935130232322109cedd83 (2018.01-RC1) to 2019.07-devel.

cladmi, 2019-04-25:
From a first investigation, the `frdm-k64f` cpu has both

https://github.com/RIOT-OS/RIOT/blob/215d5cdc1c7cb5c61fcdbda150e083759f655c50/cpu/kinetis/include/vendor/MK64F12.h#L10395
and 
https://github.com/RIOT-OS/RIOT/blob/215d5cdc1c7cb5c61fcdbda150e083759f655c50/cpu/kinetis/include/vendor/MK64F12.h#L10479

defined which makes it use 
https://github.com/RIOT-OS/RIOT/blob/215d5cdc1c7cb5c61fcdbda150e083759f655c50/cpu/kinetis/include/cpu_conf_kinetis.h#L159-L161
instead of the previous value

https://github.com/RIOT-OS/RIOT/blob/22c52bd3a837127e34f2fcb8d9abb094e1d8f9c7/boards/frdm-k64f/include/periph_conf.h#L271-L273

Patching to use the `SCGC6` makes it work.

fjmolinas, 2019-08-02:
When looking at `kinetis k64` reference manual under (`AIPS-Lite0` and `AIPS-Lite1`):

> Modules that are disabled via their clock gate control bits in the SIM registers disable the
> associated AIPS slots. Access to any address within an unimplemented or disabled
> peripheral bridge slot results in a transfer error termination.

If the peripheral Bridge for `AIPS-Lite-0` and `AIPS-Lite-1`, `RNGA` is only mapped in `SCGC6`, so it shouldn't use `SCGC3` for `k64`.

See **4.5** and **4.5.2** in [RM](https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/K64P144M120SF5RM.pdf).

closed 2019-08-09 by jcarrano

cladmi, 2019-08-12:
I did not know about https://github.com/fjmolinas/RIOT/blob/0ce2081817af689905843885c9a32f4ec50fbd23/cpu/kinetis/include/cpu_conf_kinetis_k.h
When I tried fixing it, I was defining it in the board but was parsed after the `cpu_conf` files.

Thanks for addressing this.