RIOT-OS/RIOT #11460: SPI MISO of SPI0 should be pulled down in FRDM-KW41Z

taherrera, 2019-04-26:
My FRDM-KW41Z was reseting when communicating with the on-board FLASH (AT45DB041E) via SPI. I was using this configuration to init:
```

/* from board.h */
#define FLASH_SPI_DEV 	FRDM_NOR_SPI_DEV
#define FLASH_SPI_MODE 	SPI_MODE_0
#define FLASH_SPI_CS 	FRDM_NOR_SPI_CS
#define FLASH_SPI_CLK	FRDM_NOR_SPI_CLK

spi_init(FLASH_SPI_DEV);
spi_init_cs(FLASH_SPI_DEV,FLASH_SPI_CS);
```
Reading the datasheet of the AT45DB041E it stated: 

> The SO pin will be in a high-impedance state whenever the device is deselected (CS is deasserted).

When I probed the MISO pin, it was floating, and for some reason, the chip did not reset when measuring it with and osciloscope, and it sometimes worked fine if I sticked my finger on the MISO pin.

Crearly a floating pin problem on an input pin... I have had these symptoms back when I used PIC18.

This solved the problem:

```
spi_init(FLASH_SPI_DEV);
gpio_init(GPIO_PIN(PORT_C,18),GPIO_IN_PD);
spi_init_cs(FLASH_SPI_DEV,FLASH_SPI_CS);

```
I tried changing things in cpu/kinetics/spi.c but it wasnt compiling..

ps: I tried this on 2 different FRDM-KW41Z boards with same results.
ps2: I am using a RIOT branch I dont know if this is solved in HEAD, since I need to use KW41Z wireless transceiver. The last commit of the repo is Dec 17 2018: 09ad33f05f458d93d150c6c34501615e63e39fd9

taherrera, 2019-04-26:
I added a 41k resistor from C18 to GND and it is now working.

taherrera, 2019-04-28:
Acording to the schematic and the user guide, the pin that drives the MISO is conected to gpio PTC18  (MISO) and an other gpio: PTC1. PTC1 could be set as input pulldown. I will try it out and inform.

benemorius, 2019-04-30:
You're right. Currently the default configuration doesn't enable pullups for spi miso and also uart rx.

The bit definition for turning on the pullup is `GPIO_IN_PU`. You can add it to periph_conf.h like so:

`.pcr      = GPIO_AF_2 | GPIO_IN_PU,`

This will enable the pullup on all spi pins, which should be harmless but of course there's no need for pullups on the driven pins.

Alternately, you can add it to spi.c like so:

`gpio_init_port(spi_config[bus].pin_miso, spi_config[bus].pcr | GPIO_IN_PU);`

This way the pullup is only enabled on miso.

Probably we should change all the board configurations to enable pullups where appropriate. It looks like many boards have no pullups by default, which is fine for development, but a floating uart rx pin will definitely cause intermittent boot failures in the field.

PeterKietzmann, 2019-05-21:
@taherrera have you tried @benemorius's suggestion?

taherrera, 2019-05-22:
Hi @PeterKietzmann , yes I tried the alternative sugestion (gpio_init_port(...)) and it seems to be working. I used a pull down on MISO though

PeterKietzmann, 2019-05-22:
@taherrera thanks for testing! For know you should be good to proceed, right? However, SPI is push-pull so enabling extra resistors from within the driver or its configuration is not the right fix. I leave the issue up to @MrKevinWeiss and @MichelRottleuthner

aabadie, 2019-05-22:
Maybe have a look at #11542 which provides a way to configure gpio modes from outside. This is useful for optimizing low-power consumption with LoRa radios.

taherrera, 2019-05-22:
@PeterKietzmann yes, I am currently using @benemorius suggestion and I have not experienced unexpected resets on the frdm-kw41z ! I will use this configuration for now. Thank you both for the help

MrKevinWeiss, 2019-05-23:
Hey @taherrera,
A few things would be helpful.  Is there a test or application or proceedure I can run to try to reproduce the problem (I tried just initializing the spi and sending some bytes but could not fail it on master).

Another thing that should be noted, even if there is a problem and a pullup resistor fixes it, that doesn't mean that the root cause is fixed.  The device shouldn't reset due to the MISO not being pulled, this might be something completely different.

I should have some available time coming up and will investigate further.

taherrera, 2019-05-24:
Hi @MrKevinWeiss 

I dont have the exact application that fired the resets on the board, I recall this happening when writing to the external flash present on the board when the resets ocurred.

I will try to reproduce again this issue and I will send you the application.

MrKevinWeiss, 2019-05-24:
Perfect thank you!

benemorius, 2019-05-29:
I don't have this hardware to test with, and I haven't personally encountered this with SPI before, but after some consideration it's my opinion that unwanted resets can be fully explained by having MISO floating while the SPI peripheral is powered on.

> Reading the datasheet of the AT45DB041E it stated:
> 
> > The SO pin will be in a high-impedance state whenever the device is deselected (CS is deasserted).

If this is true then a pullup/pulldown is strictly required unless you ensure that CS is asserted all the while that MISO is configured as an input, and otherwise you must expect undefined behavior.

Configuring a floating pin as an input means its CMOS input buffer is free to oscillate, and it will sometimes oscillate fast enough that the two input FETs spend too much time simultaneously turned on which provides a path from VCC to GND which results in anomalous current draw and a subsequent supply voltage brownout or instability. I've had some incredibly enigmatic problems from floating UART inputs so this is something I can't help but pay attention to.

This used to be a less frequent concern, as many SPI slave devices have a weak pullup on their output even in their lowest power state, but lately a lot of devices forego the pull resistor to facilitate modern low-power requirements. A superfluous pullup can easily waste a few microamps or even much more, and that's becoming very relevant these days. When the microcontroller is solely responsible for the pull resistor, as the industry is tending towards, then there's no problem with phantom current draw. But then it's essential that the designer make sure there's a pull resistor enabled or else instead of 3uA extra current draw you get an unreliable board.

The observation that touching the pin fixes the problem is a classic confirmation, despite the fact that it's not a very rigorous test. Even if the resets turn out to be from a totally unrelated problem, floating CMOS inputs still need pull resistors, without exception.

https://electronics.stackexchange.com/questions/7179/is-it-really-a-bad-idea-to-leave-an-mcu-input-pin-floating

PeterKietzmann, 2019-08-20:
@benemorius thanks for the explanation. It's interesting read and I never though about it in this way. @taherrera did you reproduce again so you could share the application as you suggested in May?

taherrera, 2019-08-20:
Hi @PeterKietzmann, I tried to reproduce the problem, but sadly I could not make it reset :disappointed:, if I have some time I will try again. I will attach the code I was trying out when this happened in case somebody wants to try it out (It is slightly modified though, it is not exactly the code that fired this problem): [estructural_test_flash.tar.gz](https://github.com/RIOT-OS/RIOT/files/3519881/estructural_test_flash.tar.gz)

closed 2020-03-23 by stale[bot]