RIOT-OS/RIOT #11468: gnrc_networking: setting the address

roberthartung, 2019-04-30:
In `examples/gnrc_networking/Makefile` it is documented that `GNRC_IPV6_STATIC_LLADDR` can be used to define the address, however this is not used anywhere. Before just removing this piece of code from the Makefile, what is the suggested way to set a local address?

miri64, 2019-04-30:
The address link-local address and global addresses usually is auto-configured according to RFCs [4862](https://tools.ietf.org/html/rfc4862) and [6775](https://tools.ietf.org/html/rfc6775) (the latter only if your device uses [6Lo](https://tools.ietf.org/html/rfc4944)). The idea behind `GNRC_IPV6_STATIC_LLADDR` was that if, for any reason, you don't want or can't (e.g. if your link layer doesn't have addresses, see #10483) auto-configure the link-local IPv6 address, you can use `-DGNRC_IPV6_STATIC_LLADDR="<link-local-address>"` to define one at compile time. However, its usage was apparently removed with 71a7dbf91856dc221a5615e29587c6b8309b08d2, so I would rather advocate for a re-implementation than a removal.

paxter, 2019-09-04:
I ran into the same problem. On many places this is documented but it's not working obviously. This is very confusing and would be very helpful for testing.
Just to be sure: There is no possibility at the moment to set up a fixed local link address for testing purposes?

miri64, 2020-03-08:
This was fixed in https://github.com/RIOT-OS/RIOT/pull/13122

closed 2020-03-08 by miri64