RIOT-OS/RIOT #11472: Warnings from objcopy and size (binutils 2.32)

brummer-simon, 2019-04-30:
Hi Everyone,

I ran a Systemupdate on my Arch Linux Notebook, updating gcc and binutils. After the update,
building any RIOT application for native shows Warnings from the objcopy and size commands. I added a screenshot building the hello-world example. I don't know if this causes any issues but warnings from tools like binutils get me usually nervous ;).

[image]

jcarrano, 2019-05-27:
I'm annoyed by that too. It is weird. It complains that segment GNU_PROPERTY (composed of the section ".note.gnu.property" has size zero, but readelf shows it has size 0x10.

miri64, 2019-07-30:
Saw it as well, but don't know how to fix it :-/

aabadie, 2020-07-02:
Is this still an issue ?

miri64, 2020-07-02:
Did not see it for a while, but not sure.

brummer-simon, 2020-07-03:
I didn't see it either in a long time

miri64, 2020-07-03:
Then let's close. Seemed to have been a toolchain issue anyways.

closed 2020-07-03 by miri64