RIOT-OS/RIOT #11494: emcute_mqttsn example setup contains invalid steps

jvijtiuk, 2019-05-07:
#### Description

#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

miri64, 2019-05-07:
Hi @jvijtiuk, welcome to RIOT!

> The command 'ifconfig 5 add fec0:affe::99' fails with the following output:
> SIOGIFINDEX: No such device

This does not read like an error in RIOT. Did you type this within the native application or in Linux?

miri64, 2019-05-07:
Looking at the README with some distance it is indeed not clear, that you have to start the RIOT application first ;-).

miri64, 2019-05-07:
@jvijtiuk does #11495 help with that.

jvijtiuk, 2019-05-07:
Hi @miri64 #11495 resolves the issue, I was running the command on the Linux host, not inside the RIOT native application. Thank you for the quick response :)

closed 2019-05-07 by jvijtiuk