RIOT-OS/RIOT #11515: gnrc: ipv6: UDP: Can't send IPv6 UDP packets over ethernet interface when at86rf233 module is used

roubra, 2019-05-13:
Hi,

we have ported RIOT Release 2018.07 to the XMC4700 Relax Board from Infineon. We also ported an Ethernet driver for the integrated MAC. This makes it possible to send udp packets over IPv6. If the module at86rf233 is now integrated, the packets are apparently sent through the IEEE802.15.4 stack and cannot be sent correctly via the Ethernet interface. I tested the whole thing with a quite simple program. 

Hex dump of working frame:
```
33 33 00 00 00 01 01 02  45 19 00 00 86 DD 60 00
00 00 00 0F 11 40 FE 80  00 00 00 00 00 00 03 02
45 FF FE 19 00 00 FF 02  00 00 00 00 00 00 00 00
00 00 00 00 00 01 B1 F3  B1 F3 00 0F 33 56 48 65
6C 6C 6F 21 00 00 00 00  00 00 00 00 00 00 00 00
```

hex dump of wrong frame:
```
00 E0 4C 13 59 0C 01 02  45 19 00 00 86 DD 60 00
00 00 00 0F 11 40 FE 80  00 00 00 00 00 00 03 02
45 FF FE 19 00 00 FE 80  00 00 00 00 00 00 03 02
45 FF FE 19 00 01 C0 77  B1 F3 00 0F DE 38 48 65
6C 6C 6F 21 00 00 03 02  45 FF FE 19 00 00 01 01
```
```
#include <stdio.h>
#include "net/af.h"
#include "net/protnum.h"
#include "net/ipv6/addr.h"
#include "net/sock/udp.h"
#include "xtimer.h"
uint8_t buf[7];

#define _TEST_ADDR_LOCAL    { 0xfe, 0x80, 0, 0, 0, 0, 0, 0, 0x03, 0x02, 0x45, 0xff, 0xfe, 0x19, 0x00, 0x00 }
#define _TEST_ADDR_REMOTE   { 0xfe, 0x80, 0, 0, 0, 0, 0, 0, 0x03, 0x02, 0x45, 0xff, 0xfe, 0x19, 0x00, 0x01 }

int main(void)
{
    static const sock_udp_ep_t remote = { .addr = { .ipv6 = _TEST_ADDR_REMOTE },
                                          .family = AF_INET6,
                                          .netif = 6,
                                          .port = 45555 };
    while (1) {
        if (sock_udp_send(NULL, "Hello!", sizeof("Hello!"), &remote) < 0) {
            puts("Error sending message");
            return 1;
        }
        xtimer_sleep(1);
    }
    return 0;
}
```
Makefile

```
GNRC_NETIF_NUMOF := 2

# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps
USEMODULE += gnrc_netdev_default
USEMODULE += auto_init_gnrc_netif
USEMODULE += gnrc_ipv6_default
USEMODULE += gnrc_icmpv6_echo
USEMODULE += gnrc_sock_udp
USEMODULE += at86rf233
USEMODULE += xmc4x00_eth

```

We want to use the at86rf233 as receiver of a WSN and send the data over Ethernet. 
How can I prevent the message from going through the IEEE802.15.4 stack?

```
Installed compiler toolchains
-----------------------------
             native gcc: gcc.exe (Rev2, Built by MSYS2 project) 8.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
```
Kind regards,
roubra

miri64, 2019-05-23:
Hello @roubra and welcome to the RIOT community. I'm not one hundred percent sure what the problem is? We have similar setup's e.g. when using a `samr21-xpro` as border router with `ethos` and there it works. Are the packets going out the wrong interface?

Also I noticed you are using `gnrc_ipv6_default`. Since your node has two interfaces I propose to use `gnrc_ipv6_router_default` instead, so forwarding is activated.

roubra, 2019-06-03:
Hi @miri64,
thanks for the feedback. After further tests I noticed that the Ethernet interface does not respond to ICMPv6 responses. As a result, RIOT tries to send the packets via the wrong interface. An interim solution is a fixed entry in nib.c. I suspect that I configured the Ethernet interface incorrectly. I will now contact the manufacturer and ask for support. I will get back to you with new information.

miri64, 2019-06-03:
If it's a forwarding issue it is unlikely the manufacturer can help you. Can you please tell me what exactly is happening, so I might be able to help you?

miri64, 2019-12-05:
Ping @roubra. Can it be that the fix of https://github.com/RIOT-OS/RIOT/issues/12545 fixed your issue as well?

roubra, 2019-12-09:
Hi,
I'll try it out and give you feedback.

closed 2021-02-14 by stale[bot]