RIOT-OS/RIOT #11522: doc: XTIMER_BACKOFF inconsistency

fjmolinas, 2019-05-13:
### Description

Looking into XTIMER_BACKOFF documentation I've run upon some inconsistency over the meaning of the value.

In multiple places XTIMER_BACKOFF is refered to as having usec units, eg:

https://github.com/RIOT-OS/RIOT/blob/568a3514557fe301aaf8b2413b76c701dac98492/sys/include/xtimer.h#L474-L484

https://github.com/RIOT-OS/RIOT/blob/568a3514557fe301aaf8b2413b76c701dac98492/sys/include/xtimer.h#L136-L145

But it is also referred  as ticks:

https://github.com/RIOT-OS/RIOT/blob/568a3514557fe301aaf8b2413b76c701dac98492/sys/include/xtimer.h#L160-L169

But when looking at where it is used, e.g tracing ` xtimer_usleep()`. We can actually see that the value is used a ticks since it is converted from usec to ticks inmediatly.

```
static inline void xtimer_usleep(uint32_t microseconds)
{
    _xtimer_tsleep32(_xtimer_ticks_from_usec(microseconds));
}
```

And when tracing the value from `_xtimer_tsleep32() `to  `_xtimer_tsleep()` to `_xtimer_set64() ` to `_xtimer_set()` we end up seeing that XTIMER_OFFSET is compared against ticks.

https://github.com/RIOT-OS/RIOT/blob/568a3514557fe301aaf8b2413b76c701dac98492/sys/xtimer/xtimer_core.c#L136-L155

This leads to unclear definition of XTIMER_BACKOFF since documentation makes you think of usec and not ticks. This has an impact on some tests, eg https://github.com/RIOT-OS/RIOT/blob/568a3514557fe301aaf8b2413b76c701dac98492/tests/ps_schedstatistics/main.c#L48

Where a variable representing ticks is passed to a function requesting usec. This had already been brought up in #8969.

### Proposed steps to solve the problem

- [ ] change documentation to make sure XTIMER_BACKOFF has a clear unit ( ticks or usec), or add XTIMER_BACKOFF_US
- [ ] fix tests that use XTIMER_BACKOFF as an unproper reference ([tests/ps_schechstatistics](https://github.com/RIOT-OS/RIOT/tree/master/tests/ps_schedstatistics), [tests/thread_float](https://github.com/RIOT-OS/RIOT/tree/master/tests/thread_float), etc.)
- [ ] fix XTIMER_BACKOFF definition for concerned boards [frdmk-kw41z](https://github.com/RIOT-OS/RIOT/blob/master/boards/frdm-kw41z/include/board.h)

#### Related Issues

#8969

SemjonKerner, 2019-05-17:
I would disagree, that all your findings support your proposed steps towards fixing.
From my experience xtimer indeed becomes increasingly unrelieable below 30 microseconds. I think the information for users should include microseconds instead of ticks, since it is more intuitive.

Of course current usage of XTIMER_BACKOFF must be reviewed, whether it is used to compare against us or ticks then.
So from my point of view it is not a problem of documentation, but of proper casting. I'd even prefer if this value of 30us is reviewed properly, to determine a more exact bound.

SemjonKerner, 2019-05-17:
I guess the straight forward approach is the best here. I'd suggest casting `XTIMER_BACKOFF` in those cases where it gets compared to ticks with [xtimer_ticks_from_usec](http://doc.riot-os.org/group__sys__xtimer.html#gae457bf7ed1f4ec0f4f6767a35a4fd181).

From tracing with grep, I guess these are only following two lines:
https://github.com/RIOT-OS/RIOT/blob/568a3514557fe301aaf8b2413b76c701dac98492/sys/xtimer/xtimer_core.c#L147
https://github.com/RIOT-OS/RIOT/blob/568a3514557fe301aaf8b2413b76c701dac98492/sys/xtimer/xtimer_core.c#L201

Also I realized there are varying redefinitions of `XTIMER_BACKOFF` for different boards. Assuming the developers who set this values stuck to the documentation, all those values are supposed to be microseconds, not ticks and hence should be used accordingly, instead of bloating the documentation.

fjmolinas, 2019-05-17:
@SemjonKerner Thanks for your comments! 

> I would disagree, that all your findings support your proposed steps towards fixing.
From my experience xtimer indeed becomes increasingly unreliable below 30 microseconds. I think the information for users should include microseconds instead of ticks, since it is more intuitive.

The first step I proposed was changing the documentation to ticks or us. I was hopping for discussion and did  #11538 to spark this since I don't know much about xtimer. But I probably skipped a step and should have waited for a reaction here, so I'm going back one step :)

> Also I realized there are varying redefinitions of XTIMER_BACKOFF for different boards. Assuming the developers who set this values stuck to the documentation, all those values are supposed to be microseconds, not ticks and hence should be used accordingly, instead of bloating the documentation.

Well the thing is this is not clear, since documentation is ambiguous they could have done it either way, and the way that would have worked properly is if the y used it as ticks.

fjmolinas, 2019-05-17:
> So from my point of view it is not a problem of documentation, but of proper casting.

It is a problem of documentation. The fact that it is ambiguous doesn't make it clear if when implemented it was thought of as a per board configuration variable in ticks or us. I don't know the details of the concept behind the implementation. That being said, you are right 'us' is more intuitive. I would like @kaspar030 input on this since its his implementation.

> I'd even prefer if this value of 30us is reviewed properly, to determine a more exact bound.

+1, and the reasoning for it

> Also I realized there are varying redefinitions of XTIMER_BACKOFF for different boards. Assuming the developers who set this values stuck to the documentation, all those values are supposed to be microseconds, not ticks and hence should be used accordingly, instead of bloating the documentation.

Going back to this, looking at discussion thread https://github.com/RIOT-OS/RIOT/issues/7347 the reasoning behind the value doesn't seem to be more than works/doesn't work. The fact that the discussion makes sense when comparing XTIMER_BACKOFF to 'us' is only because the timer in most cases is 1Mhz. But this is not always the case as in some 8 bit MCU or when using slow low power timers (frdm-kw41z).

kaspar030, 2019-05-20:
More fallout from #5608.

Before that PR, there was no ticks vs. us distinction. When that PR got merged, we didn't go through every xtimer variable in order to specify whether it should be us or ticks. We should have. :)

IMO this should be specified in "us", as users might be be working with it on a high level, and we should spare them the ticks handling.

SemjonKerner, 2019-11-21:
Hey @fjmolinas , I didn't pursue this issue further. The board specific xtimer_backoffs make it very complicated to implement a clean solution. In favor of ztimer I'd abandon this issue.

fjmolinas, 2019-11-22:
> Hey @fjmolinas , I didn't pursue this issue further. The board specific xtimer_backoffs make it very complicated to implement a clean solution. In favor of ztimer I'd abandon this issue.

I know, and I haven't either :) . But this is still an issue, so should remain open (at least until ztimer becomes the default).

maribu, 2022-09-16:
With xtimer on its way out and ztimer being used instead now every, I'd say this is a wontfix.

If anyone disagrees, please reopen.

closed 2022-09-16 by maribu