RIOT-OS/RIOT #11523: xtimer_periodic_wakeup crashing at high frequencies on frdm-kw41z

taherrera, 2019-05-13:
#### Description
The function xtimer_periodic_wakeup does not work or is not reliable at high frequencies (100Hz or more) on the frdm-kw41z. After some time (depending on the frequency) it tends to crash and returns at very high frequencies (160kHz in following example)

#### Steps to reproduce the issue

Test code:

```
#include <stdio.h>
#include <xtimer.h>
#include "periph/gpio.h"

#define APP_GPIO_PIN	GPIO_PIN(PORT_C,6) /* J1 last pin */

int main(void)
{
    puts("Periodic Timer test");
	gpio_init(APP_GPIO_PIN, GPIO_OUT);
	xtimer_init();
	xtimer_ticks32_t last_time = xtimer_now();
	while(1){
		xtimer_periodic_wakeup(&last_time, 2000); /* with 1000 us it crashes almost immediately */
		gpio_toggle(APP_GPIO_PIN);
	}
    return 0;
}
```
makefile:

```
APPLICATION = test-alarm

BOARD ?= frdm-kw41z

RIOTBASE ?= $(CURDIR)/../..

DEVELHELP ?= 1

QUIET ?= 0

USEMODULE += xtimer

include $(RIOTBASE)/Makefile.include
```
#### Notes

I tested the code above with a samr21-xpro and it works fine @1kHz (changing the GPIOs). I tried this on two frdm-kw41z and it crashes almost immediately @1kHz.

#### Versions
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "14.04.6 LTS, Trusty Tahr"
                 Kernel: Linux 4.4.0-146-generic i686 i686

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 4.9.4-2ubuntu1~14.04.1) 4.9.4
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 2.8.12.2
               cppcheck: missing
                doxygen: 1.8.6
                 flake8: missing
                    git: git version 1.9.1
                   make: GNU Make 3.81
                openocd: Open On-Chip Debugger 0.9.0-dev-gbdd8312 (2018-04-05-13:14)
                 python: Python 2.7.6
                python2: Python 2.7.6
                python3: Python 3.4.3
             coccinelle: missing

benemorius, 2019-07-30:
I was able to confirm this on master but not on my local working branch. I'll try to isolate the difference.

[image]

benemorius, 2019-07-30:
On master if I change `XTIMER_BACKOFF` from `5` to `92` then I no longer get this bug. I got that value from a discussion about xtimer ticks/us discrepancies and I'll link it if I find it again. I haven't confirmed what value is actually correct.

benemorius, 2019-07-31:
> I got that value from a discussion about xtimer ticks/us discrepancies and I'll link it if I find it again

https://github.com/RIOT-OS/RIOT/pull/11546#issuecomment-499046537

miri64, 2020-07-05:
So this seems to have been a configuration issue and more of a duplicate of https://github.com/RIOT-OS/RIOT/issues/11522 I close this. Re-open if you disagree.

closed 2020-07-05 by miri64