RIOT-OS/RIOT #11525: API change, uart input not working anymore on previously working setups

cladmi, 2019-05-14:
#### Description

Since https://github.com/RIOT-OS/RIOT/pull/11310 applications that were relying on `getchar` and were previously working stopped working.
I do not know if the impact is on other functions by default too or not.

This is an API change and was review without putting the label and merged without respecting the 2 ACKs for it.

I noticed this when running the test for https://github.com/RIOT-OS/RIOT/pull/11493 with the last master.
I got weird "time took less time than expected" and thought it was missing a synchronization mechanism, so ignored this when merging as was unrelated.

The test was relying on `getchar` which now does not block and silently returns.

#### Steps to reproduce the issue

I found it by doing:

`make -C tests/xtimer_usleep test`

`make -C tests/xtimer_usleep flash`

#### Expected results

The firmware waits for input:

```
�main(): This is RIOT! (Version: 2019.04-devel-308-g2bfa0-HEAD)                                                                                                                                                                                                                           
Running test 5 times with 7 distinct sleep times                                                                                                                                                                                                                                          
Please hit any key and then ENTER to continue                                                                                                                                                                                                                                             
```

#### Actual results

The firmware does not wait.

```
make: Entering directory '/home/harter/work/git/RIOT/tests/xtimer_usleep'                                                                                                                                                                                                                 
/home/harter/work/git/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200"                                                                                                                                                                                                         
Twisted not available, please install it if you want to use pyterm's JSON capabilities                                                                                                                                                                                                    
2019-05-14 13:10:39,720 - INFO # Connect to serial port /dev/ttyACM0                                                                                                                                                                                                                      
Welcome to pyterm!                                                                                                                                                                                                                                                                        
Type '/exit' to exit.                                                                                                                                                                                                                                                                     
2019-05-14 13:10:42,435 - INFO # main(): This is RIOT! (Version: 2019.07-devel-266-gdde4f)                                                                                                                                                                                                
2019-05-14 13:10:42,439 - INFO # Running test 5 times with 7 distinct sleep times                                                                                                                                                                                                         
2019-05-14 13:10:42,443 - INFO # Please hit any key and then ENTER to continue                                                                                                                                                                                                            
2019-05-14 13:10:42,458 - INFO # Slept for 10052 us (expected: 10000 us) Offset: 52 us                                                                                                                                                                                                    
2019-05-14 13:10:42,513 - INFO # Slept for 50052 us (expected: 50000 us) Offset: 52 us
2019-05-14 13:10:42,529 - INFO # Slept for 10286 us (expected: 10234 us) Offset: 52 us
2019-05-14 13:10:42,590 - INFO # Slept for 56832 us (expected: 56780 us) Offset: 52 us
2019-05-14 13:10:42,608 - INFO # Slept for 12174 us (expected: 12122 us) Offset: 52 us
2019-05-14 13:10:42,712 - INFO # Slept for 98817 us (expected: 98765 us) Offset: 52 us
2019-05-14 13:10:42,792 - INFO # Slept for 75052 us (expected: 75000 us) Offset: 52 us
2019-05-14 13:10:42,807 - INFO # Slept for 10052 us (expected: 10000 us) Offset: 52 us
2019-05-14 13:10:42,862 - INFO # Slept for 50052 us (expected: 50000 us) Offset: 52 us
2019-05-14 13:10:42,877 - INFO # Slept for 10286 us (expected: 10234 us) Offset: 52 us
2019-05-14 13:10:42,940 - INFO # Slept for 56832 us (expected: 56780 us) Offset: 52 us
2019-05-14 13:10:42,956 - INFO # Slept for 12174 us (expected: 12122 us) Offset: 52 us
2019-05-14 13:10:43,060 - INFO # Slept for 98817 us (expected: 98765 us) Offset: 52 us
2019-05-14 13:10:43,140 - INFO # Slept for 75052 us (expected: 75000 us) Offset: 52 us
2019-05-14 13:10:43,155 - INFO # Slept for 10052 us (expected: 10000 us) Offset: 52 us
2019-05-14 13:10:43,211 - INFO # Slept for 50052 us (expected: 50000 us) Offset: 52 us
```

#### Versions

After https://github.com/RIOT-OS/RIOT/pull/11310

So current master https://github.com/RIOT-OS/RIOT/commit/e40f483acfa75170821d5da354d9824b907eecdf

kaspar030, 2019-05-14:
> This is an API change and was review without putting the label and merged without respecting the 2 ACKs for it.

This was not an API change. getchar works as expected (from ```man 3 getchar```:

```
 fgetc(), getc() and getchar() return the character read as an unsigned char cast to an int or EOF on end of file or error.
```

I agree that making previously working code not work anymore sucks, though.

The options for fixing are:

1. make uart_stdio_rx optional but default
2. add the dependency to applications that need it.
3. make all stdin reads panic() if support is not enabled
4. other ideas?

Unfortunately adding "uart_stdio_rx" doesn't cut it for e.g., native. There needs to be a level of pseudo modules, e.g., "stdio_rx (or just stdin)" which depends on "uart_stdio_rx" iff "uart_stdio" is selected...

MrKevinWeiss, 2019-05-14:
I mean the purpose of this was for the pm layer to have an easier time right?  Can we just add a module that disables the RX and make it a dependency of the pm layer?

> make uart_stdio_rx optional but default

I thought it is optional by default, do you me use the uart_stdio_rx by default?

> add the dependency to applications that need it.

Doesn't help with backwards compatibility, I would tend to say no

> make all stdin reads panic() if support is not enabled

Probably doesn't hurt regardless

cladmi, 2019-05-14:
> > make uart_stdio_rx optional but default
> 
> I thought it is optional by default, do you me use the uart_stdio_rx by default?

I think it was meant to be "enabled by default".

MrKevinWeiss, 2019-05-14:
Ah ya... that then.

kaspar030, 2019-05-14:
> I mean the purpose of this was for the pm layer to have an easier time right? Can we just add a module that disables the RX and make it a dependency of the pm layer?

Well, the pm layer is enabled by default, that won't work.

MrKevinWeiss, 2019-05-14:
I guess we are narrowing down our options then.

cladmi, 2019-05-14:
It is a RIOT code base API change. If this is not something we describe as an API change, please let me know.

What is wrong here, is not that it needed to be changed, but that it was changed like this.

Touching low level dependencies with uart is sure tricky and can have consequences. The pull request does not have a testing procedure for this change… and does not give a testing procedure for showing it indeed solves the target issue.

The API change can be seen as it done in a single commit, where the shell dependencies must be updated as the same time to not be in a broken state.
If there was no API change, it could and should have been at least two commits.

It is a single line commit message "sys: make uart_stdio RX optional".
No more details on consequences and why.
Why does a commit named like this changes the `shell` dependencies without explanation?

No mention that anything that implicitly depended on this before may break.

All this minimal information led to this PR being handled as a minor change when it should not have been.

To prevent this, I think that we should move in a direction that contributor *should* put more information in the first place, and reviewer *should* ask for more information when there is not enough.
Even if both understand together the issue, another reader, now or in the future, may find this completely obscure.

> I guess we are narrowing down our options then.

The other solution is to now revert this and handle it with more time and mark it with an API change warning if needed.

jcarrano, 2019-05-14:
>  fgetc(), getc() and getchar() return the character read as an unsigned char cast to an int or EOF on end of file or error.

By that logic I can make every function that can fail in RIOT fail (and apps that use it break) and it will still not be an API change?

**This is an OS**: I we expect there to ever be more apps than our own examples/test then we cannot just break apps without a _serious_ reason.

kaspar030, 2019-05-14:
> By that logic I can make every function that can fail in RIOT fail (and apps that use it break) and it will still not be an API change?

No, by that logic it is possible for getchar() to fail with EOF if the underlying stdio returns an error when reading from it.

> I we expect there to ever be more apps than our own examples/test then we cannot just break apps without a serious reason.

Yes, and we're trying hard not to. What's your point?

kaspar030, 2019-05-14:
(seems like github ate my mail reply, so here's it copy&pasted:)

> It is a RIOT code base API change. If this is not something we describe
> as an API change, please let me know.

This is a *configuration* change. *nothing* of the getchar API has been
changed. This is more like we decided that unless otherwise requested,
we do "close(STDIO_FILENO);" at program start.

That might have been a mistake, but this was technically *not an API
change*.

> To prevent this, I think that we should move in a direction that
> contributor /should/ put more information in the first place, and
> reviewer /should/ ask for more information when there is not [...]

I understand your frustration that something you expected working
suddenly broke. I'm sorry that happened.

Let's move on and fix it.

> The other solution is to now revert this and handle it with more time
> and mark it with an API change warning if needed.

Reverting makes no sense, there is a *strong* use case for disabling
uart rx. Changing to the previous default (by selecting uart_stdio_rx by
default) would make more sense.

MrKevinWeiss, 2019-05-14:
> Changing to the previous default (by selecting uart_stdio_rx by
default) would make more sense.

Yup, can we agree to that?

aabadie, 2019-05-14:
> Changing to the previous default (by selecting uart_stdio_rx by default) would make more sense.

Maybe I missed something or I misunderstand the issue but from what I get, on STM32, doing this will make power management less intuitive => STOP mode will be blocked once by UART. We can disable this behaviour at driver level but this other disadvantages. Sorry again if I'm wrong.

kaspar030, 2019-05-14:
> Maybe I missed something or I misunderstand the issue but from what I get, on STM32, doing this will make power management less intuitive => STOP mode will be blocked once by UART.

That is correct. But there's a point to not breaking stdin by default...

jcarrano, 2019-05-14:
> ...on STM32, doing this will make power management less intuitive

Not any worse than it was before.

aabadie, 2019-05-14:
> Not any worse than it was before.

Maybe but for sure worse than it is now :)

aabadie, 2019-05-14:
Regarding the actual state of low-power management on STM32, it works well and is easy to use as it is know with `uart_stdio_rx` being disabled by default and when not using the shell in an application: for an autonomous application (like a trendy LoRaWAN application), calling unblock from the application code or simply setting PM_BLOCKER_INITIAL macro from the Makefile does the job pretty well. I have good examples for this.

Enabling `uart_stdio_rx` by default would break that and from the users perspective this is not good IMHO as it will require them to explicitly disable something not directly related to make low power work out of the box.

If you need to read some characters from stdin (over UART or whatever) but don't need the shell, then you could just pull `uart_stdio_rx` explicitly.

jcarrano, 2019-05-14:
@aabadie For sure that have to discuss whether we disable by default (or not) features that conflict with PM. Also, I have no idea about PM in STM32, so I won't question it.

This discussion, however, is about a different topic altogether. It is about a change that was _merged_ which changed the behaviour of RIOT in a way that applications are now _broken_.

aabadie, 2019-05-14:
> There needs to be a level of pseudo modules, e.g., "stdio_rx (or just stdin)" which depends on "uart_stdio_rx" iff "uart_stdio" is selected...

That would be my preferred solution (was about to propose something like this when I read this in the initial issue comment)/

aabadie, 2019-05-14:
> It is about a change that was merged which changed the behaviour of RIOT in a way that applications are now broken.

Let me rephrase the end of this sentence to avoid any confusion:
`It is about a change that was merged which changed the behaviour of RIOT in a way that applications not using the shell but expecting some input from stdio are now broken.`

jcarrano, 2019-05-14:
> ...applications not using the shell but expecting some input...

Yes, sure, not _all_ applications are broken- that's a good thing.

On the side of broken ones, for example, would be any app that uses "ethos" but does not explicitly declare it uses "uart_rx":

https://github.com/RIOT-OS/RIOT/blob/e4bc5d4718558ad308625b30841ee523dc136042/sys/stdio_uart/stdio_uart.c#L65

The issue is that we _assume_ there are external application we don't know of. If it was just about our examples and test apps then OK, no big deal, we add the dependency manually.

cladmi, 2019-05-15:
> I understand your frustration that something you expected working
> suddenly broke. I'm sorry that happened.
> 
> Let's move on and fix it.

Sure I wait for the fix !

cladmi, 2019-06-13:
`tests/lua_loader` also broken https://github.com/RIOT-OS/RIOT/pull/11686

PR #11598 (2019-06-20): Add stdin pseudo module and set it as dependency for applications using getchar

### Contribution description

fixes #11525

This PR introduces the stdio_rx pseudomodule that is now required when an application needs input from UART (or any other bus used for stdio), e.g. when using the getchar function.

`git grep getchar` shows that 3 applications in the RIOT code base are using getchar: `tests/xtimer_usleep`, `tests/posix_time` and another one related to nrf in `dist/tools`. The Makefiles of these application has been updated.

This PR also introduces a new test application to avoid regress [... cut]
closed 2019-06-20 by kaspar030