RIOT-OS/RIOT #11526: Timer gets disabled after interrupt

Alexander-M1, 2019-05-14:
#### Description
Periodic timer interrupt for cyclic time sensitive tasks. I am not sure, if this is a bug, or intentional feature, but on timer interrupt the bellow line would disable any future interrupts:
`//TIMER_0_DEV.INTENCLR.reg = TC_INTENCLR_MC0;`

As part of RIOT > cpu > samd21 > periph > timer.c, function void TIMER_0_ISR(void):
   ```
    /* Timer 0, channel 0 */
    /* Interrupt pending and interrupt enabled */
    if (TIMER_0_DEV.INTFLAG.bit.MC0 && TIMER_0_DEV.INTENSET.bit.MC0) {
    	/* Clear interrupt pending flag (should be auromatic, when reading */
        TIMER_0_DEV.INTFLAG.reg = TC_INTFLAG_MC0;
    	/* Disable Match or Capture interrupt */
        //TIMER_0_DEV.INTENCLR.reg = TC_INTENCLR_MC0;
    	/* Call callback */
        if(config[TIMER_0].cb) {
            config[TIMER_0].cb(config[TIMER_0].arg, 0);
        }
    }
```

### Useful links

kaspar030, 2019-05-14:
> I am not sure, if this is a bug, or intentional feature, but on timer interrupt the bellow line would disable any future interrupts

That is intentional. The file is part of the periph/timer implementation, which allows only one active timer at a time (it is a slim abstraction of hardware timers).

The corresponding ```timer_set_absolute()``` enables the interrupt, e.g. here:

https://github.com/RIOT-OS/RIOT/blob/e4bc5d4718558ad308625b30841ee523dc136042/cpu/samd21/periph/timer.c#L147

If the timer enable flag wouldn't be disabled, the same callback would be re-executed every time the timer has overflowed and counted back up to the compare value.

Alexander-M1, 2019-05-15:
> That is intentional. The file is part of the periph/timer implementation, which allows only one active timer at a time (it is a slim abstraction of hardware timers).
>
> The corresponding [...]

Wouldn't the correct approach for doing that include enabling one-shop operation?

[image]

closed 2020-03-02 by stale[bot]