RIOT-OS/RIOT #11544: i2c_read_regs / i2c_write_regs and endianess

fedepell, 2019-05-19:
#### Description
I'm working on a 8kb EEPROM connected via I2C to a SAML21. I'm having strange wrap around issues on memory writing. After some debug my feeling is that the `i2c_write_regs` and `i2c_read_regs` functions do not take into account the endianess of the CPU and do work only with big endian code.

Example trying to write to register 0x10 the value 1:
`   
i2c_write_reg(I2C_DEV(0), 0x50, 0x10, 1, I2C_REG16);
`
Putting some debug prints I see that the write request will generate an address on the bus:
```
0x10
0x0
```
While I would expect the MSB to come first (by specs, but please correct me if I'm wrong as I'm not very used to I2C):
```
0x0
0x10
```
Looking at `i2c_write_regs` the register is received as an `uint16_t` which is then passed down to `i2c_write_bytes` that gets it as `const void *` that is then written physicall in a loop as `const uint8_t *`.
Therefore in little endian machines the LSB will come first, while for big endian machines MSB will come first (as it should).

I think the two regs function should take the endianess into account and always pass the registers data with MSB first to the underneath functions. I believe this should be the place to do it, so the calls to the CPU specific functions underneath already always receive the data as it should be written.

I would gladly do a PR once I get confirmation that I'm not seeing something wrong. Worst case I would suppose this should be documented if it is correct as it is.

Swapping beforehand the register data in my application seems to solve all the troubles.

#### Steps to reproduce the issue
Write some i2c registers using `i2c_write_regs` and see if the data gets written where it should.

#### Versions
RIOT master

fedepell, 2019-05-19:
@dylad : pinging you as I see you as author of drivers/periph/i2c

keestux, 2019-05-19:
@benpicco I'm not sure if this is a bug. The I2C spec doesn't talk about register addresses, as far as I can see, only about device addresses (7 or 10 bit). The spec also doesn't talk about endianness, at all.

Now, there aren't many uses of 16 bit register addresses. So that could be an explanation why nobody bothered. I happen to know of one in the SHT2x driver where the identifier is read from a 16 bit register address. The SHT2x documentation isn't very clear about the actual address. It just gives an example of the bytes to transfer and I have adapted the register address to make it work with the I2C driver.

Could it perhaps be that endianness interpretation is a device issue?

fedepell, 2019-05-19:
@keestux : many thanks for checking this out! I don't have indeed much practice with I2C so that's why I was very cautious on this before doing a PR or so :) 

In my case the device is an ST EEPROM model M24C64, you can find the datasheet [here](https://www.st.com/resource/en/datasheet/m24c64-f.pdf). The datasheet indeed is not so very explicit about ordering, but it mentions in 5.1 first the most significant address byte. Indeed by using it the other way around then I get into wraparound problems as mentioned.

One more similar chip I found from Microchip, [datasheet here](http://ww1.microchip.com/downloads/en/DeviceDoc/21754M.pdf) also uses the most significant byte first and here it's explicitly seen in figure 6-1.

I've searched a bit around and found a couple of examples and they were usually using big endianess for such memories, but indeed also found for example [here](https://www.raspberrypi.org/forums/viewtopic.php?t=122923) a mention that it may purely depend on the device (in this case is the data, not the address).

I wonder if _registers_ are not in the I2C specs and are device dependant, then how "useful" are the `_reg_` functions? Or maybe we could put a big fat warning in the doc.

dylad, 2019-05-20:
Thanks for spotting @fedepell 
For the story:
The I2C rework/AI are inherit from #6577 and related discussions in #9548 #9539.

During the rework and to minimize the embargo on I2C, we focus only on extending the API and try not to destroy all previous work. So basically, we did what we could for ease new features (like 16-bit registers)  but we did not test it because we were lacking time and hardware so we just focus on what we have.
The current API is still not perfect and I would be more than happy to enhance it !

keestux, 2019-05-20:
> In my case the device is an ST EEPROM model M24C64, you can find the datasheet [here](https://www.st.com/resource/en/datasheet/m24c64-f.pdf). The datasheet indeed is not so very explicit about [...]

The datasheet is not very clear. OK, you proved with trial and error that the register address must be send big endian.

> 
> One more similar chip I found from Microchip, [datasheet here](http://ww1.microchip.com/downloads/en/DeviceDoc/21754M.pdf) also uses the most significant byte first and here it's explicitly seen in figure 6-1.

That datasheet is really clear about it, memory address must be send big endian.

Still, I couldn't find any I2C documentation that formally describes this big endian behaviour. But let me quickly add that I'm no I2C expert.

Have you already thought about a PR?

fedepell, 2019-05-21:
@dylad : many thanks for the links! I'll check the tickets in detail for sure!

@keestux: I've looked a bit further but again came until now to a similar conclusion as yours, as it seems like it's device specific (althought inclined, at least for EEPROMS which are one of the fews that need 16 byte adressing as other simpler I2C devices don't have that much registers, to be big endian) and there is not even mention of "register" in I2C.

About a possible PR:
assuming we need to handle both cases (if we conclude it's device specific) I would go for having therefore a set of these reg function, one that works with big endian and one with little endian, and in each of these we make sure the format is correct (using the functions defined in `include/byteorder.h`).
Either we keep the current one named as it is (and force it to be big endian, as I think it was originally tested for) and create just and additional little endian one, or we remove the current one (but this is an API change) and add explicitly a _be and a _le.
Let me know if this sounds reasonable (also which would be best of the two API solutions), I can sketch the code then in a PR.

kaspar030, 2019-05-21:
> Either we keep the current one named as it is (and force it to be big endian, as I think it was originally tested for) and create just and additional little endian one, or we remove the current one (but this is an API change) and add explicitly a _be and a _le.

+1 for keeping the current one, but defining it to work in big endian (network) byte order.
It'll require the least changes (probably almost none apart from documentation as we currently only support little endian machines, and on those the current ```uint8_t *ptr = (uint8_t *)&uint16_value)``` will work that way).

As soon as we support a big endian host, its i2c implementation needs to make sure the byte order is correct. Drivers who need little endian byte order can either adapt their defines statically or convert on every i2c reg call.

fedepell, 2019-05-21:
Just sketched down a possible PR to evaluate:
https://github.com/RIOT-OS/RIOT/compare/master...fedepell:i2c_reg_endianess?diff=split&expand=1

dylad, 2019-05-21:
Another solution could be introduce one or several flags for specific endian operations.
e.g keep the current API as it, add a new flag to translate reg or data to big endian and pass it the each functions that need it.

fedepell, 2019-05-21:
@dylad : oh I see, indeed, so use some of unused bits of the passed `flags` (ie. we create a `I2C_ADDR_LE` or so). Sounds also good, even better more compact code wise!

Let's see other's comments in case and I can then rewrite!

keestux, 2019-05-21:
@fedepell be careful not to mix up "address" and "register (address)". I thought the discussion here was about the register address. In that sense your suggested `I2C_ADDR_LE` makes no sense to me.
The enum `i2c_flags_t` is defining bits that can be or-ed together. There is a value for `I2C_REG16`. If we want two choices (little and big endian) then maybe we can add `I2C_REG16_LE` and have the old for big endian.

keestux, 2019-05-21:
Also notice that we avoided endian issues for 10bit addressing :-) Mostly. Only ESP32 actually does support 10bit addressing.
```
    if (flags & I2C_ADDR10) {
        return -EOPNOTSUPP;
    }
```

fedepell, 2019-05-22:
@keestux : sorry for the addr/reg confusion, was writing the comment in a hurry :)

I've now done a proposal in line with @dylad suggestion which is also much cleaner. So added a flag `I2C_REG16_LE` which works for LE 16-bit registers and made sure the original `I2C_REG16` gets converted to BE. 

Here is the diff:
https://github.com/RIOT-OS/RIOT/compare/master...fedepell:i2c_reg_endianess_2?diff=split&expand=1 

Hope looks in better shape now!

Thanks!
F.

fedepell, 2019-05-27:
I opened a PR (#11585) to make it easier to discuss. Thanks!

maribu, 2019-08-16:
The `i2c_write_regs()` in `cpu/stm32_common/periph/i2c_1.c` has not yet been updated accordingly. As it does rely on the common implementation of `i2c_read_regs()` in `drivers/periph_common/i2c.c`, different endianess is used for reading and writing registers :-/

Btw: Is there a reason why `i2c_1.c` does have a custom `i2c_write_regs()`? If not, I'd say remove it to make maintaining easier.

@MrKevinWeiss: Do you want to address this?

MrKevinWeiss, 2019-08-16:
> Btw: Is there a reason why i2c_1.c does have a custom i2c_write_regs()? If not, I'd say remove it to make maintaining easier.

This is because the `i2c_write_bytes` cannot do a split frame continuous write as the next byte must be set before the current byte is transmitting or a stop must be sent.  To support the common driver `i2c_write_regs` we must be able to call write_bytes with `I2C_NOSTOP` which would be the register then a write_bytes with `I2C_NOSTART` which would be the buffer.  This can be done but then we would not be able to support a repeated start write.
It has to do with the `I2C_CR2_RELOAD` bit.  I implemented it in a way where `i2c_write_reg` is continuous, as the API states, and if a repeated start is needed it can be done with split frame.

I think the only other way is to allocate the register space before the buffer and try to send the whole thing.  I didn't think that was a valid solution though.

Also it was my bad for missing that it needed special handling...  As an apology I present to you #12023.
:smile:

PR #11585 (2019-05-29): periph/i2c: handle i2c register write/read helper function correctly in big endian
### Contribution description
In RIOT I2C we have helper functions to read/write I2C devices (mostly EEPROMS) that need a 16-bit addressing. The standard does not dictate the endianess of this addressing but all the examples found are big-endian. The current code doesn't correctly force a translation to big-endian, therefore it will not work on little-endian architectures.
This PR fixes this by making sure the 16-bit address is always converted into big-endian.

### Testing procedure
The test in `tests/periph_i2c [... cut]
PR #12023 (2019-08-16): cpu/stm32: Fix 16 bit reg endianness for i2c_1

### Contribution description

Since i2c_1 i2c_write_regs is cpu specific reg endianness must be swapped there
Change reg from little endian to big endian

### Testing procedure
Use any STM32 F0, F3, F7, L0 and L4 board.
Read a 16 bit register and make sure it is big endian and not little endian.
You can use [PHiLIP](https://github.com/riot-appstore/PHiLIP/) as the slave device since it can be set to 16 bit register mode and have the endianness controlled.

### Issues/PRs references

Fixes #11544
closed 2019-08-16 by maribu