RIOT-OS/RIOT #11584: no output when running  RIOT/examples/hello-world/

jhx2002, 2019-05-27:
setup:

- my board is stm32f407vet6

-  `cd RIOT-19.04/examples/hello-world`

- native mode is OK

- `make BOARD=stm32f4discovery all flash` ,  using stlink-v2 .  is OK

-  use TTL-USB converter lineup usb on UBUNTU18.04 and J6 on stm32f407vet6

[image]

- `make BOARD=stm32f4discovery BAUDRATE=115200 PORT=/dev/ttyUSB0 term`

   This step occur problem,  log is 
```
huaxing@X1:~/oproj/riot/RIOT/examples/hello-world$ sh run.sh
/bin/sh: 1: arm-none-eabi-gcc: not found
/home/huaxing/oproj/riot/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB0" -b "115200"
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2019-05-27 10:32:20,966 - INFO # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type '/exit' to exit.

```

jhx2002, 2019-05-27:
[image]

jhx2002, 2019-05-27:
[image]

PeterKietzmann, 2019-05-27:
@jhx2002 the stm32f4discovery configures pins PA2/3 for STDIO TX/RX (see [periph_conf.h](https://github.com/RIOT-OS/RIOT/blob/master/boards/stm32f4discovery/include/periph_conf.h#L66)). Please connect your converter to these pins and retry. Furthermore, your board is not officially supported by RIOT so don't be surprised when some configurations don't work out of the box.

dylad, 2019-05-27:
> /bin/sh: 1: arm-none-eabi-gcc: not found

Did you install your toolchain and did you add it to your PATH variable ?

jhx2002, 2019-05-27:
The example Success  after your suggestion.  @PeterKietzmann
Yes.  I install it in docker images, build and flash in docker container.
 But run outside of docker container @dylad 

Thank you.

jcarrano, 2019-05-27:
Solved

closed 2019-05-27 by jcarrano