RIOT-OS/RIOT #11586: Connect to MQTT-SN Gateway failed from RIOT virtual machine

jhx2002, 2019-05-27:
Please see:

https://github.com/eclipse/paho.mqtt-sn.embedded-c/issues/152

smlng, 2019-05-28:
Hi @jhx2002 welcome to RIOT,

you must not mix IP versions, i.e. don't bind the MQTT broker to `192.168.0.104` (IPv6) but rather configure it to use IPv6 with `::` (any address). 

---

Further, can you please elaborate and clarify your setup a bit more. What is you overall goal you want to achieve from a high level perspective, i.e. without talking about what runs where and how. 

If I understood your setup correctly there are 3 players/nodes in you scenario:

1. a RIOT node running emcute example
2. a MQTT-Broker/Gateway
3. a MQTT Linux (?) node using MQTT-SN tools

The RIOT node is running in an VM and the other 2 parts are running on the host, where the VM is running too, right - so everything in one box?

jhx2002, 2019-05-29:
Thank you @smlng .
Test a case, running gateway with ipv6. got error:
```

20190529 181941.706 MQTT-SN/gateway1 has been started.

 ConfigFile: ./gateway.conf
 ClientList: ./clients.conf
 PreDefFile: ./predefinedTopic.conf
 SensorN/W:  UDP Multicast 225.1.1.1:1883 Gateway Port 10000
 Broker:     ::C0A8:0078 : 1883, 8883
 RootCApath: (null)
 RootCAfile: (null)
 CertKey:    (null)
 PrivateKey: (null)

20190529 181943.000   error:getaddrinfo(): Address family for hostname not supported
Error: BrokerSendTask: MQTT-SN/gateway1QoS-1 can't connect to the broker. errno=0 Success 

```

closed 2019-08-17 by jhx2002