RIOT-OS/RIOT #11603: NATIVEINCLUDES Does not include USEMODULE_INCLUDES from log_printfnoformat

crest42, 2019-05-28:
#### Description
If I try to build an application, which uses a module (log_printfnoformat) that requires an include path in USEMODULE_INCLUDES, but my build fails with:

```
➜  RIOT/tests/log ‹master*› » QUIET=0 LANG=C make all 
Building application "log" for "native" with MCU "native".

'/home/robin/git/RIOT/dist/tools/genconfigheader/genconfigheader.sh' -DDEVELHELP -Werror -Wall -Wextra -pedantic -std=gnu99 -m32 -fstack-protector-all -ffunction-sections -fdata-sections -DDEBUG_ASSERT_VERBOSE -DRIOT_APPLICATION=\"log\" -DBOARD_NATIVE=\"native\" -DRIOT_BOARD=BOARD_NATIVE -DCPU_NATIVE=\"native\" -DRIOT_CPU=CPU_NATIVE -DMCU_NATIVE=\"native\" -DRIOT_MCU=MCU_NATIVE -fno-delete-null-pointer-checks -fdiagnostics-color -Wstrict-prototypes -Wold-style-definition -fno-common -Wall -Wextra -Wformat=2 -Wformat-overflow -Wformat-truncation -Wmissing-include-dirs -DMODULE_AUTO_INIT -DMODULE_BOARD -DMODULE_CORE -DMODULE_CORE_MSG -DMODULE_CPU -DMODULE_LOG -DMODULE_LOG_PRINTFNOFORMAT -DMODULE_NATIVE_DRIVERS -DMODULE_PERIPH -DMODULE_PERIPH_COMMON -DMODULE_PERIPH_GPIO -DMODULE_PERIPH_PM -DMODULE_PERIPH_UART -DMODULE_SYS -DRIOT_VERSION=\"2019.07-devel-479-g3f425\" \
	| '/home/robin/git/RIOT/dist/tools/lazysponge/lazysponge.py' --verbose '/home/robin/git/RIOT/tests/log/bin/native/riotbuild/riotbuild.h'
Keeping old /home/robin/git/RIOT/tests/log/bin/native/riotbuild/riotbuild.h (7873b9dce997c1b8e120b6302e22f558)
DIRS=" " "make" -C /home/robin/git/RIOT/tests/log -f /home/robin/git/RIOT/makefiles/application.inc.mk
make[1]: Entering directory '/home/robin/git/RIOT/tests/log'
"make" -C /home/robin/git/RIOT/boards/native
make[2]: Entering directory '/home/robin/git/RIOT/boards/native'
gcc \
	-DRIOT_FILE_RELATIVE=\"boards/native/board.c\" \
	-DRIOT_FILE_NOPATH=\"board.c\" \
	-Werror -Wall -Wextra -pedantic -std=gnu99 -m32 -fstack-protector-all -ffunction-sections -fdata-sections -fno-delete-null-pointer-checks -fdiagnostics-color -Wstrict-prototypes -Wold-style-definition -fno-common -Wall -Wextra -Wformat=2 -Wformat-overflow -Wformat-truncation -Wmissing-include-dirs -include '/home/robin/git/RIOT/tests/log/bin/native/riotbuild/riotbuild.h'  -DNATIVE_INCLUDES -I/home/robin/git/RIOT/boards/native/include/ -I/home/robin/git/RIOT/core/include/ -I/home/robin/git/RIOT/drivers/include/ -I/home/robin/git/RIOT/cpu/native/include -I/home/robin/git/RIOT/sys/include -MD -MP -c -o /home/robin/git/RIOT/tests/log/bin/native/board/board.o /home/robin/git/RIOT/boards/native/board.c
In file included from /home/robin/git/RIOT/boards/native/board.c:21:
/home/robin/git/RIOT/core/include/log.h:98:10: fatal error: log_module.h: No such file or directory
 #include "log_module.h"
          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/home/robin/git/RIOT/Makefile.base:83: /home/robin/git/RIOT/tests/log/bin/native/board/board.o] Error 1
make[2]: Leaving directory '/home/robin/git/RIOT/boards/native'
make[1]: *** [/home/robin/git/RIOT/Makefile.base:20: ALL--/home/robin/git/RIOT/boards/native] Error 2
make[1]: Leaving directory '/home/robin/git/RIOT/tests/log'
make: *** [/home/robin/git/RIOT/tests/log/../../Makefile.include:464: /home/robin/git/RIOT/tests/log/bin/native/application_log.a] Error 2

```

I Only tested this with "USEMODULE = log_printfnoformat" so far, but the problem may also appear with other (pseudo-)modules.

Problem Description:

"USEMODULE = printfnoformat" invokes sys/log/Makefile.include

```
ifneq (,$(filter log_printfnoformat,$(USEMODULE)))
  USEMODULE_INCLUDES += $(RIOTBASE)/sys/log/log_printfnoformat
endif
```

As you can see in the code the include directory of log_printfnoformat is correctly added to USEMODULE_INCLUDE. USEMODULE_INCLUDE is later appended to INCLUDES. Later INCLUDES is overwritten in boards/native/Makefile

```
MODULE = board

DIRS = drivers

include $(RIOTBASE)/Makefile.base

INCLUDES = $(NATIVEINCLUDES)
```

I assume this should only add $NATIVEINCLUDES to INCLUDES and thus the last line should look like the following: 

```
INCLUDES += $(NATIVEINCLUDES)
```

Changing this line results in working builds on my system. If anybody could back my assumption or provide a better solution I will happily create a small PR. 

Alternatively one could reduce impact heavily with the following change in sys/log/Makefile.include

```
diff --git a/sys/log/Makefile.include b/sys/log/Makefile.include
index 05720548d..b7320744f 100644
--- a/sys/log/Makefile.include
+++ b/sys/log/Makefile.include
@@ -1,3 +1,6 @@
 ifneq (,$(filter log_printfnoformat,$(USEMODULE)))
   USEMODULE_INCLUDES += $(RIOTBASE)/sys/log/log_printfnoformat
+  ifeq ($(BOARD),native)
+    NATIVEINCLUDES += -I$(RIOTBASE)/sys/log/log_printfnoformat
+  endif
 endif
```
#### Steps to reproduce the issue

1. Create a new Application which includes "log.h" in its main.c
2. Add "USEMODULE += log_printfnoformat" to the applications Makefile
3. Run "BOARD=native make all"

#### Expected results
Build should run without any error

#### Actual results
Build fails with compiler errors due to missing include path

#### Versions

```
Operating System Environment
-----------------------------
       Operating System: "Arch Linux" 
                 Kernel: Linux 5.1.2-arch1-1-ARCH x86_64 unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (GCC) 8.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 9.1.0
                avr-gcc: avr-gcc (GCC) 9.1.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: clang version 8.0.0 (tags/RELEASE_800/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.1.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.14.4
               cppcheck: Cppcheck 1.86
                doxygen: 1.8.15
                 flake8: missing
                    git: git version 2.21.0
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 3.7.3
                python2: Python 2.7.16
                python3: Python 3.7.3
             coccinelle: spatch version 1.0.7 compiled with OCaml version 4.07.1
```

aabadie, 2019-05-29:
This is fixed by #11572

crest42, 2019-05-29:
Thanks. Looks like it was a little bit too late to use the search function correctly -.-

crest42, 2019-05-29:
Since you referenced this issue should I close it or leave it open ?

aabadie, 2019-05-29:
Better leave it open to have a trace of the issue. And this issue proposes an alternative fix to #11572 that should be discussed.

closed 2019-08-06 by aabadie

jzySaber1996, 2024-05-10:
Hello, we're investigating the issue reports in security. Since we observed that this issue may relate to a potential vulnerability, has it been disclosed in CVE already? Hope to receive your reply.